ROBOTIS-GIT/dynamixel-workbench #176: Controlling individual positions/velocities/torques of several Dynamixels through pubsub

Achllle, 2018-08-24:
I'm looking for a way to publish angular velocities to a topic for several motors at once. How can this be done without writing a C++ node?

I launched the `velocity_control` example node from `dynamixel_workbench_controllers` which exposes a `/cmd_vel` topic but that broadcasts the same command to all servos.

This is a very basic feature and I can't find it anywhere in the documentation.

routiful, 2018-08-27:
Sorry, we have provided velocity_control example using only two motors.

If you want to add motors, you are supposed to modify code.

Achllle, 2018-08-27:
I modified the velocity_control example node to subscribe to a topic that can take in velocity command messages for an arbitrary number of dynamixels through the `Float64MultiArray` message. Let me know if you're interested and want me to open a PR

elthef, 2018-08-27:
I interested. Possible to send a copy to me.
Edwin

Sime92, 2018-08-28:
@Achllle Your modification sounds useful, I am also interested in a copy.
I am trying to get multiple Dynamixel PRO servos to follow a trajectory generated by moveit. So far the only easy possibility I see is to write a trajectory controller which outputs velocities. Is there a more convenient way to do this? Otherwise your modification would be of great help.
Thanks in advance,
Simon

Achllle, 2018-08-30:
@elthef @Sime92 I opened a PR for it #178

hamidnrn, 2018-09-14:
I need velocity control fro Dynamixel x430-w210r? could u plz provide it

Achllle, 2018-09-17:
@hamidnrn Please open a separate issue as you are requesting a different feature.

closed 2018-11-15 by gpwhd07