ROBOTIS-GIT/dynamixel-workbench #177: udev rules / symlinks unsupported?

Achllle, 2018-08-25:
I can connect to a series of Dynamixels using the U2D2 when I specify the USB port it's been plugged into (e.g. `/dev/ttyUSB0`). I assigned a symlink to the U2D2 using the following rule that also sets the permissions:

```
SUBSYSTEM=="usb", ATTR{idVendor}=="0403", ATTR{manufacturer}=="FTDI", MODE:="0777", SYMLINK+="U2D2"
```
As a result, the device can be found through `ls -l /dev/U2D2`:

```
lrwxrwxrwx 1 root root 15 Aug 24 22:04 /dev/U2D2 -> bus/usb/001/022
```
Which definitely corresponds to the U2D2 (through checking bus and port with `lsusb`)
In my launch file I changed the parameter `device_name` to "/dev/U2D2". When I launch the file now however I get an error saying that the dynamixel manager couldn't find the motors: `Not found Motors, Please check id and baud rate`.

Am I missing something? Otherwise it seems that the workbench can't handle symlinks or udev rules.

gpwhd07, 2018-08-27:
Hi @Achllle ,
I need more information.
Could you run it before changing it from "/ dev / ttyUSB0" to "/ dev / U2D2"?

If you didn't check that, I think the dynamixel id or baud rate was not correct. 
You can check that, refer to [here](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/).
http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/

Achllle, 2018-08-27:
Yes I can get it to work if I use `/dev/ttyUSB0`. The only difference is when the udev rule is loaded. ID and baud rate are definitely correct.
Can you get this to work on your machine?

Achllle, 2018-08-30:
That did the trick, thanks a lot!

closed 2018-08-30 by Achllle