ROBOTIS-GIT/dynamixel-workbench #179: Using dynamixel_workbench functions in other packages

AlvaroFerGar, 2018-08-31:
Hello everyone!!

I'm using ROS kinetic in my Raspberry Pi 2 with ubuntu MATE. I comunicate my AX-12+ with an USB2Dynamixel.

My goal was to be able to control the position and the speed of the servos. I followed the tutorials of the [emanual](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/) and created my own _my_dynamixel_workbench-tutorial_ package.

As expected, I was able to run the position controller tutorial but no the velocity controller tutorial, as long as is not designed for my type of motors. So I decided to make a simple program that uses the _dynamixel_workbench_toolbox_ to change the _[Moving Speed](http://support.robotis.com/en/product/actuator/dynamixel/ax_series/dxl_ax_actuator.htm)_ of my motors with the _goalSpeed_ function.
 
For this purpose i use one terminal to:
  ```
  $ roslaunch dynamixel_workbench_single_manager single_manager.launch
  ```
(_roscore_ and _position_control.launch_ also tested)
 
And another terminal to run my node:

  ```
  $ rosrun my_dynamixel_workbench-tutorial vel_ax12
  ```
Where _vel_ax12_ correspond to this code, which doesn't leave any arror when is built:
```
//vel_ax12.cpp

#include "/opt/ros/kinetic/include/dynamixel_workbench_toolbox/dynamixel_workbench.h"
#include "ros/ros.h"
int main(int argc, char **argv)
{
  int id=10; //This corresponds with the id of one of my motors
  int goal=10; 

DynamixelWorkbench prueba;
prueba.goalSpeed(id,goal);

ROS_INFO("sent!\n");

while(1){}
}
```
 After executing this I only get a nice:
**`Segmentation fault`** 

I leave here my CMakeLists.txt, that can be my main problem:
```
cmake_minimum_required(VERSION 2.8.3)
project(my_dynamixel_workbench-tutorial)

add_compile_options(-std=c++11)

find_package(catkin REQUIRED COMPONENTS
  roscpp
  std_msgs
  dynamixel_workbench_toolbox
)

catkin_package(
  INCLUDE_DIRS include
# LIBRARIES dynamixel_workbench
  CATKIN_DEPENDS roscpp std_msgs
#  DEPENDS system_lib
)

###########
## Build ##
###########

include_directories(
# include
  ${catkin_INCLUDE_DIRS}
)
INCLUDE_DIRECTORIES(/opt/ros/kinetic/include/dynamixel_workbench_toolbox)
LINK_DIRECTORIES(/opt/ros/kinetic/include/dynamixel_workbench_toolbox)

add_executable( vel_ax12  src/vel_ax12.cpp )
target_link_libraries(vel_ax12 ${catkin_LIBRARIES})
add_dependencies(vel_ax12 ${catkin_EXPORTED_TARGETS})
```
I know it can be a beginner question but I didn't find any tutorial on how to use the toolbox functions on my personal code.

Thank you very much in advance!!!

routiful, 2018-09-07:
Hi :)

You can not need to launch `single_manager.launch`

Attached example might be help you to modify your `vel_ax12.cpp` file 

https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/OpenCR/examples/08.%20DynamixelWorkbench/k_Speed/k_Speed.ino

I think your CMakelist looks fine :)

AlvaroFerGar, 2018-09-10:
Hello and thank you very much for the answer!

I modified my `vel_ax12.cpp` file to look more like the `.ino` that you have shared.

```
//vel_ax12.cpp

#include "/opt/ros/kinetic/include/dynamixel_workbench_toolbox/dynamixel_workbench.h"
#include "ros/ros.h"

#define BAUDRATE  1000000
#define ID 10

int main(int argc, char **argv)
{
DynamixelWorkbench dxl_wb;

dxl_wb.begin("/dev/ttyUSB0",BAUDRATE);

//dxl_wb.ledOn(ID);
//dxl_wb.ping(ID);

dxl_wb.wheelMode(ID);

dxl_wb.goalSpeed(ID,goal);
sleep(3000);
dxl_wb.goalSpeed(ID,-1*goal);

while(1)
{}

}
```

After typing in the terminal the `sudo chmod a+rw /dev/ttyUSB0` instruction, I executed the node with an active _roscore_ and it continues to leave the _Segmentation fault_ message. The use of `dxl_wb.ping`, `dxl_wb.ledOn` or `dxl_wb.jointMode` gives the same error. However, the use of `dxl_wb.begin()` alone doesn't give any problems.

Although it seems like a simple problem I didn't find the way to solve it   :/

Thanks in advance!!

routiful, 2018-09-11:
You are supposed to `scan` or `ping` to get Dynamixel.
I give you a example, but this is not tested

```
//vel_ax12.cpp

#include <dynamixel_workbench_toolbox/dynamixel_workbench.h> // If this line is not working, please check your CMakeLists and package.xml 
#include <ros/ros.h>

#define BAUDRATE  1000000
#define ID 10

int main(int argc, char **argv)
{
DynamixelWorkbench dxl_wb;

dxl_wb.begin("/dev/ttyUSB0",BAUDRATE);

dxl_wb.ping(ID);
dxl_wb.ledOn(ID);

dxl_wb.wheelMode(ID);

dxl_wb.goalSpeed(ID,goal);
sleep(3000);
dxl_wb.goalSpeed(ID,-1*goal);

while(1)
{}

}
```

routiful, 2018-09-12:
We provided other example for wheel mode of Dynamixel. Please check below node

https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/master/dynamixel_workbench_controllers/src/velocity_control.cpp

AlvaroFerGar, 2018-09-12:
Yes, the `velocity control` is in the tutorials of the _emanual_ and it failed when I ran it. I think that it doesn't work for my Dynamixel AX-12+ 
```
  printf("-----------------------------------------------------------------------\n");
  printf("        dynamixel_workbench controller; velocity control example       \n");
  printf("              -This example supports MX2.0 and X Series-               \n");
  printf("-----------------------------------------------------------------------\n");
  printf("\n");
```
However, I will take a look at the example and make some changes in `vel_ax12.cpp` to see if I can make it work. I'll post if I have nice results.

Thanks!

routiful, 2018-09-12:
Great. If you have any problem, please feel free to open issue :)

closed 2018-09-12 by routiful