ROBOTIS-GIT/dynamixel-workbench #183: position_control

auliakhilmi, 2018-09-20:
ISSUE TEMPLATE ver. 0.1.0

I've been followed this tutorial in http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/

I've been read the closed issue in https://github.com/ROBOTIS-GIT/dynamixel-workbench/issues/163 and https://github.com/ROBOTIS-GIT/dynamixel-workbench/issues/103

1. Which distro of ROS have you installed? Kinetic
2. Which OS have you installed? Ubuntu 16.04
3. Which Dynamixel have you used? MX-28(Version 40)

Launch File
```
<launch>
	
 <arg name="device_name"                default="/dev/ttyUSB0"/>
 <arg name="baud_rate"                  default="1000000"/>

 <arg name="scan_range"                 default="10"/>

 <arg name="profile_velocity"           default="200"/>
 <arg name="profile_acceleration"       default="50"/>

 <param name="device_name"              value="$(arg device_name)"/>
 <param name="baud_rate"                value="$(arg baud_rate)"/>

 <param name="scan_range"               value="$(arg scan_range)"/>

 <param name="profile_velocity"         value="$(arg profile_velocity)"/>
 <param name="profile_acceleration"     value="$(arg profile_acceleration)"/>

 <node name="position_control" pkg="dynamixel_workbench_controllers" type="position_control" required="true" output="screen"/>

 </launch>
```
 
4. Write down the commands you used in order

  ```    
  $ roscore
  $ roslaunch wb_tut position_control.launch
  ```
 
5. Copy and Paste your error message on the terminal

  ```
SUMMARY
========

PARAMETERS
 * /baud_rate: 1000000
 * /device_name: /dev/ttyUSB0
 * /profile_acceleration: 50
 * /profile_velocity: 200
 * /rosdistro: kinetic
 * /rosversion: 1.12.13
 * /scan_range: 10

NODES
  /
    position_control (dynamixel_workbench_controllers/position_control)

ROS_MASTER_URI=http://localhost:11311

process[position_control-1]: started with pid [6209]
-----------------------------------------------------------------------
  dynamixel_workbench controller; position control example             
-----------------------------------------------------------------------

MODEL   : MX-28
ID      : 1

-----------------------------------------------------------------------
================================================================================REQUIRED process [position_control-1] has died!
process has died [pid 6209, exit code -11, cmd /opt/ros/kinetic/lib/dynamixel_workbench_controllers/position_control __name:=position_control __log:=/home/eros/.ros/log/0dee29e0-bd2a-11e8-8701-e0ca944d3d91/position_control-1.log].
log file: /home/eros/.ros/log/0dee29e0-bd2a-11e8-8701-e0ca944d3d91/position_control-1*.log
Initiating shutdown!
================================================================================
[position_control-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
  ```
  
6. Please, describe detailedly what difficulty you are in 
    I did well tutorial for Single Manager in Command Line and GUI, and before I come to dynamixel_workbench I used to dynamixel_controller (http://wiki.ros.org/dynamixel_controllers) and did well.

Thanks for help

gpwhd07, 2018-09-27:
Hi, @auliakhilmi
I tested to control MX-28 using your Launch File.
It was successfully worked.

```
SUMMARY
========

PARAMETERS
 * /baud_rate: 1000000
 * /device_name: /dev/ttyUSB0
 * /profile_acceleration: 50
 * /profile_velocity: 200
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /scan_range: 10

NODES
  /
    position_control (dynamixel_workbench_controllers/position_control)

ROS_MASTER_URI=http://localhost:11311

process[position_control-1]: started with pid [9083]
-----------------------------------------------------------------------
  dynamixel_workbench controller; position control example             
-----------------------------------------------------------------------

MODEL   : MX-28
ID      : 1

-----------------------------------------------------------------------
```

Please check if your "position control" node is the most current version. 
And I want to know your more detail environment.

auliakhilmi, 2018-09-30:
I installed it by using this command
```
$ sudo apt-get install ros-kinetic-dynamixel-workbench
$ sudo apt-get install ros-kinetic-dynamixel-workbench-msgs
```
also with this command,
```
$ sudo apt-get install ros-kinetic-dynamixel-sdk
$ sudo apt-get install ros-kinetic-qt-build
```

or is it possible to crash with other dynamixel libraries such as (http://wiki.ros.org/dynamixel_controllers)?
I will try to reinstall my ROS and post the result here,

gpwhd07, 2018-10-15:
Thank you. 
You reinstall ROS, dynamixel-workbench  and its dependencies, then the source code work successfully, is that right?
Is this problem cleared?

If we don't have the other problem, this issue will be closed.

closed 2018-10-15 by gpwhd07