ROBOTIS-GIT/dynamixel-workbench #184: MX-28AT velocity control not working

Chick92, 2018-10-03:
Hi, I'm able to follow the tutorials to get joint mode to work on this servo - MX-28AT with Firmware version 40, however when i try and run the velocity tutorials, i get the following error - 

[ERROR] [1538527062.543018307]: Not found Motors, Please check id and baud rate
================================================================================REQUIRED process [velocity_control-2] has died!
process has finished cleanly
log file: /home/ubuntu/.ros/log/88acf0fc-c6a4-11e8-b2a8-30e37a9b6de7/velocity_control-2*.log
Initiating shutdown!
================================================================================

I'm able to run the single manager GUI and control the velocity of the servo in wheel mode, but i can't get it to work using the velocity controller. 

Does the Dynamixel workbench support veleocity control of the MX-28 series? 

Ubuntu 16.04 Ros Kinetic

routiful, 2018-10-03:
Hello :)

You are supposed to prepare two Dynamixels and set ID 1 and 2 with 57600 Baudrate.
Please check this first.

Chick92, 2018-10-03:
Baud rate is not the problem, I adjusted it accordingly. 

How would you go about doing velocity control of one servo?

routiful, 2018-10-03:
We have not provide that example. But we are preparing application more flexibility.

I think similar issue might be help you.
https://github.com/ROBOTIS-GIT/dynamixel-workbench/issues/176

Chick92, 2018-10-03:
You appear to have missed a key step in your tutorials then. What is the point of us spending multiple thousands of £ on robotis servos under the promise that they are ROS compatible when the manufacturer supported package is this poorly done?

closed 2018-10-03 by Chick92

routiful, 2018-10-03:
Sorry for inconvenience.

Have you any suggestions or contributions about this package, please write down your thought :) 
Because I am preparing big updates this year.