ROBOTIS-GIT/dynamixel-workbench #191: Low pub/sub rate for position_control

Scottapotamas, 2018-10-23:
## Hardware/Software

Using ROS Kinetic, on Ubuntu 16.04, Odroid XU4 (armhf).

1x MX-64AT flashed with latest firmware (uses protocol 2.0), running at 4Mbit through a U2D2 adaptor. No chained motors.

## Issue

Running the `dynamixel_workbench_controllers` position control node works, I successfully publish movements and achieve motion, can view /dynamixel_state etc, but it runs around 7Hz.

```
odroid@odroid:~/flying_fish$ rostopic hz /dynamixel_state 
subscribed to [/dynamixel_state]
average rate: 7.854
	min: 0.124s max: 0.128s std dev: 0.00147s window: 7
average rate: 7.830
	min: 0.124s max: 0.129s std dev: 0.00109s window: 15
```
I publish goal positions to `/goal_dynamixel_position` at 100hz.

___

If I run the `dynamixel_workbench_single_manager` monitor and controller, I can interact with the servo using CLI, and the message is publishing at 60Hz.

Any ideas?

routiful, 2018-10-23:
/dynamixel_state topic is filled with some frequently used information about dynamixel.
Because of using itemRead() function, the topic is slow compared by other topics.
I will make this more faster using syncRead function or other methods. 
From now, you can change it using syncRead function if your Dynamixels supports protocol 2.0.

Scottapotamas, 2018-10-23:
Thanks. Where should I look to implement syncread?

Is it expected that servo goals/joints are only transmitted at low rates (7-8hz)? Setpoint rate is more important to me than stats.

routiful, 2018-10-23:
Send value to address is used to syncWrite(). It worked fine with high frequency(e.g. 125hz).

https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/master/dynamixel_workbench_controllers/src/position_control.cpp#L48

I give you a simpe example how to handle syncRead(). 

https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/OpenCR/examples/08.%20DynamixelWorkbench/q_Sync_Read_Write/q_Sync_Read_Write.ino

gpwhd07, 2018-10-30:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

Scottapotamas, 2018-11-13:
I've already done that. 

Was 16, changed it to 1. Doesn't make any difference to the performance of the servo.

Scottapotamas, 2018-11-27:
I checked out the latest develop branch for the workbench and msgs.

Validated latency_timer is set to 1, dynamixel driver publishes at 30hz currently but the latency is still really high.

[image]

Scottapotamas, 2018-11-28:
@routiful is there a recommended example program/setup I could use to test the latency and troubleshoot the servo?

Are there any additional troubleshooting instructions you could recommend?

Scottapotamas, 2018-12-03:
@routiful Bump?

routiful, 2018-12-03:
Hi :)

Sorry for late response.
I upgraded new version of dynamixel-workbench.
Attached image you uploaded seems last version.

You can control dynamixel by service (/dynamixel_workbench/dynamixel_command) or trajectory_msgs(/dynamixel_workbench/joint_trajectory)

Furthermore, you can write PID gain by service. I recommend that position-p-gain set higher 4000.
http://emanual.robotis.com/docs/en/dxl/x/xm430-w210/#position-p-gain

Scottapotamas, 2018-12-03:
Thanks for the response, I've updated and built workbench and msgs from current master, and am running the node. It seems to run at 100hz with much lower CPU than previous setup, but I haven't been able to send position targets yet.

Is there documentation or example commands for:

- Manually testing position control by service command?
- Reading/Writing the PID gain by service?

I couldn't find which set of arguments to use with the following:
```
rosservice call /dynamixel_workbench/dynamixel_command "command: '_' id: 1 addr_name: '_' value: _"
```

Motor is on ID 1, I tried a few addresses and values out of [the MX64 table](http://emanual.robotis.com/docs/en/dxl/mx/mx-64/#led) and got `comm_result: False`.

In the source, https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/5a6114bb1b31ed107bf8adde14a503e6ce2e3a51/dynamixel_workbench_controllers/src/dynamixel_workbench_controllers.cpp#L742 it seems like it expects a string for the addr_name, rather than the register value?

routiful, 2018-12-03:
Sorry, I am writing e-manual for new version. It will be uploaded this week.

First, the new version is to be set /joint_states doesn't update when dynamixel is moving.
Because when simultaneously sync read and write to dynamixel make control_frequency would be lower. 

https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/5a6114bb1b31ed107bf8adde14a503e6ce2e3a51/dynamixel_workbench_controllers/src/dynamixel_workbench_controllers.cpp#L352

So you should delete if statement when you want to check dynamixel present position.
I will make this parameter, user can select it.

Second, I attached example for MX64

```
rosservice call /dynamixel_workbench/dynamixel_command "command:  id: 1 addr_name: 'P_Gain' value: 2000"
```

Scottapotamas, 2018-12-04:
Thanks for the response @routiful. I removed the `is_moving` check and didn't have any visible difference (but it makes it easier to check servo behaviour during testing).

The P_Gain setting didn't succeed, error was `[dynamixelCommandMsgCallback]: [DynamixelTool] Can't find Item`, but I was able to set `Goal_Position` and `Torque_Enable` through rosservice, which returned true. I tried a series of alternative spelling/capitalisations but the key wasn't found. Thoughts?

I used the windows configuration tool to set the P_Gain to 2000 (was 835) for now, and updated my ROS system to use the joint_trajectory functionality and retested the performance,

[image]

Seems to be about the same latency as previously. I also felt this latency while trying to control the servo using the onscreen position dial in R+ manager.

Can you tell me what latency _you_ would normally expect for a dynamixel in this kind of situation, I think 40-80ms should be achievable?

Scottapotamas, 2018-12-11:
@routiful Any chance you could comment on expected latency of a servo for this style of position tracking?

routiful, 2018-12-11:
Sorry for late reply.
When you publish joint_trajectory_msgs, the controller receives trajectory information and makes more smoothly trajectory using minimum jerk

https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/5cbf5e82f6c87b997b6789ded83bde76d88e2fc2/dynamixel_workbench_controllers/src/dynamixel_workbench_controllers.cpp#L719

So, the graph is different when you shown.

You can use service to test what you want to whow. You can use 'Goal_Position' value and 'P_Gain'

[image]

Achllle, 2019-08-02:
What was the solution to the latency issue @Scottapotamas @routiful ? I seem to be running into the same or a similar issue where latency is high and peaks when you send commands to the dynamixels (some spikes up to 0.2ms).

jackwilkinson255, 2020-07-22:
Hi @Scottapotamas 

Did you solve this problem? I'm having the same issue