ROBOTIS-GIT/dynamixel-workbench #192: Support for Arbotix-M

piraka9011, 2018-10-25:
ISSUE TEMPLATE ver. 0.1.0

Before you open issue, please refer to [ROBOTIS e-Manual](http://emanual.robotis.com/)

1. Which distro of ROS have you installed?
Kinetic

2. Which OS have you installed?
Ubuntu 16.04 LTS (Xenial Xerus)

3. Which Dynamixel have you used?
AX-18A x6
 
4. Write down the commands you used in order
  ```
roslaunch dynamixel_workbench_single_manager single_manager.launch
  ```
 
5. Copy and Paste your error message on terminal

  ```
  SUMMARY
  ========

  PARAMETERS
   * /baud_rate: 1000000
   * /device_name: /dev/ttyUSB0
   * /ping: False
   * /ping_id: 1
   * /rosdistro: kinetic
   * /rosversion: 1.12.13
   * /scan_range: 10

  NODES
    /
      single_dynamixel_controller (dynamixel_workbench_single_manager/single_dynamixel_controller)
      single_dynamixel_monitor (dynamixel_workbench_single_manager/single_dynamixel_monitor)

  auto-starting new master
  process[master]: started with pid [14750]
  ROS_MASTER_URI=http://localhost:11311

  Please Check USB Port authorization and
Baudrate [ex : 9600, 57600, 115200, 1000000, 2000000]
...Failed to find dynamixel!
================================================================================REQUIRED process [single_dynamixel_monitor-2] has died!
process has died [pid 14766, exit code 1, cmd /home/piraka/dynamixel_ws/devel/lib/dynamixel_workbench_single_manager/single_dynamixel_monitor __name:=single_dynamixel_monitor __log:=/home/piraka/.ros/log/2c32b1a8-d84d-11e8-b34c-702084074675/single_dynamixel_monitor-2.log].
log file: /home/piraka/.ros/log/2c32b1a8-d84d-11e8-b34c-702084074675/single_dynamixel_monitor-2*.log
Initiating shutdown!
================================================================================
  ```
  
6. Please, describe detailedly what difficulty you are in 

I was wondering whether this package supports the Arbotix-M controller? I have been using the old python-based [arbotix_ros](https://github.com/vanadiumlabs/arbotix_ros) but would like to transition to the new DynamixelSDK in CPP. I have been receiving the above error when trying to connect to the Arbotix-M, but no error when using the arbotix_ros package.

I have confirmed my baud rate and servo ID in dynaManager (one by one) and confirmed that servo works with arbotix_ros.

routiful, 2018-10-26:
Hi :)

It didn't supports the Arbotix-M controller and I havn't tested.
But if that controller may work like `USB2Dynamixel`, you can use this package.

Before you launch single_manager, please connect only ONE dynamixel to board.

piraka9011, 2018-11-03:
So I tried this package with different launch files/params (Version, Baud, name) but to no avail. It appears that the communication protocol this package uses isn't the same. I've been using serial communication based on the package [from Vanadium Labs](https://github.com/vanadiumlabs/arbotix_ros). It uses Python.

My servos are recognized by DynaManager tho. (But not my MX-12W, I contacted Dynamixel support which hasn't gotten back to me since last week).

I've refactored and modified Vanadium's controller so it can work on the project we're using [here](https://github.com/piraka9011/affordance_arm).

I have some sense as to how to add serial support for this package in C++ but unfortunately cannot commit the time.

routiful, 2018-11-13:
Sorry for inconvenience. 
But I am ready to new release :) Please feel free to open issue about any problem!

closed 2018-11-13 by routiful