ROBOTIS-GIT/dynamixel-workbench #193: REQUIRED process [position_control-2] has died!

juanjqo, 2018-10-26:
ISSUE TEMPLATE ver. 0.1.0

Before you open issue, please refer to [ROBOTIS e-Manual](http://emanual.robotis.com/)

1. Which distro of ROS have you installed? (ex, indigo, kinetic, melodic)
   ROS Melodic.

2. Which OS have you installed? (ex, Ubuntu 16.04 LTS (Xenial Xerus), Linux Mint 18.x, ...)
  Ubuntu 18.04 LTS

3. Which Dynamixel have you used? (ex, MX-28, XM430-W350, ...)
 AX12, MX-28, MX-64
 
4. Write down the commands you used in order

  ```

  $  roslaunch my_dynamixel_workbench_tutorial position_control.launch 

>  <launch>
 <arg name="device_name"                default="/dev/ttyACM0"/>
 <arg name="baud_rate"                  default="1000000"/>

 <arg name="scan_range"                 default="60"/>

 <arg name="profile_velocity"           default="200"/>
 <arg name="profile_acceleration"       default="20"/>

 <param name="device_name"              value="$(arg device_name)"/>
 <param name="baud_rate"                value="$(arg baud_rate)"/>

 <param name="scan_range"               value="$(arg scan_range)"/>

 <param name="profile_velocity"         value="$(arg profile_velocity)"/>
 <param name="profile_acceleration"     value="$(arg profile_acceleration)"/>

 <node name="position_control" pkg="dynamixel_workbench_controllers" type="position_control" required="true" output="screen"/>
 </launch>

  ```
 
5. Copy and Paste your error message on terminal
~$ roslaunch my_dynamixel_workbench_tutorial position_control.launch 
roslaunch my_dynamixel_workbench_tutorial position_control.launch 
... logging to /home/juanjqo/.ros/log/c6e2db48-d941-11e8-9cc7-f81654d52e97/roslaunch-pop-os-27898.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://pop-os:44621/

SUMMARY
========

PARAMETERS
 * /baud_rate: 1000000
 * /device_name: /dev/ttyACM0
 * /profile_acceleration: 20
 * /profile_velocity: 200
 * /rosdistro: melodic
 * /rosversion: 1.14.3
 * /scan_range: 60

NODES
  /
    position_control (dynamixel_workbench_controllers/position_control)

auto-starting new master
process[master]: started with pid [27908]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to c6e2db48-d941-11e8-9cc7-f81654d52e97
process[rosout-1]: started with pid [27921]
started core service [/rosout]
process[position_control-2]: started with pid [27924]
================================================================================REQUIRED process [position_control-2] has died!
process has died [pid 27924, exit code -11, cmd /opt/ros/melodic/lib/dynamixel_workbench_controllers/position_control __name:=position_control __log:=/home/juanjqo/.ros/log/c6e2db48-d941-11e8-9cc7-f81654d52e97/position_control-2.log].
log file: /home/juanjqo/.ros/log/c6e2db48-d941-11e8-9cc7-f81654d52e97/position_control-2*.log
Initiating shutdown!
================================================================================
[position_control-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
  
6. Please, describe detailedly what difficulty you are in 

    - I am using a Poppy humanoid robot, which is compose of 25 Dynamixel motors with the same baud rate 1000000 and with the ID's as follows
2 AX-12 [36, 37], 19 MX-28 [11, 12, 14, 15, 21, 22, 24, 25, 33, 34, 35, 41, 42, 43, 44, 51, 52, 53, 54], 4 MX-64 [13, 23, 31, 32].

When I set "scan_range" = 20, the launch file works. Of course, in that case, only five motors are started. 

Eg:

> roslaunch my_dynamixel_workbench_tutorial position_control.launch 

... logging to /home/juanjqo/.ros/log/6fc1f61a-d940-11e8-9cc7-f81654d52e97/roslaunch-pop-os-25825.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://pop-os:34751/

SUMMARY
========

PARAMETERS
 * /baud_rate: 1000000
 * /device_name: /dev/ttyACM0
 * /profile_acceleration: 20
 * /profile_velocity: 200
 * /rosdistro: melodic
 * /rosversion: 1.14.3
 * /scan_range: 20

NODES
  /
    position_control (dynamixel_workbench_controllers/position_control)

auto-starting new master
process[master]: started with pid [25839]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 6fc1f61a-d940-11e8-9cc7-f81654d52e97
process[rosout-1]: started with pid [25850]
started core service [/rosout]
process[position_control-2]: started with pid [25853]
-----------------------------------------------------------------------
  dynamixel_workbench controller; position control example             
-----------------------------------------------------------------------

MODEL   : MX-28
ID      : 11

MODEL   : MX-28
ID      : 12

MODEL   : MX-64
ID      : 13

MODEL   : MX-28
ID      : 14

MODEL   : MX-28
ID      : 15

-----------------------------------------------------------------------

However, when I set  "scan_range" = 60, 
 in order to start all the motors, the launch file crash.

routiful, 2018-10-30:
Dynamixel-Workbench is set to 16 dynamixel of same series. Because of memory problem in embedded board.

https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/676880b4900877e4d6b68f21b87b0153cf04fbe3/dynamixel_workbench_toolbox/include/dynamixel_workbench_toolbox/dynamixel_tool.h#L42

You are supposed to change it what you want.

gpwhd07, 2018-11-15:
This issue will be closed since there were no actions for a while. 
If you have more question, you can reopen this issue to show this issue to the users whenever. Thanks.

closed 2018-11-15 by gpwhd07

juanjqo, 2018-12-06:
Hi guys!
I modified the code line  DXLInfo dxl_info_[16];  to  DXLInfo dxl_info_[30];
But the problem continues, the launch files crash. 

================================================================================REQUIRED process [position_control-2] has died!
process has died [pid 10630, exit code -11, cmd /opt/ros/melodic/lib/dynamixel_workbench_controllers/position_control __name:=position_control __log:=/home/juanjqo/.ros/log/6bbacad8-f952-11e8-8c4e-3863bbbce9bc/position_control-2.log].
log file: /home/juanjqo/.ros/log/6bbacad8-f952-11e8-8c4e-3863bbbce9bc/position_control-2*.log
Initiating shutdown!
================================================================================

juanjqo, 2018-12-06:
In addition, when I set the "scan_range" to 31, the launch detects the motors and crashes. I realize that a motor is detected as NX-64. This is expected?   (since the motor is a MX-64).  

Thanks

-----------------------------------------------------------------------
  dynamixel_workbench controller; position control example             
-----------------------------------------------------------------------

MODEL   : MX-28
ID      : 11

MODEL   : MX-28
ID      : 12

MODEL   : MX-64
ID      : 13

MODEL   : MX-28
ID      : 14

MODEL   : MX-28
ID      : 21

MODEL   : MX-28
ID      : 22

MODEL   : MX-64
ID      : 23

MODEL   : MX-28
ID      : 24

MODEL   : NX-64
ID      : 31

-----------------------------------------------------------------------
================================================================================REQUIRED process [position_control-2] has died!
process has died [pid 11358, exit code -11, cmd /opt/ros/melodic/lib/dynamixel_workbench_controllers/position_control __name:=position_control __log:=/home/juanjqo/.ros/log/96888f10-f953-11e8-8c4e-3863bbbce9bc/position_control-2.log].
log file: /home/juanjqo/.ros/log/96888f10-f953-11e8-8c4e-3863bbbce9bc/position_control-2*.log
Initiating shutdown!
================================================================================