ROBOTIS-GIT/dynamixel-workbench #194: Synchronize Write Servo Error

auliakhilmi, 2018-10-27:
1. Which distro of ROS have you installed? Kinetic Kame

2. Which OS have you installed? Ubuntu 16.04.3 Xenial Xerus

3. Which Dynamixel have you used? (18 Servo) 8 for MX-28, 10 for MX-64

 
4. Write down the commands you used in order

  ```
  $ roscore    
  $ roslaunch motion motion_controller.launch
  ```
Here inside the motion_controller.launch
```
<launch>
  <arg name="use_robot_name"         default="eros"/>
  <arg name="device_name"            default="/dev/ttyUSB1"/>
  <arg name="baud_rate"              default="1000000"/>
  <arg name="protocol_version"       default="2.0"/>
  <arg name="joint_controller"       default="position_mode"/>

  <node pkg="motion" type="motion_control" name="motion_control" required="true" output="screen">
    <param name="robot_name"           value="$(arg use_robot_name)"/>
    <param name="device_name"          value="$(arg device_name)"/>
    <param name="baud_rate"            value="$(arg baud_rate)"/>
    <param name="protocol_version"     value="$(arg protocol_version)"/>
    <param name="joint_controller"     value="$(arg joint_controller)"/>
  </node>
</launch>
``` 

5. Copy and Paste your error message on terminal

  ```
  process[motion_control-1]: started with pid [3621]
[ERROR] [1540628713.458865538]: Not found Joints, Please check id and baud rate
================================================================================REQUIRED process [motion_control-1] has died!
process has finished cleanly
log file: /home/eros/.ros/log/bf04804a-d9c1-11e8-af17-94c69116c2e2/motion_control-1*.log
Initiating shutdown!
================================================================================
[motion_control-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
  ```
  
6. Please, describe detailedly what difficulty you are in 

Inside motion_control.cpp I call function 

`joint_controller_->syncWrite("Goal_Position", goal_position);` 

and joint_controller is 

`DynamixelWorkbench *joint_controller_;`

I modified the source code from https://github.com/ROBOTIS-GIT/open_manipulator/blob/master/open_manipulator_dynamixel_ctrl/src/dynamixel_controller.cpp

When I run syncWrite just for MX-28 (without MX-64), there is no error and the program run correctly. Also when I run syncWrite just for MX-64 (without MX-28), there is no error and the program run correctly. But when I run MX-64 and MX-28 together, this error happened.

routiful, 2018-10-30:
I can't help you how to make your code.
Can you give me more information about `motion_controller`?

auliakhilmi, 2018-10-30:
My code is nearly same with https://github.com/ROBOTIS-GIT/open_manipulator/blob/master/open_manipulator_dynamixel_ctrl/src/dynamixel_controller.cpp

I just modified the number of JOINT_NUM, we know that in open_manipulator the type of the servo that used in is same, but here I use 2 types (MX-64 and MX-28), and the error happened.

I'm not setting the velocity, but just set the goal_position for every servo.

routiful, 2018-10-31:
You are supposed to modify some lines for 8 motor.
For example, you can find that the vector was filled with only four motor.

https://github.com/ROBOTIS-GIT/open_manipulator/blob/8b045d98336fd71a02a6cbc8a31b4cb7bb2730b1/open_manipulator_dynamixel_ctrl/src/dynamixel_controller.cpp#L40

auliakhilmi, 2018-10-31:
Here my code:

```
#include "motion/motion_control.h"

using namespace dynamixel;

MotionControl::MotionControl()
    :node_handle_(""),
     priv_node_handle_("~")
{
    robot_name_   = priv_node_handle_.param<std::string>("robot_name", "eros");
    
    std::string device_name   = priv_node_handle_.param<std::string>("device_name", "/dev/ttyUSB0");
    uint32_t dxl_baud_rate    = priv_node_handle_.param<int>("baud_rate", 1000000);
    protocol_version_         = priv_node_handle_.param<float>("protocol_version", 2.0);

    joint_mode_   = priv_node_handle_.param<std::string>("joint_controller", "position_mode");

    joint_id_.push_back(priv_node_handle_.param<int>("joint1_id", 1)); //MX-28
    joint_id_.push_back(priv_node_handle_.param<int>("joint2_id", 2)); //MX-28
    joint_id_.push_back(priv_node_handle_.param<int>("joint3_id", 3)); //MX-28
    joint_id_.push_back(priv_node_handle_.param<int>("joint4_id", 4)); //MX-28
    joint_id_.push_back(priv_node_handle_.param<int>("joint5_id", 5)); //MX-28
    joint_id_.push_back(priv_node_handle_.param<int>("joint6_id", 6)); //MX-28
    joint_id_.push_back(priv_node_handle_.param<int>("joint7_id", 7)); //MX-28
    joint_id_.push_back(priv_node_handle_.param<int>("joint8_id", 8)); //MX-28
    joint_id_.push_back(priv_node_handle_.param<int>("joint9_id", 9)); //MX-64
    joint_id_.push_back(priv_node_handle_.param<int>("joint10_id", 10)); //MX-64
    joint_id_.push_back(priv_node_handle_.param<int>("joint11_id", 11)); //MX-64
    joint_id_.push_back(priv_node_handle_.param<int>("joint12_id", 12)); //MX-64
    joint_id_.push_back(priv_node_handle_.param<int>("joint13_id", 13)); //MX-64
    joint_id_.push_back(priv_node_handle_.param<int>("joint14_id", 14)); //MX-64
    joint_id_.push_back(priv_node_handle_.param<int>("joint15_id", 15)); //MX-64
    joint_id_.push_back(priv_node_handle_.param<int>("joint16_id", 16)); //MX-64
    joint_id_.push_back(priv_node_handle_.param<int>("joint17_id", 17)); //MX-64
    joint_id_.push_back(priv_node_handle_.param<int>("joint18_id", 18)); //MX-64

    joint_controller_   = new DynamixelWorkbench;
      
    joint_controller_->begin(device_name.c_str(), dxl_baud_rate);

    getDynamixelInst();

    initPublisher();

    ROS_INFO("motion_control [node] : Init OK!");
}

MotionControl::~MotionControl(){
    for (uint8_t num = 0; num < JOINT_NUM; num++)
    joint_controller_->itemWrite(joint_id_.at(num), "Torque_Enable", false);
    ros::shutdown();
}

void MotionControl::getDynamixelInst()
{
  uint16_t get_model_number;
  for (uint8_t index = 0; index < JOINT_NUM; index++)
  {

    if (joint_controller_->ping(joint_id_.at(index), &get_model_number) != true)
    {
      ROS_ERROR("Not found Joints, Please check id and baud rate");

      ros::shutdown();
      return;
    }
  }

  setOperatingMode();
  setSyncFunction();
}

void MotionControl::setOperatingMode()
{
  if (joint_mode_ == "position_mode")
  {
    for (uint8_t num = 0; num < JOINT_NUM; num++)
      joint_controller_->jointMode(joint_id_.at(num));
  }
  else if (joint_mode_ == "current_mode")
  {
    for (uint8_t num = 0; num < JOINT_NUM; num++)
      joint_controller_->currentMode(joint_id_.at(num));
  }
  else
  {
    for (uint8_t num = 0; num < JOINT_NUM; num++)
      joint_controller_->jointMode(joint_id_.at(num));
  }
}

void MotionControl::setSyncFunction()
{
  joint_controller_->addSyncWrite("Goal_Position");

  if (protocol_version_ == 2.0)
  {
    joint_controller_->addSyncRead("Present_Position");
    joint_controller_->addSyncRead("Present_Velocity");
  }
}

void MotionControl::initPublisher(){
    joint_states_pub_ = node_handle_.advertise<sensor_msgs::JointState>(robot_name_ + "/goal_joint_position", 10);
}

void MotionControl::basicMotion(){
    goal_position[0]=GoalPosition[0];
    goal_position[1]=GoalPosition[2];
    goal_position[2]=GoalPosition[4];
    //goal_position[3]=GoalPosition[14];
    ROS_INFO("[%d][%d][%d]",goal_position[0],goal_position[1],goal_position[2]);
    sendAllServo();
}
void MotionControl::sendAllServo(){
    joint_controller_->syncWrite("Goal_Position", goal_position);
}

int main(int argc,char** argv)
{
    ros::init(argc, argv, "motion_control");
    MotionControl motion_control;
    ros::Rate loop_rate(ITERATION_FREQUENCY);
    while(ros::ok()){
      motion_control.basicMotion();
      loop_rate.sleep();
      ros::spinOnce();
    }
}
```
Variable `GoalPosition` is defined as GoalPosition[18]
and `goal_position` is defined as goal_position[JOINT_NUM]
When i used JOINT_NUM 8 with using MX-28 and id from 1-8, it work properly, servo move to the right position.
When i used JOINT_NUM 10 with using MX-64 and id from 9-18, it work properly also, the servo moved,,
but when i used JOINT_NUM 18 with using MX-28 and MX-64 id from 1-18, it stopped working, the error shows like what i wrote above.

I also tried just use JOINT_NUM 10 with using MX-28 id 1-8 and MX-64 id 9-10, it stopped working.
and my oppinion is syncwrite can't did well in more than one type of servo.

closed 2018-12-12 by gpwhd07