ROBOTIS-GIT/dynamixel-workbench #195: Potential buffer overflow or... i'm just stupid.

Le0nX, 2018-10-29:
On this line we take `get_id[id_cnt]` as an array.
https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/676880b4900877e4d6b68f21b87b0153cf04fbe3/dynamixel_workbench_toolbox/src/dynamixel_workbench_toolbox/dynamixel_driver.cpp#L194

 But on this line   we pass `dxl_id_` to method which is just a `uint8_t` variable... https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/676880b4900877e4d6b68f21b87b0153cf04fbe3/dynamixel_workbench_single_manager/src/single_dynamixel_monitor.cpp#L70 
In case when we have a lot of servos, we will get **accessing array out of bounds** because our "array" is just 1 byte long.

routiful, 2018-10-30:
Hello :)

Dynamixel id can be set 0 to 253. So I set that variable `uint8_t`. How many servos you want to use?

Le0nX, 2018-10-30:
Hi! 
Yeah, I understand that there is SINGLE monitor class & it should be used with only one servo, BUT this class contains code from `DynamixelDriver` class which IS USED for accessing array of servos. So... I think that you should have your own version of scan() method in `SingleDynamixelMonitor` class to avoid this kind of mistake, when you iterating `uint8_t` variable instead of an array.

What I want to say is, imagine a situation:
0) I'm dumb user & don't know anything about your code and `SingleDynamixelMonitor`
1) I just take and connect 30 servos to u2d2 and then plug in it to my PC.
2) Then I run `roslaunch my_dynamixel_workbench_tutorial single_manager.launch`
3) What should happen? Your code will create an object of `SingleDynamixelMonitor` with `uint8_t dxl_id_` then it will call a method of DynamixelDriver object `scan(uint8_t *get_id, uint8_t *get_id_num, uint8_t range)` and pass address of `dxl_id_` into it.
4) In this method...we will run in for cycle and scan for available servos AND iterate our 1byte long "array" aka `uint8_t dxl_id_`. So... if i have 30 servos connected we will overrflow out `uint8_t dxl_id_` by 29 bytes because of this code: 
```
 if (packetHandler_1->ping(portHandler_, id, &model_number) == COMM_SUCCESS)
    {
      get_id[id_cnt] = id;
      setTools(model_number, id);
      id_cnt++;
      protocol_version = 1.0;
    }
```

Le0nX, 2018-10-30:
I can write a patch and then create a pull request if you want. I would be happy to contribute to your project.

routiful, 2018-10-30:
I understand what's your purpose. And thank you for your idea.

But after it is not overflowed, this package only provides UI for just SINGLE dynamixel.
If it changed get multiple dynamixels, I might to make new UI for handle these. But I don't have plan to support it.

I can show you an example what you needed. It is working on linux and mac (NOT ROS)

https://github.com/ROBOTIS-GIT/dynamixel-workbench/tree/develop/dynamixel_workbench_toolbox

I will change single manager based on that.
If you want to contribute it, please feel free to create PR.