ROBOTIS-GIT/dynamixel-workbench #197: Issue with Changing PID parameters of the Dynamixel through R+ Manager

TheBard2273, 2018-10-30:
Hi,

I'm using the Turtlebot3 Burger setting on a new design and I want to change P, I and D gains of my dynamixels through R+ manager 2.0.
R+ manager detects the motors and I will change P from 800 to 3000, D from 0 to 250 and I from 0 to 50, but after downloading the turtlebot3_core sketch on the OpenCR board, the changes will cancel and R+ manager shows the same gain (800,0,0) for both motors. is there something I'm missing?
Should I do anything after changing the numbers before downloading the turtlebot3_core on OpenCR and reconnecting it to the RPi ?

Thanks.

routiful, 2018-10-31:
Hi :)

Address of P,I and D gain is saved in RAM of Dynamixel. So the value of this address will be erased after power off.

You are supposed to set it in turtlebot3_core.

https://github.com/ROBOTIS-GIT/OpenCR/blob/9da4ae9fd4506df6e068c2a6d381f268a0ba371a/arduino/opencr_arduino/opencr/libraries/turtlebot3/src/turtlebot3/turtlebot3_motor_driver.cpp#L36

Please add below code on above init() function after setTorque(true).

```
// Set P gain
packetHandler_->write2ByteTxRx(portHandler_, DXL_LEFT_ID, 84, 3000, &dxl_error); 
// Set D gain
packetHandler_->write2ByteTxRx(portHandler_, DXL_LEFT_ID, 80, 250, &dxl_error);
```

You can find control address to follow below link

http://emanual.robotis.com/docs/en/dxl/x/xm430-w210/#control-table-of-ram-area

ashotaro, 2018-11-05:
Hi,  
I also try to solve same problem in [open-manipulator](https://github.com/ROBOTIS-GIT/open_manipulator).  
  
How can I change the Gains? Let me know

ashotaro, 2018-11-08:
I solved it by "itemWrite" which is the function in dynamixel_workbench_toolbox.
I execute it when a dynamixel is connected.
Thx

routiful, 2018-11-13:
Great :) If you have any futher issue, please feel free to open issue.

closed 2018-11-13 by routiful