ROBOTIS-GIT/dynamixel-workbench #198: Dynamixel-workbench error with the seventeenth motor

keiser30, 2018-11-09:
I modify the dynamixel-workbench monitor controller and it worked correctly for 16 connected MX-64 motors. But, when I connect the 17th motor I get an extraneous error that I can not solve.

I am currently working with ROS Kinetic on Ubuntu 16.04 and Dynamixel MX-64 with protocol 2.0

the commands that I am using are:
  $ sudo chmod 777 /dev/ttyUSB0
  $ roslaunch marta marta_monitor.launch
 

The error message on terminal

  ```
  SUMMARY
  ========

  PARAMETERS
   * /baud_rate: 57600
   * /device_name: /dev/ttyUSB0
   * /rosdistro: kinetic
   * /rosversion: 1.12.14
   * /scan_range: 200

  NODES
    /
      marta_dynamixel_monitor (marta/marta_dynamixel_monitor)

  auto-starting new master
  process[master]: started with pid [3697]
  ROS_MASTER_URI=http://localhost:11311

  setting /run_id to 98b9f23e-e34f-11e8-bc56-e4f89c6d29af
  process[rosout-1]: started with pid [3710]
  started core service [/rosout]
  process[single_dynamixel_monitor-2]: started with pid [3718]
  [ID] 3, [Model Name] MX-64-2, [BAUD RATE] 57600 [VERSION] 2.0
  [ID] 7, [Model Name] MX-64-2, [BAUD RATE] 57600 [VERSION] 2.0
  [ID] 8, [Model Name] MX-64-2, [BAUD RATE] 57600 [VERSION] 2.0
  [ID] 9, [Model Name] MX-64-2, [BAUD RATE] 57600 [VERSION] 2.0
  [ID] 10, [Model Name] MX-64-2, [BAUD RATE] 57600 [VERSION] 2.0
  [ID] 11, [Model Name] MX-64-2, [BAUD RATE] 57600 [VERSION] 2.0
  [ID] 12, [Model Name] MX-64-2, [BAUD RATE] 57600 [VERSION] 2.0
  [ID] 13, [Model Name] MX-64-2, [BAUD RATE] 57600 [VERSION] 2.0
  [ID] 14, [Model Name] MX-64-2, [BAUD RATE] 57600 [VERSION] 2.0
  [ID] 15, [Model Name] MX-64-2, [BAUD RATE] 57600 [VERSION] 2.0
  [ID] 16, [Model Name] MX-64-2, [BAUD RATE] 57600 [VERSION] 2.0
  [ID] 17, [Model Name] MX-64-2, [BAUD RATE] 57600 [VERSION] 2.0
  [ID] 18, [Model Name] MX-64-2, [BAUD RATE] 57600 [VERSION] 2.0
  [ID] 19, [Model Name] ??&B, [BAUD RATE] 57600 [VERSION] 2.0
  [ID] 21, [Model Name] MX-64-2, [BAUD RATE] 57600 [VERSION] 2.0 
  [ID] 22, [Model Name] MX-64-2, [BAUD RATE] 57600 [VERSION] 2.0 
  [ID] 23, [Model Name] MX-64-2, [BAUD RATE] 57600 [VERSION] 2.0 
  [ID] 24, [Model Name] MX-64-2, [BAUD RATE] 57600 [VERSION] 2.0 
  [ID] 25, [Model Name] MX-64-2, [BAUD RATE] 57600 [VERSION] 2.0 
ES MI CODIGO! 23 3
marta_dynamixel_monitor : Init Success!
================================================================================REQUIRED process [marta_dynamixel_monitor-2] has died!
  process has died [pid 3718, exit code -11, cmd /home/marta2/catkin_ws/devel/lib/marta_dynamixel_monitor __name:=marta_dynamixel_monitor __log:=/home/marta2/.ros/log/98b9f23e-e34f-11e8-bc56-e4f89c6d29af/marta_dynamixel_monitor-2.log].
  log file: /home/marta2/.ros/log/98b9f23e-e34f-118e-bc56-e4f89c6d29af/marta_dynamixel_monitor-2*.log
  Initiating shutdown!
  ================================================================================
  [marta_dynamixel_monitor-2] killing on exit
  [rosout-1] killing on exit
  [master] killing on exit
  shutting down processing monitor...
  ... shutting down processing monitor complete
  ```
It can be seen that the 17th motor is the one that is giving problem, it appears as if it does not recognize the model of that motor, but the others do. Then I do not know what has happened.

routiful, 2018-11-13:
Sorry for late reply.
I restricted the number of Dynamixel as 16

https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/676880b4900877e4d6b68f21b87b0153cf04fbe3/dynamixel_workbench_toolbox/include/dynamixel_workbench_toolbox/dynamixel_tool.h#L42

due to memory problem on embedded board.

You are supposed to change that.

KurtE, 2018-11-22:
@routiful,

I believe that some of the changes we did for saving memory, that I changed the code to detect when a tool was filled with 16 servos, to not over write memory AND in addition, if there were still available tools (< 5 used), it would allocate another tool and add that servo to the new tool object... 

However it looks like you have been busy making changes here and I have not checked to see if that code is still here or not. 

Please forgive my random thought here, but have been wondering if it might make sense to  turn the the Tools/Items objects on it's head a little to maybe remove some of these restrictions.  

That is Wonder if it would make sense to make all of the Tool information be const (Program space).  We already did that with the information for ControlItem objects and it looks like you have now done the same for ModelInfo.   So wondering if all of that information should be put into Another const table for all of the information associated with a Servo type.  Then wonder if you simply have a table that maps Servo ID to Servo Type index,  if that would make sense.  
You could easily do it two ways: 
a) Simple byte array where index is Servo ID and content is the Servo type (maybe 0xff says not there).
That would eat up maybe 255 bytes... 

b) Or could restrict to lets say no more than 64 servos and have array of ServoID, ServoType structures...  So in this case 128 bytes total... 

There are pros and cons of both... a) You can easily convert Servo to type info without scanning...  b) you can walk through all servo quickly without looking through all blanks.... Note: some of that would be minimized by remembering the min/max servo number used... 

Again sorry for being a bit random here...

routiful, 2018-11-23:
@KurtE 

Hi!, First of all I have developed to upgrade Dynamixel-Workbench since last two weeks based on your contribution. It is really awesome and I will let you know this update and to ask review my code.(But you already knew about this haha)

I sincerely thankful your contribution and I am ready to carefully listen your review.
But I didn't understand perfectly your comments, so I give to you some questions.

> That is Wonder if it would make sense to make all of the Tool information be const (Program space)

Yes, we did :)

> So wondering if all of that information should be put into Another const table for all of the information associated with a Servo type.

It means to combine ControlItem and ModelInfo struct is more resonable and it makes more quickly find servo and reduce memory?

KurtE, 2018-11-23:
Hi @routiful,

I have not thought this completely out, ut the idea is suppose you create a structure like the DynamixelTool object, but remove the dxl_id_ and dxl_cnt_ fields.   Actually you could simply extend the information in the ControlItem table...  

Then suppose you create all of this as const data (program space) and suppose that you have an array of these objects or array of pointers to these objects for each of your servo types

Example assume your ModelInfo structure is extended to hold pointer to array of ControlItem and count and ID and ... 
Then you could have an array like: 

```
static const ModelInfo *model_infos[] = 
    {&info_AX, &info_RX, &info_EX...};
```

Then as I mentioned in previous one, you have array like:
```
uint8_t servo_id_to_model_info[254];
```
where maybe you init the array to 0xff;  

So if you wish to then map a servo ID to the tool info(model_info), you simply need to look at the array and then grab the data out of the array mentioned... 

Or again if you did not wish to use 254 or so bytes for servo indexes.  you could create a simple structure 
(or array), like:
```
typedef struct {
    uint8_t servoID;
    uint8_t modelIndex;
} IdToModel;

#define MAX_SERVOS 32  // Choose your max 64?
IdToModel servo_id_to_model_info[MAX_SERVOS];
uint8_t count_servos = 0;
```

But I understand this would impact the code in lots of places...  

Hope that makes sense?

routiful, 2018-11-29:
Sorry for late reply.
I understand your suggestions. But all servo doesn't need to get their own model_info.
Tool including model_info is more reasonable.

KurtE, 2018-11-29:
@routiful,

Not a problem,  I was simply trying to give you a possible alternative, which would maybe remove some of the limitations, like only 5 tools and only 16 items per the tool, without wasting much memory.

Note: for the fun of it, I thought I would try migrating your current stuff onto OpenCR and maybe run some of the memory usage stuff, that was showing me why we were running out of memory on OpenCM...

But it looks like you made sufficient changes to your current stuff, like how in the Monitor program you do the "info" command.    So maybe I will wait until you might resolve, that function names changed...

Also wondering if the stuff in your p_Monitor.cpp will work on the Arduino boards. 
That is if you look at the start of it:
```
        if (strcmp(cmd, "info") == 0)
        {
          uint8_t id = atoi(param[0]);

          const ControlItem *control_item =  dxl_wb.getControlTable(id);
          uint8_t the_number_of_control_item = dxl_wb.getTheNumberOfControlItem(id);

          uint16_t last_register_addr = control_item[the_number_of_control_item-1].address;
          uint16_t last_register_addr_length = control_item[the_number_of_control_item-1].data_length;

          uint32_t getAllRegisteredData[last_register_addr+last_register_addr_length];
...
```
If I am reading this correct the variable getAllRegisteredData is on the stack and is not a constant size.  Not sure if GCC supports this?  Maybe.   But: assuming I understand this properly.   If I tried this on a OpenCM/CR board with a Dynamixel Pro servo, whose last servo register number in the table is 892 size of 1
The Array would be 893 elements long of which each is 4 bytes so:  3572 bytes on stack... Which on a board like OpenCM with 20K RAM, could be an issue...   Especially since it will then call Dynamixel SDK expecting to then read in all 893 bytes...      Actually looking at it, wondering if correct that is the variable getAllRegisteredData should be uint8_t... as you then expect to build return values out of it and your indexes will be off?

routiful, 2018-12-02:
I will check it your point out. I agree getAllRegisteredData is enough to uint8_t. I will test it.
By the way, I changed "info" command to "control_table" and merged it into this branch and OpenCR, OpenCM. 
If you have a time, please test it and feel free to feedback :)

KurtE, 2018-12-02:
Hi again @routiful,

I have synced up my dev branch and so far it appears to be working.   I have not tried very much yet.

I was curious about memory usage, so I did a quick update of your p_monitor app, to hopefully verify we were not still trashing memory.  Looks like some of our memory reductions have helped. 

Here is the OpenCM version with memory usages guessing: 
[p_Monitor-memory-181202a.zip](https://github.com/ROBOTIS-GIT/dynamixel-workbench/files/2637025/p_Monitor-memory-181202a.zip)

Luckily I don't have any Pro servos to test it with, I did it with an XL430-W250 and at least at first glance the data looks valid:
```
Estimated global data size: 10104
starting Heap info: start: 20002778 current: 20002cb4
Start Stack info: end: 20005000 current: 20004fe0
Try to init memory
After init memory
-------------------------------------
Set begin before scan or ping
-------------------------------------
help
begin  (BAUD)
scan   (RANGE)
ping   (ID)
control_table (ID)
id     (ID) (NEW_ID)
baud   (ID) (NEW_BAUD)
torque_on (ID)
torque_off (ID)
joint  (ID) (GOAL_POSITION)
wheel  (ID) (GOAL_VELOCITY)
write  (ID) (ADDRESS_NAME) (DATA)
read   (ID) (ADDRESS_NAME)
sync_write_handler (Ref_ID) (ADDRESS_NAME)
sync_write (ID_1) (ID_2) (HANDLER_INDEX) (PARAM_1) (PARAM_2)
sync_read_handler (Ref_ID) (ADDRESS_NAME)
sync_read (ID_1) (ID_2) (HANDLER_INDEX)
bulk_write_handler
bulk_write_param (ID) (ADDRESS_NAME) (PARAM)
bulk_write
bulk_read_handler
bulk_read_param (ID) (ADDRESS_NAME)
bulk_read
reboot (ID) 
reset  (ID) 
end
-------------------------------------
Press Enter Key
Heap ptr: 20002cb4  Usage: 1340
Stack Max: 20004c10, usage: 1008
Estimated unused memory: 8028
[CMD] : begin 1000000

Succeed to init : 1000000
Heap ptr: 20002dd4  Usage: 1628
Stack Max: 20004bd8, usage: 1064
Estimated unused memory: 7684
[CMD] : scan

Find 1 Dynamixels
id : 1 model name : XL430-W250
Heap ptr: 20002dd4  Usage: 1628
Stack Max: 20004698, usage: 2408
Estimated unused memory: 6340
[CMD] : control_table 1

	Model_Number : 36
	Firmware_Version : 42
	ID : 1
	Baud_Rate : 3
	Return_Delay_Time : 250
	Drive_Mode : 0
	Operating_Mode : 3
	Secondary_ID : 255
	Protocol_Version : 2
	Homing_Offset : 0
	Moving_Threshold : 10
	Temperature_Limit : 72
	Max_Voltage_Limit : 140
	Min_Voltage_Limit : 60
	PWM_Limit : 117
	Acceleration_Limit : 255
	Velocity_Limit : 9
	Max_Position_Limit : 255
	Min_Position_Limit : 0
	Shutdown : 52
	Torque_Enable : 0
	LED : 0
	Status_Return_Level : 2
	Registered_Instruction : 0
	Hardware_Error_Status : 0
	Velocity_I_Gain : 232
	Velocity_P_Gain : 100
	Position_D_Gain : 16
	Position_I_Gain : 0
	Position_P_Gain : 128
	Feedforward_2nd_Gain : 0
	Feedforward_1st_Gain : 0
	Bus_Watchdog : 0
	Goal_PWM : 117
	Goal_Velocity : 9
	Profile_Acceleration : 0
	Profile_Velocity : 0
	Goal_Position : 220
	Realtime_Tick : 22
	Moving : 0
	Moving_Status : 1
	Present_PWM : 0
	Present_Load : 0
	Present_Velocity : 0
	Present_Position : 220
	Velocity_Trajectory : 0
	Position_Trajectory : 220
	Present_Input_Voltage : 121
	Present_Temperature : 32
Heap ptr: 200035dc  Usage: 3684
Stack Max: 20004400, usage: 3072
Estimated unused memory: 3620
```
Note: I was at a loss to figure out how some of this code actually worked.  That is with the code:
```
      if (cmd[0] == "control_table")
      {
        uint8_t id = cmd[1].toInt();

        const ControlItem *control_item =  dxl_wb.getControlTable(id);
        uint8_t the_number_of_control_item = dxl_wb.getTheNumberOfControlItem(id);

        uint16_t last_register_addr = control_item[the_number_of_control_item-1].address;
        uint16_t last_register_addr_length = control_item[the_number_of_control_item-1].data_length;

        uint32_t getAllRegisteredData[last_register_addr+last_register_addr_length];

        if (control_item != NULL)
        {
          result = dxl_wb.readRegister(id, (uint16_t)0, last_register_addr+last_register_addr_length, getAllRegisteredData, &log);
```
How that read would work as you are addressing the array with normal indexes, which  are normally at byte boundaries, but your array is 4 bytes per element... But the data is correct...  

So for XL430 servos, you try to read in I believe 147 bytes so allocated: 588 bytes here 

So then tracked the code to:  DynamixelDriver::readRegister(uint8_t id, uint16_t address, uint16_t length, uint32_t *data, const char **log)

And then found you allocate another new stack variable:  uint8_t data_read[length];
So this one is only 147 bytes.   Which you call off to Dynamixel SDK to do the read. 
And then after the read completes, you then move the actual read bytes into your uint32_t array, which your sketch extracts the bytes out of the uint32_t array... 

So it appears to work...  Personally I find this sort of overloading of this API, a bit odd, but...

Example suppose I call it like:
uint32_t my_array[10];
readRegister(1, 0, 10, my_array);

Then the 10 elements in the array are filled with one byte of data for each quad word...
likewise if I called it with: readRegister(1, 0, 3, my_array);
The first three elements of the array will be filled in with byte elements. 

But if I call it with: readRegister(1, 0, 4, my_array);
Only the first element will be filled in with a DWORD of data...  (likewise for 2)

And for memory usage it would be interesting to try it out with some of the Pro servos, who would try to read in 893 elements  Just those two extra arrays would take up something like 3730 extra bytes on the stack.  Which is more than I estimated is still available...

routiful, 2018-12-19:
@KurtE 
Origianl problem @keiser30 commented is not proceeded.
I will open new issue for your comment.

@keiser30 If you have any futher problem or question, please re-open this issue page.

closed 2018-12-19 by routiful