ROBOTIS-GIT/dynamixel-workbench #199: ...Failed to find dynamixel!

rothti, 2018-11-12:
Hello,
I'm trying to do the "dynamixel workbench" tutorial but when I do the command:
"$ roslaunch my_dynamixel_workbench_tutorial single_manager.launch"
I keep receiving the fact that it "failed to find Dynamixel".
The complete error message is joined as an image.
[image]
I am using ros-kinetic on ubuntu 16.04 and my hardware is an Opencm9.04c microcontroller, a XL-320 servo and a LN-101 downloader.
Similar topics exist but they are closed and I couldn't a find a solution who will work.

Already thank you for your help and time,

Timo

routiful, 2018-11-13:
Hi :)

First, upload firmware to opencm9.04 to work as usb2dynamixel
Second, check usb port and give permission to that
Third, check dynamixel firmware, id, baudrate

rothti, 2018-11-13:
Thank you for your answer :)
I have already worked on the usb port, it seems ok.
It might be the firmware I'm struggling with. I can't find the right one to upload.

routiful, 2018-11-14:
Sorry for inconvenience.

First, you are supposed to install R+ manager 2.0  latest version and upgrade 
http://en.robotis.com/service/downloadpage.php?ca_id=10

Second, firmware recovery to OpenCM and set baudrate same as Dynamixel

Then you can use it usb2dynamixel.

closed 2018-12-12 by gpwhd07