ROBOTIS-GIT/dynamixel-workbench #203: How to install ros-kinetic-dynamixel-workbench on Raspbian Stretch

kasliwalr, 2018-11-27:
Hi, 

Here is my system description:
ROS distro: ROS kinetic
OS: Raspbian Stretch 4.14.79
Dynamixel: XL430-250-T

I want to install the _ros-kinetic-dynamixel-workbench_ on raspberry Pi running Raspbian Stretch. 
I did the ros-kinetic installation following the process specified [here](http://wiki.ros.org/action/fullsearch/ROSberryPi/Installing%20ROS%20Kinetic%20on%20the%20Raspberry%20Piaction=fullsearch&context=180&value=linkto%3A%22ROSberryPi%2FInstalling+ROS+Kinetic+on+the+Raspberry+Pi%22)

So I tried the following, but it didn't work
```
> rosinstall_generator ros_comm dynamixel-workbench --rosdistro kinetic --deps --wet-only --tar>kinetic-custom_ros.rosinstall
The following not released packages/stacks will be ignored: dynamixel-workbench   # error message
```
I also tried apt-get install, but the package does not show up in ros-repos
```
> sudo apt-get install ros-kinetic-dynamixel-workbench
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-kinetic-dynamixel-workbench
```

closed 2018-11-28 by kasliwalr

routiful, 2018-11-28:
Hi, did you solve this issue?

kasliwalr, 2018-11-28:
Yes, I did.  I downloaded the dynamixel_workbench, analyzed the package
file for dependencies, installed dependencies and then built
dynamixel_workbench using catkin_make. I am now able to run the
dynamixel_workbench examples on Raspberry Pi. So it seems to be working.

thanks,
rishabh

routiful, 2018-11-28:
Great Thanks!