ROBOTIS-GIT/dynamixel-workbench #204: Dynamixel node crashes

Volgnor, 2018-11-29:
Hi, 

I'm encountering a problem using a Dynamixel **XM540-W270-R**. I have **ROS Kinetic** on **Ubuntu 16.04**. 

I am trying to use the actuator to move a control surface but the node dies when starting external equipment that moves the structure where the actuator is placed. I have tried the Single Monitor that is available on the workbench_toolbox to check if it is a problem with the node I have written but it dies in the same way.

Are you guys aware of problems due to electromagnetic interference with the Dynamixel actuators?

Thanks

routiful, 2018-11-30:
Did you check your ID and Baudrate?

Volgnor, 2018-11-30:
Hi, yes. I get the connection with the actuator but the node dies when the external equipment is powered. 

How are communications failures handled? It seems like the underlying libraries (Dynamixel SDK) fail as soon as there is a small communications issue, which kills the node.

routiful, 2018-12-13:
Sorry for late reply.
Can you please try to new version? 
It might be more helpful to debug.
http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/