ROBOTIS-GIT/dynamixel-workbench #206: Compatibility (openCM/openCR) between released and develop branches.

KurtE, 2018-12-06:
Again sorry for not following the standard template...

Wondering about if compatibility between the version 1 of the dynamixel-workbench library and the version 2 is important or not. 

Example if I currently load up the example program: I_Position_Speed from the current release version of Arduino device OpenCR and compiles, but then if I switch to the develop branch, and try to build it, there are compiler errors, like: 
```
C:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\hardware\OpenCR\1.3.2\libraries\OpenCR\examples\08. DynamixelWorkbench\l_Position_Speed\l_Position_Speed.ino: In function 'void loop()':

l_Position_Speed:45:32: error: call of overloaded 'goalPosition(int, int)' is ambiguous

   dxl_wb.goalPosition(DXL_ID, 0);

                                ^
...
```
Obviously you can and probably have updated these examples in the develop branch, but I would suspect that there are several applications out there that probably used the older syntax.

routiful, 2018-12-12:
I uploaded new example for new version.
You can check h_Position.ino

KurtE, 2018-12-12:
Hi @routiful, 

Thanks, I will try it out.

But what I am wondering is, is there any requirement (or goal), that applications that were developed using Version 1 of the library will compile unchanged and work with Version 2 of the library.   

Note: This has no impact on me, as I all of my personal code uses the Dynamixel SDK, but have dealt with similar issues on other platforms/libraries. 

Obviously you have a many options here. 
a) You can simply let the developers find out about the changes using the compiler
b) Have information up on the wiki or the like on changes and what is needed to be changed
c) When it makes sense, add back in APIs that were changed, in many cases these might be simple wrapper of the new method(s), and maybe added in compiler messages...

Again, I mainly opened the issue, when I tried compiling some older code and it failed, and thought maybe it would be a good idea to raise it, so you might choose what makes sense.  

Feel free to close it

routiful, 2018-12-13:
I didn't consider the compatible with version 1. This is my fault.
I will choose options you give to me and apply as soon as possible.

Thanks!

closed 2018-12-13 by routiful