ROBOTIS-GIT/dynamixel-workbench #213: dynamixel_workbench_toolbox/dynamixel_driver.cpp:391  error: ‘class dynamixel::PacketHandler’ has no member named ‘clearMultiTurn’

happythingsmaker, 2018-12-28:
ISSUE TEMPLATE ver. 1.0.0

Before you open issue, please refer to [ROBOTIS e-Manual](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/)

1. How to setup? (ex, U2D2, OpenCR,...)
I tried this for my PC (Ubuntu 16.04 x64)
http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/#position-controller

I've struggled to build "dynamixel_workbench" but I couldn't make it.
It was neither ROS nor make. 
I just reverted the commit (5cbf5e82f6c87b997b6789ded83bde76d88e2fc2) and build it for a temporary build and it worked.

2. Which Dynamixel have you used? and how many? **(Please describe below format to all connected Dynamixels)**
0
 
3. Write down the commands you used in order
$ mkdir ~/issue
$ cd ~/issue
$ git clone https://github.com/ROBOTIS-GIT/dynamixel-workbench.git
$ git clone https://github.com/ROBOTIS-GIT/dynamixel-workbench-msgs.git
$ git clone https://github.com/ROBOTIS-GIT/DynamixelSDK.git
$ git clone https://github.com/ROBOTIS-GIT/open_manipulator_msgs.git
$ sudo apt-get install ros-kinetic-moveit-core ros-kinetic-moveit-ros-planning ros-kinetic-moveit-ros-planning-interface

$ cd ~/issue/dynamixel-workbench/dynamixel_workbench_toolbox/examples
$ mkdir -p build && cd build
$ cmake ..
$ make

 
4. Copy and Paste your error message on terminal
[  2%] Building CXX object CMakeFiles/dynamixel_workbench.dir/home/chans/issue/dynamixel-workbench/dynamixel_workbench_toolbox/src/dynamixel_workbench_toolbox/dynamixel_item.cpp.o
[  5%] Building CXX object CMakeFiles/dynamixel_workbench.dir/home/chans/issue/dynamixel-workbench/dynamixel_workbench_toolbox/src/dynamixel_workbench_toolbox/dynamixel_driver.cpp.o
/home/chans/issue/dynamixel-workbench/dynamixel_workbench_toolbox/src/dynamixel_workbench_toolbox/dynamixel_driver.cpp: In member function ‘bool DynamixelDriver::clearMultiTurn(uint8_t, const char**)’:
/home/chans/issue/dynamixel-workbench/dynamixel_workbench_toolbox/src/dynamixel_workbench_toolbox/dynamixel_driver.cpp:391:47: error: ‘class dynamixel::PacketHandler’ has no member named ‘clearMultiTurn’
   sdk_error.dxl_comm_result = packetHandler_->clearMultiTurn(portHandler_, 
                                               ^
CMakeFiles/dynamixel_workbench.dir/build.make:86: recipe for target 'CMakeFiles/dynamixel_workbench.dir/home/chans/issue/dynamixel-workbench/dynamixel_workbench_toolbox/src/dynamixel_workbench_toolbox/dynamixel_driver.cpp.o' failed
make[2]: *** [CMakeFiles/dynamixel_workbench.dir/home/chans/issue/dynamixel-workbench/dynamixel_workbench_toolbox/src/dynamixel_workbench_toolbox/dynamixel_driver.cpp.o] Error 1
CMakeFiles/Makefile2:511: recipe for target 'CMakeFiles/dynamixel_workbench.dir/all' failed
make[1]: *** [CMakeFiles/dynamixel_workbench.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

5. Please, describe detailedly what difficulty you are in 
build failed.

P.S.
I tried to revert the commit "5cbf5e82f6c87b997b6789ded83bde76d88e2fc2"

$ git revert 5cbf5e82f6c87b997b6789ded83bde76d88e2fc2

and it worked well.

============================================
commit 5cbf5e82f6c87b997b6789ded83bde76d88e2fc2
Author: Darby Lim <thlim@robotis.com>
Date:   Tue Dec 11 10:42:33 2018 +0900

    add clearmultiturn
============================================

P.S. I love ROBOTIS!

routiful, 2019-01-02:
Hi, @happythingsmaker :)

I appereciated your inquiry!
Last month, SDK was updated.
You should re-download and re-install SDK first.
Please follow below link and try latest version of workbench.
http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/library_setup/cpp_linux/#cpp-linux

happythingsmaker, 2019-01-04:
Hello, Mr. Taehun Lim. Thanks for your reply.
okay, I will try it again.
It is somehow confusing because I thought that I've never installed the SDK before.
For the clean test, I am going to test it using a new R-PI and share the result to you.
:)

closed 2019-02-27 by gpwhd07