ROBOTIS-GIT/dynamixel-workbench #214: skipping ID number 0 issue when the searching dynamixel

happythingsmaker, 2018-12-30:
https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/851a13b0485ee54b157196e7a1094d4f497c6cc4/dynamixel_workbench_toolbox/src/dynamixel_workbench_toolbox/dynamixel_driver.cpp#L331

I have tried to find my robot's motors several times using 
$ rosrun dynamixel_workbench_controllers find_dynamixel /dev/ttyUSB0

and I found that "find_dynamixel" always could not find ID number 0. 
I thought that it was from a cable fault or so. But it wasn't.
It seemed like starting from ID number 1.

this line should be changed in this code so that the code can start to find from a motor whose id is ID number 0 
- return scan(get_id, get_the_number_of_id, 1, range, log);
+ return scan(get_id, get_the_number_of_id, 0, range, log);

Thanks ahead of time.

routiful, 2019-01-02:
Thank you for your contribution.

I've updated your comment to master branch

https://github.com/ROBOTIS-GIT/dynamixel-workbench/pull/215

happythingsmaker, 2019-01-04:
I'm feeling happy that I could contribute it :) 
thanks for the update !

routiful, 2019-01-07:
Thanks again. If you have any further issue, please feel free to get ticket :)

closed 2019-01-07 by routiful