ROBOTIS-GIT/dynamixel-workbench #218: How to use moveit bridge and controller tutorial with only two motors?

jcgarciaca, 2019-01-11:
I am trying to use `moveit bridge` according with the tutorial [here](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/), but since I just have 2 dynamixel motors when dragging interactive marker and push `Plan and execute`, the following error appears:

```
[ INFO] [1547241450.580576002]: Name : joint2, ID : 1, Model Number : 1030
[ INFO] [1547241450.620516952]: Name : joint3, ID : 2, Model Number : 1030
[ERROR] [1547241504.711027076]: groupSyncWrite addparam failed
================================================================================REQUIRED process [open_manipulator-1] has died!
process has died [pid 21534, exit code -11, cmd /home/personal/robotic_arm/catkin_ws/devel/lib/dynamixel_workbench_controllers/dynamixel_workbench_controllers /dev/ttyUSB0 57600 __name:=open_manipulator __log:=/home/personal/.ros/log/8d2e687c-15e1-11e9-a8eb-40a8f0a985e8/open_manipulator-1.log].
log file: /home/personal/.ros/log/8d2e687c-15e1-11e9-a8eb-40a8f0a985e8/open_manipulator-1*.log
Initiating shutdown!
================================================================================

```
Is it possible to test `moveit` with `bridge` and `controller` when using less motors than joints? It is for instance to take movement from two first joints and command my two motors. Actually, my final setup is quite similar because the robot has 3 joints, but only two are dynamixel motors.

routiful, 2019-01-14:
When you get planned trajectory from MoveIt!, moveit_bridge node are passed that into controller.
https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/6709e01dfde2eeb40435f441fe4a89eb067bfc30/dynamixel_workbench_moveit_bridge/src/moveit_bridge.cpp#L186

After that, the dynamixel_controller call a syncread function to get present position of Dynamixels using joint_names variable in trajectory_msgs.
getPresentPosition
https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/6709e01dfde2eeb40435f441fe4a89eb067bfc30/dynamixel_workbench_controllers/src/dynamixel_workbench_controllers.cpp#L678

Because some joints which are not Dynamixels does not has ID, however, the syncread functions would not work.
https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/6709e01dfde2eeb40435f441fe4a89eb067bfc30/dynamixel_workbench_controllers/src/dynamixel_workbench_controllers.cpp#L238

So, you should add some code to delete joint information about NOT dynamixels.
https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/6709e01dfde2eeb40435f441fe4a89eb067bfc30/dynamixel_workbench_moveit_bridge/src/moveit_bridge.cpp#L186

jcgarciaca, 2019-01-14:
@routiful,
Thank you for your answer, it worked.
I just have an extra question, does the `dynamixel_workbench_controllers` node uses feedback motor position for control?

routiful, 2019-01-15:
1. No, as you can see a config file(joint_2_0.yaml), the initial param was set '3'(Position_control) into [Operating_Mode](http://emanual.robotis.com/docs/en/dxl/x/xm430-w350/#operating-mode11)

https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/9bc9f791460b3ad44d4e535b174d252f55d0c3a0/dynamixel_workbench_controllers/config/joint_2_0.yaml#L8

If you want to use current motor position, you should set that param to '0'

2. I attached refer for this questions
https://ros-planning.github.io/moveit_tutorials/doc/time_parameterization/time_parameterization_tutorial.html#speed-control

jcgarciaca, 2019-01-15:
Hi.
I am not sure how to ask it...
Do you know why when I press `Plan and Execute` button the robot in `rviz` shows its movement (and the real motors move too), but seems like arm remains in `zero` position (black robot in image below), such that if I press again the button then a trajectory is planned from `zero` position instead of last position. In this case, the robot (and motors) shouldn't move since the robot is already in target position, but actually they move fast to zero and then move according with the generated trajectory.

[image]

jcgarciaca, 2019-01-17:
Thank you for your help.
I think it works when using the interactive marker and `Plan and Execute` button. Nevertheless, it fails when I plan a set of target waypoints and use `ExecuteTrajectoryAction`. In this case, the action returns an `ABORTED` status with `Cannot execute trajectory since ~allow_trajectory_execution was set to false` message. If I change `allow_trajectory_execution` to `true` [here](https://github.com/ROBOTIS-GIT/open_manipulator/blob/05812f6cccc53ff0e72ed657639d0d89b8939a69/open_manipulator_controller/launch/open_manipulator_moveit.launch#L22) then run `roslaunch open_manipulator_controller open_manipulator_moveit.launch sample_duration:=0.030`, the following error appears: `[FATAL] [1547764341.890861350]: Parameter '~moveit_controller_manager' not specified. This is needed to identify the plugin to use for interacting with controllers. No paths can be executed`.

<rosparam file="$(find motoman_mh6full_moveit_config)/config/controllers.yaml"/>

Do you know how could I solve it?

I added below lines to `open_manipulator_moveit_controller_manager.launch.xml`, but it is still a controller list missing.
```
  <arg name="moveit_controller_manager" default="moveit_simple_controller_manager/MoveItSimpleControllerManager"/>
  <param name="moveit_controller_manager" value="$(arg moveit_controller_manager)"/>
```
Then I created a `controller.yaml` file defined as:
```
controller_list:
  - name: ''
    action_ns: joint_trajectory_action
    type: FollowJointTrajectory
    joints: [joint1, joint2, joint3, joint4]
```
But `roslaunch open_manipulator_controller open_manipulator_moveit.launch sample_duration:=0.030` shows the following messages:
```
[ WARN] [1547765463.452313711]: Waiting for /joint_trajectory_action to come up
[ INFO] [1547765466.851636309]: Failed to call service get_planning_scene, have you launched move_group? at /tmp/binarydeb/ros-kinetic-moveit-ros-planning-0.9.15/planning_scene_monitor/src/planning_scene_monitor.cpp:498
[ INFO] [1547765467.065878662]: Constructing new MoveGroup connection for group 'arm' in namespace ''
[ WARN] [1547765469.452610338]: Waiting for /joint_trajectory_action to come up
[ERROR] [1547765475.452994387]: Action client not connected: /joint_trajectory_action
```

routiful, 2019-01-18:
I think this further question is about MovIt! configuration.
I assume that you can find releated question on [ROS ANSWER](https://answers.ros.org/questions/)

closed 2019-01-29 by gpwhd07