ROBOTIS-GIT/dynamixel-workbench #220: Are H54-200-S500-R and H54-100-S500-R supported?

garaemon, 2019-01-16:
Are H54-200-S500-R and H54-100-S500-R supported on dynamixel-workbench-controllers?

1. How to setup?
U2D2

2. Which Dynamixel have you used? and how many? **(Please describe below format to all connected Dynamixels)**

   - Model Name
H54-200-S500-R and H54-100-S500-R

   - ID
222 and 111

   - Baud Rate of Dynamixels
57600

   - Protocol Version
2

3. Write down the commands you used in order

  ```
  $ roscore    
  $ roslaunch dynamixel_workbench_controllers dynamixel_controllers.launch
  ```

yaml file is

```yaml
rarm2_j:
  ID: 111
  Return_Delay_Time: 0
  Operating_Mode: 0
rarm3_j:
  ID: 222
  Return_Delay_Time: 0
  Operating_Mode: 0

```
 
4.
roslaunch doest not report any error but `/joint_states` values are strange.

```
$ rostopic echo -n 1 /joint_states
header: 
  seq: 425
  stamp: 
    secs: 1547631091
    nsecs:  63027000
  frame_id: ''
name: [rarm2_j, rarm3_j]
position: [-19109.78125, -19109.78125]
velocity: [-325496.9375, -325496.9375]
effort: [-65415.421875, -65415.421875]
  ```
  
5. Please, describe detailedly what difficulty you are in 
- The values of position, velocity and effort looks strange.
- [Address values defined in the header](https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/master/dynamixel_workbench_controllers/include/dynamixel_workbench_controllers/dynamixel_workbench_controllers.h#L47) is different from [the addresses in the manual](http://emanual.robotis.com/docs/en/dxl/pro/h54-200-s500-r/#present-position
).

routiful, 2019-01-17:
Thank you for your comment :)
I will add some code to handle this issue soon.

routiful, 2019-01-24:
@garaemon 

I updated fixed code to support PRO.
Please pull master branch.
https://github.com/ROBOTIS-GIT/dynamixel-workbench/pull/221

garaemon, 2019-01-24:
Thanks @routiful 

I updated to ce02e31a59570c2cd6eb53feeb51c3e378546c1d and I confirmed that my H54-200-S500-R and H54-100-S500R provide correct joint angles with current control mode.

However, my XM540-W150 provides strange joint angles.

If I use 9bc9f791460b3ad44d4e535b174d252f55d0c3a0 (= 2.0.0), XM540-W150 provides correct joint angles and H54200-S500R provides wired values.

The following are console output when I use ce02e31a59570c2cd6eb53feeb51c3e378546c1d:
```
$ roslaunch test.launch
... logging to /home/rueda/.ros/log/37f10ccc-1f80-11e9-8513-fb2f44d3ffde/roslaunch-gitai-cockpit-32718.l$
g
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
...
[ INFO] [1548299452.653562556]: Name : rarm1_pitch_j, ID : 221, Model Number : 54024
[ INFO] [1548299452.701477614]: Name : rarm2_roll_j, ID : 111, Model Number : 53768
[ INFO] [1548299452.749458975]: Name : rarm3_yaw_j, ID : 222, Model Number : 54024
[ INFO] [1548299452.797490041]: Name : rarm4_pitch_j, ID : 112, Model Number : 53768
[ INFO] [1548299452.845488394]: Name : rarm5_yaw_j, ID : 51, Model Number : 1130
[ INFO] [1548299452.893475539]: Name : rarm6_roll_j, ID : 52, Model Number : 1130
[ INFO] [1548299452.941464856]: Name : rarm7_pitch_j, ID : 53, Model Number : 1130
[ INFO] [1548299453.485605848]: [DynamixelDriver] Succeeded to add sync write handler
[ INFO] [1548299453.486092353]: [DynamixelDriver] Succeeded to add sync write handler
```

```
$ rostopic echo -n /dynamixel/dynamixel_state
dynamixel_state: 
  - 
    name: "rarm1_pitch_j"
    id: 221
    present_position: -20368
    present_velocity: 0
    present_current: 0
  - 
    name: "rarm2_roll_j"
    id: 111
    present_position: 1836
    present_velocity: 0
    present_current: 0
  - 
    name: "rarm3_yaw_j"
    id: 222
    present_position: 2132
    present_velocity: 0
    present_current: 0
  - 
    name: "rarm4_pitch_j"
    id: 112
    present_position: 67534
    present_velocity: 0
    present_current: 0
  - 
    name: "rarm5_yaw_j"
    id: 51
    present_position: -1945990398
    present_velocity: -1912435454
    present_current: -28670
  - 
    name: "rarm6_roll_j"
    id: 52
    present_position: -1945990398
    present_velocity: -1912435454
    present_current: -28670
  - 
    name: "rarm7_pitch_j"
    id: 53
    present_position: -1945990398
    present_velocity: -1912435454
    present_current: -28670
---

```

routiful, 2019-01-24:
I had a mistake when readhandler are set.
I modified the code and test XM series.
Please check again in master branch

https://github.com/ROBOTIS-GIT/dynamixel-workbench/pull/224/commits/0fd3702d9313f2008a5ae566ec83bf6ca27ccb37

garaemon, 2019-01-24:
Hmm, I tested the latest master (529aaded82fa1876779791fab3705d2606f7028a) but XM540-W150s still does not work.

And dynamixel_workbench_controllers outputs error messages;
```
$ roslaunch test.launch
...
[ INFO] [1548307951.680589631]: Name : rarm1_pitch_j, ID : 221, Model Number : 54024
[ INFO] [1548307951.728538562]: Name : rarm2_roll_j, ID : 111, Model Number : 53768
[ INFO] [1548307951.776522547]: Name : rarm3_yaw_j, ID : 222, Model Number : 54024
[ INFO] [1548307951.824544007]: Name : rarm4_pitch_j, ID : 112, Model Number : 53768
[ INFO] [1548307951.872542783]: Name : rarm5_yaw_j, ID : 51, Model Number : 1130
[ INFO] [1548307951.920499727]: Name : rarm6_roll_j, ID : 52, Model Number : 1130
[ INFO] [1548307951.968519992]: Name : rarm7_pitch_j, ID : 53, Model Number : 1130
[ INFO] [1548307952.512908775]: [DynamixelDriver] Succeeded to add sync write handler
[ INFO] [1548307952.513285546]: [DynamixelDriver] Succeeded to add sync write handler
[ERROR] [1548307952.592505146]: groupSyncRead getdata failed
[ERROR] [1548307952.624497706]: groupSyncRead getdata failed
[ERROR] [1548307952.656513666]: groupSyncRead getdata failed
[ERROR] [1548307952.688501234]: groupSyncRead getdata failed
```

```
$ rostopic echo -n 1 /dynamixel/dynamixel_state
dynamixel_state: 
  - 
    name: "rarm1_pitch_j"
    id: 221
    present_position: -26125
    present_velocity: 0
    present_current: 14528
  - 
    name: "rarm2_roll_j"
    id: 111
    present_position: 3255
    present_velocity: 0
    present_current: 32764
  - 
    name: "rarm3_yaw_j"
    id: 222
    present_position: 2375
    present_velocity: 0
    present_current: 19432
  - 
    name: "rarm4_pitch_j"
    id: 112
    present_position: 91038
    present_velocity: 0
    present_current: 32764
  - 
    name: "rarm5_yaw_j"
    id: 51
    present_position: -1945990398
    present_velocity: -1912435454
    present_current: 15824
  - 
    name: "rarm6_roll_j"
    id: 52
    present_position: -1945990398
    present_velocity: -1912435454
    present_current: 32764
  - 
    name: "rarm7_pitch_j"
    id: 53
    present_position: -1945990398
    present_velocity: -1912435454
    present_current: -18986
---
```

garaemon, 2019-01-24:
And also, present current values of H54200-S500R are not zero even though the control mode is current control and no goal current is specified.

routiful, 2019-01-24:
I've tested XH540-W150. The read value is shown right
Please double check your local dynamixel-workbench package.

```
[ INFO] [1548309362.401439803]: Name : pan, ID : 1, Model Number : 1130
[ INFO] [1548309362.446605982]: [DynamixelDriver] Succeeded to add sync write handler
[ INFO] [1548309362.446686772]: [DynamixelDriver] Succeeded to add sync write handler
```

```
---
dynamixel_state: 
  - 
    name: "pan"
    id: 1
    present_position: 1486
    present_velocity: -302
    present_current: -53
---
dynamixel_state: 
  - 
    name: "pan"
    id: 1
    present_position: 1433
    present_velocity: -302
    present_current: -54
---
dynamixel_state: 
  - 
    name: "pan"
    id: 1
    present_position: 1391
    present_velocity: -302
    present_current: -56
---
dynamixel_state: 
  - 
    name: "pan"
    id: 1
    present_position: 1344
    present_velocity: -302
    present_current: -58
---
dynamixel_state: 
  - 
    name: "pan"
    id: 1
    present_position: 1296
    present_velocity: -302
    present_current: -50
---
dynamixel_state: 
  - 
    name: "pan"
    id: 1
    present_position: 1248
    present_velocity: -302
    present_current: -54
---
dynamixel_state: 
  - 
    name: "pan"
    id: 1
    present_position: 1196
    present_velocity: -303
    present_current: -54
---
```

routiful, 2019-01-24:
> And also, present current values of H54200-S500R are not zero even though the control mode is current control and no goal current is specified.

Because motor are controlled to maintain it's desired position, the current value is not always zero.

garaemon, 2019-01-24:
I found that if I use XM540-W150 and XM430-W210 motors, they work well (`/dev/ttyUSB1`).

However If I use XM540-W150, PRO-H54-100-S500-R and PRO-200-S500-R, XM540-W150 motors does not work well (`/dev/ttyUSB0`).

Cannot I mix PRO-H54 and XM dynamixels on the same usb bus?

* `/dev/ttyUSB0`

```
$ rosrun dynamixel_workbench_controllers find_dynamixel /dev/ttyUSB0
[ INFO] [1548310114.209240855]: Succeed to init(9600)
[ INFO] [1548310114.209263537]: Wait for scanning...
[ INFO] [1548310135.993718262]: Find 0 Dynamixels
[ INFO] [1548310135.994483052]: Succeed to init(57600)
[ INFO] [1548310135.994547197]: Wait for scanning...
[ INFO] [1548310153.881290608]: Find 7 Dynamixels
[ INFO] [1548310153.881358474]: id : 51, model name : XM540-W150
[ INFO] [1548310153.881386076]: id : 52, model name : XM540-W150
[ INFO] [1548310153.881412858]: id : 53, model name : XM540-W150
[ INFO] [1548310153.881438255]: id : 111, model name : PRO-H54-100-S500-R
[ INFO] [1548310153.881465542]: id : 112, model name : PRO-H54-100-S500-R
[ INFO] [1548310153.881490245]: id : 221, model name : PRO-H54-200-S500-R
[ INFO] [1548310153.881512825]: id : 222, model name : PRO-H54-200-S500-R
[ INFO] [1548310153.881784643]: Succeed to init(115200)
[ INFO] [1548310153.881811420]: Wait for scanning...
[ INFO] [1548310171.542250528]: Find 0 Dynamixels
[ INFO] [1548310171.542829566]: Succeed to init(1000000)
[ INFO] [1548310171.542876253]: Wait for scanning...
[ INFO] [1548310188.870518067]: Find 0 Dynamixels
[ INFO] [1548310188.870858372]: Succeed to init(2000000)
[ INFO] [1548310188.870885001]: Wait for scanning...
[ INFO] [1548310206.177112296]: Find 0 Dynamixels
[ INFO] [1548310206.177501446]: Succeed to init(3000000)
[ INFO] [1548310206.177536227]: Wait for scanning...
[ INFO] [1548310223.477649535]: Find 0 Dynamixels
[ INFO] [1548310223.478132253]: Succeed to init(4000000)
[ INFO] [1548310223.478183306]: Wait for scanning...
[ INFO] [1548310240.774705871]: Find 0 Dynamixels

```

* `/dev/ttyUSB1`

```
$ rosrun dynamixel_workbench_controllers find_dynamixel /dev/ttyUSB1
[ INFO] [1548310121.466501045]: Succeed to init(9600)
[ INFO] [1548310121.466534844]: Wait for scanning...
[ INFO] [1548310143.250816046]: Find 0 Dynamixels
[ INFO] [1548310143.251077955]: Succeed to init(57600)
[ INFO] [1548310143.251097142]: Wait for scanning...
[ INFO] [1548310161.141894437]: Find 7 Dynamixels
[ INFO] [1548310161.141920870]: id : 1, model name : XM540-W150
[ INFO] [1548310161.141928447]: id : 2, model name : XM540-W150
[ INFO] [1548310161.141935059]: id : 3, model name : XM430-W210
[ INFO] [1548310161.141942721]: id : 4, model name : XM430-W210
[ INFO] [1548310161.141950703]: id : 5, model name : XM430-W210
[ INFO] [1548310161.141959088]: id : 6, model name : XM430-W210
[ INFO] [1548310161.141965935]: id : 7, model name : XM430-W210
[ INFO] [1548310161.142185158]: Succeed to init(115200)
[ INFO] [1548310161.142197269]: Wait for scanning...
[ INFO] [1548310178.803099882]: Find 0 Dynamixels
[ INFO] [1548310178.803434345]: Succeed to init(1000000)
[ INFO] [1548310178.803447954]: Wait for scanning...
[ INFO] [1548310196.131442570]: Find 0 Dynamixels
[ INFO] [1548310196.131707716]: Succeed to init(2000000)
[ INFO] [1548310196.131719949]: Wait for scanning...
[ INFO] [1548310213.437237154]: Find 0 Dynamixels
[ INFO] [1548310213.437504905]: Succeed to init(3000000)
[ INFO] [1548310213.437517759]: Wait for scanning...
[ INFO] [1548310230.735885201]: Find 0 Dynamixels
[ INFO] [1548310230.736166650]: Succeed to init(4000000)
[ INFO] [1548310230.736179520]: Wait for scanning...
[ INFO] [1548310248.036070340]: Find 0 Dynamixels
```

routiful, 2019-01-24:
Sorry for inconvinience.
For now, you can't mix them in same use bus. 
Because of PRO is not compatible with recent control table of protocol 2.0 as XM.

But we are ready for A firmware for PRO (http://emanual.robotis.com/docs/en/dxl/pro/h54-200-s500-ra/) that is same control table of protocol 2.0. 
Dynamixel-Workbench will be updated to support it.
This update will be released next month. After update, you can use them in same bus.

garaemon, 2019-01-24:
> This update will be released next month. After update, you can use them in same bus.

I understood. I'm looking forward to next month release.

closed 2019-01-24 by garaemon

routiful, 2019-01-24:
Good. I will let you know next updates in this thread.

garaemon, 2019-01-24:
@routiful thank you.