ROBOTIS-GIT/dynamixel-workbench #226: Dynamixel MX-12W Model Number Issue

hjsuh94, 2019-02-02:
1. How to setup? (ex, U2D2, OpenCR,...)
I am using OpenCR with the Arduino IDE

2. Which Dynamixel have you used? and how many? **(Please describe below format to all connected Dynamixels)**
   - Model Name : MX-12W
   - ID: 1 - 4 
   - Baud Rate of Dynamixels: 1Mbps
   - Protocol Version: 1.0
 
3. Write down the commands you used in order
  I tried the p_Monitor example from the Arduino IDE, and tried to issue wheel command to the MX-12Ws. 
 
4. Copy and Paste your error message on terminal

  [DynamixelDriver] Failed to get the Tool
  
5. Please, describe detailedly what difficulty you are in 

https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/529aaded82fa1876779791fab3705d2606f7028a/dynamixel_workbench_toolbox/include/dynamixel_workbench_toolbox/dynamixel_item.h#L36

It looks like this line has the wrong model number definition for MX-12Ws. With the higher byte the model-number should be 360 instead of 104 - this seems to have fixed the issue.

routiful, 2019-02-07:
The model number of MX-12W is `104`. 
http://emanual.robotis.com/docs/en/dxl/mx/mx-12w/#control-table-of-eeprom-area

hjsuh94, 2019-02-07:
In the old manual it says that the lower byte is 104 while the higher byte is 1, and the current code seems to read in 2 bytes to identify the model number, resulting in 104 + 256 = 360. 

http://support.robotis.com/en/product/actuator/dynamixel/mx_series/mx-12w.htm

For instance, when I use this method:
 https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/529aaded82fa1876779791fab3705d2606f7028a/dynamixel_workbench_toolbox/include/dynamixel_workbench_toolbox/dynamixel_driver.h#L103

After plugging in the MX-12W, the return value is 360, and this seems to be causing confusion with the code. Are you able to reproduce this issue?

routiful, 2019-02-08:
Sorry for inconvenience. 
I found that bug in new emanual and open issue to amend that model number.
Moreover, I made a PR to modified model number of MX-12W and it worked correctly.
https://github.com/ROBOTIS-GIT/dynamixel-workbench/pull/227

Thank you for your contribution :)

closed 2019-02-10 by hjsuh94