ROBOTIS-GIT/dynamixel-workbench #228: groupSyncRead getdata failed with 5 dynamixel(Pro and XH) Unable to get Current Position

Dtad, 2019-02-12:
I have 5 motors connected with U2D2 and I am unable to get position from four of them. 
1x H54P-100-S500-R, 
2x DYNAMIXEL Pro M42-10-S260-R 
2x,DYNAMIXEL XH430-V350-R 
Always gives follwoing errors:
[ERROR] [1549985476.041870182]: [TxRxResult] There is no status packet!
[ERROR] [1549985476.042193326]: groupSyncRead getdata failed
[ERROR] [1549985476.042358334]: groupSyncRead getdata failed

routiful, 2019-02-13:
Because of different address at Present Position ([XH](http://emanual.robotis.com/docs/en/dxl/x/xh430-w350/#present-position132), [PRO](http://emanual.robotis.com/docs/en/dxl/pro_plus/h54p-200-s500-r/#present-position580)), You couldn't get data.

You should add another syncreadhandler and modified getPresentPosition function.

I have a plan to solve this complex situation when someone want to use various model.

Dtad, 2019-02-20:
How long ROBOTIS will take to solve this problem? It would be great if its solved as soon as possible.

routiful, 2019-04-09:
@Dtad 
I'm sorry for late reply. I can't give a specific date when it updated. 
I'll give you comment to you as soon as possible.

RyanJaehyunShim, 2019-05-29:
@Dtad 

We will write on our e-manual if we add a new function which might allow using different models at the same time. Please stay tuned on our website for future updates. 

Regards,
Ryan

closed 2019-05-29 by RyanJaehyunShim

tomsepe, 2019-07-03:
I also am having this same problem:
these are the servos
[ INFO] [1562186252.178565064]: id : 1, model name : PRO-PLUS-H42P-020-S300-R
[ INFO] [1562186252.178679438]: id : 2, model name : PRO-PLUS-H42P-020-S300-R
[ INFO] [1562186252.178780010]: id : 3, model name : PRO-PLUS-H42P-020-S300-R
[ INFO] [1562186252.178878603]: id : 4, model name : PRO-PLUS-H42P-020-S300-R
[ INFO] [1562186252.178977509]: id : 5, model name : PRO-PLUS-M42P-010-S260-R
[ INFO] [1562186252.179074071]: id : 6, model name : PRO-PLUS-M42P-010-S260-R
[ INFO] [1562186252.179172456]: id : 20, model name : XM430-W350
[ INFO] [1562186252.179267455]: id : 21, model name : XM430-W350
[ INFO] [1562186252.179363236]: id : 22, model name : XM430-W350
 
If I am just connecting to the Pro models or the X-series separately I am fine. but as soon as I try to connect to both types then I get this "groupSyncRead getdata failed" message

RyanJaehyunShim, 2019-07-04:
@tomsepe 

Our dynamixel-workbench package does not support such cases currently. We will post on my eManual if we update our package and add more functions. 

Regards,
Ryan

ROBOTIS-Will, 2019-09-19:
@tomsepe , @Dtad 
The Syncread and Syncwrite functions are intended to read data from or write data to an identical address of connected DYNAMIXELs, so it seems difficult to change the way those functions work.
However, using [Indirect Address and Indirect Data](http://emanual.robotis.com/docs/en/dxl/x/xm430-w350/#indirect-address-indirect-data), this "issue" could be resolved.

For X series, assign the address to use at **Indirect Address 29** while using **Indirect Address 1** for PRO+.
The address of X series's **Indirect Data 29** starts at **634** and so does the **Indirect Data 1** of Pro+ series.
Since Indirect Address and Data is in RAM area, the configuration has to be done whenever the system reboots.
[http://emanual.robotis.com/docs/en/dxl/pro_plus/h54p-200-s500-r/#indirect-address-indirect-data](http://emanual.robotis.com/docs/en/dxl/pro_plus/h54p-200-s500-r/#indirect-address-indirect-data)

Thanks.

Dtad, 2019-09-20:
@ROBOTIS-Will It would be really great if you can elaborate a bit in terms of values. Which values should i set on the Indirect Address 29 , Indirect Data 29, Indirect Address 1 and Indirect Data 1.