ROBOTIS-GIT/dynamixel-workbench #229: MX64AT, MX28AT, and MX12W incompatible with workbench (protocol 1.0)

usfbrian, 2019-02-21:
1. **How to setup? (ex, U2D2, OpenCR,...)**
OpenCM 9.04 without the expansion

2. **Which Dynamixel have you used? and how many? _(Please describe below format to all connected Dynamixels)_**

   - **Model Name**
   2 x MX64AT, firmware 42 with Protocol 1.0 selected in R+ Manager
   MX28AT, firmware 42 with Protocol 1.0 selected in R+ Manager
   MX12W, firmware 41
   connected serially

   - **ID**
   5 (MX28)
   4 (MX64)
   3 (MX64)
   2 (MX12)

   - **Baud Rate of Dynamixels**
   1000000

   - **Protocol Version**
   1.0 

3. **Write down the commands you used in order**

  ```
#include <DynamixelWorkbench.h>

#define  DEBUG  true

#define  DEVICE_NAME  "1"  // OpenCM 9.04

#define  DYNAMIXEL_BAUD  1000000
#define  COMPUTER_BAUD  1000000

#define  MOTOR_X_ID  4
#define  MOTOR_Y_ID  3
#define  MOTOR_Z_ID  2
#define  MOTOR_W_ID  5

#define  NUM_MOTORS  4

#if (DEBUG != false)
 #define  DEBUG_PRINT(x)  Serial.print(x)
 #define  DEBUG_PRINTLN(x)  Serial.println(x)
#else
 #define  DEBUG_PRINT(x)
 #define  DEBUG_PRINTLN(x)
#endif

void report_outcome(bool, const char *); 

DynamixelWorkbench dxl_wb;

int32_t goal_velocity[NUM_MOTORS] = { 0, 0, 0, 0 };
uint8_t rx_buffer[10];

void setup()
{
  Serial.begin(COMPUTER_BAUD);
  while (!Serial)  {}  // Wait for serial connection before proceeding

  bool result = false;
  const char *log;

  uint8_t dxl_id[NUM_MOTORS] = {MOTOR_X_ID, MOTOR_Y_ID, MOTOR_Z_ID, MOTOR_W_ID};

  DEBUG_PRINTLN("init...");
  result = dxl_wb.init(DEVICE_NAME, DYNAMIXEL_BAUD, &log);
  report_outcome(result, log);

  DEBUG_PRINTLN("handler...");
  result = dxl_wb.setPacketHandler(1.0f, &log);
  DEBUG_PRINTLN( dxl_wb.getProtocolVersion() );
  report_outcome(result, log);

  DEBUG_PRINTLN("ping...");
  for (int idx = 0; idx < NUM_MOTORS; ++idx) {
    uint16_t model_number = 0;
    result = dxl_wb.ping(dxl_id[idx], &model_number, &log);
    report_outcome(result, log);
    DEBUG_PRINT("id: ");  DEBUG_PRINT(dxl_id[idx]);
    DEBUG_PRINT(" model: ");  DEBUG_PRINTLN(model_number);
  }

  DEBUG_PRINTLN("vel mode...");
  for (int idx = 0; idx < NUM_MOTORS; ++idx) {
    result = dxl_wb.wheelMode(dxl_id[idx], 0, &log);
    report_outcome(result, log);
  }

  dxl_wb.itemWrite(MOTOR_Z_ID, "Goal_Velocity", 100, &log);
  report_outcome(result, log);
  delay(1000);
  dxl_wb.itemWrite(MOTOR_Z_ID, "Goal_Velocity", -100);
  report_outcome(result, log);
  delay(1000);
  dxl_wb.itemWrite(MOTOR_Z_ID, "Goal_Velocity", 0); 
  report_outcome(result, log);
  delay(3000);
}

void loop()  {}

void report_outcome(bool outcome, const char *log)
{
  if (outcome == false) {
    DEBUG_PRINTLN(log);
    DEBUG_PRINTLN("fail");
  } else {
    DEBUG_PRINTLN("pass");
  }
}

  ```
 
4. **Copy and Paste your error message on terminal**

  ```
init...
pass
handler...
1.00
pass
ping...
pass
id: 4 model: 311
pass
id: 3 model: 311
pass
id: 2 model: 360
pass
id: 5 model: 30
vel mode...
pass
pass
pass
pass
pass
pass
pass
 
  ```
  
5. **Please, describe detailedly what difficulty you are in** 

We've created a 3d Cartesian gantry using the SDK and separate port handlers for protocol 2.0 and 1.0. The workspace of the gantry is larger than 7 turns and maximum resolution is valued, so running 3 of the motors in Protocol 2.0 is nice. However, the Z-axis uses a high-speed ball screw (the MX12) and have several bugs using a workaround method to increase the range of the multi-turn position control, as this motor uses protocol 1.0.

Instead of rewriting the codebase to use a velocity controller in the SDK, it would be much simpler to use the workbench API.

With the latest bug-fix #226 the MX12W was working for a day when all the motors were downgraded to protocol 1.0. I'm unsure what has changed, but the motor goes into Torque Enable during initialization and resists manual turning, but any velocity commands are ignored. Also, the response to wheelMode() yields the same error as issue #226 (Failed to get the Tool). I deleted the /tmp folder where the previous library builds were cached in case this was an issue, and I confirmed that the MX12W definition in my library is 360 instead of 104.

**I'd like the MX12W to respond to protocol 1.0 Goal_Velocity/Moving_Speed commands using DynamixelWorkbench, with the long-term goal being syncWrite to register 32, which was functional yesterday. The motor turns on, and other motors are able to respond to commands.**

An alternative fix would be the MX12W running Protocol 2.0. Is there any guidance on if this is possible, and when would that be expected?

Thanks for your time and assistance!

usfbrian, 2019-02-22:
Using `goalVelocity(...)` instead of `itemWrite(...)` has the MX12W operational again.
But, the syncWrite handler turns all motors except the MX12W.

In the worst case, syncWrite can move the other 3 motors and a separate command moves z, but it'd be ideal to have the code as simple as possible (specifically, serial read into the single goal_velocity array and issue the syncWrite command)

routiful, 2019-04-09:
@usfbrian 
Sorry for late reply.
[SyncWrite Inst](http://emanual.robotis.com/docs/en/dxl/protocol1/) is to be worked multiple Dynamixels with coincidence control table. As your environments, MX-12W is completely not a same as other MX series. 
You can't use syncwrite instruction with them.
I suggest that you can use [Reg Write Inst](http://emanual.robotis.com/docs/en/dxl/protocol1/#reg-write) to async other Dynamixels or simply purchase one more MX28 or 64.

routiful, 2019-04-09:
@usfbrian 
I found tha you used MX28 and MX64 with 2.0 firmware.
If you want to all MX series with protocol 1.0, you are supposed to set same 1.0 firmware.

usfbrian, 2019-04-12:
Thanks for the details about using Reg Write Inst. We chose the MX12 for its high speed as it's running a ball screw, though we also wanted Protocol 2.0 as it provides greater than plus/minus 7 turns.

I suppose until there is a high-speed motor with extended range, I'll change our design to a rack-and-pinion or use a velocity controller.

Thanks for the help!

closed 2019-04-12 by usfbrian