ROBOTIS-GIT/dynamixel-workbench #230: Following Errors: [TxRxResult] There is no status packet! ---- groupSyncRead getdata failed --- Failed to get Present Position

haltermc, 2019-02-27:
ISSUE TEMPLATE ver. 1.0.0

Before you open issue, please refer to [ROBOTIS e-Manual](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/)

1. How to setup? U2D2

2. Which Dynamixel have you used? and how many? 

   - 6 XM430-W350R & 1 XM540-W270-R all with firmware version 42

   - 1-7

   - 3000000

   - Protocol Version v2.0
 
3. Write down the commands you used in order

Term. 1:
$ roslaunch dynamixel_workbench_controllers dynamixel_controllers.launch

Term. 2:
$ rostopic pub /dynamixel_workbench/joint_trajectory trajectory_msgs/JointTrajectory "header:
  seq: 0
  stamp:
    secs: 0
    nsecs: 0
  frame_id: ''
joint_names:
- 'id_1, id_2, id_3, id_4, id_5, id_6, id_7'
points:
- positions: [0, 0, 0, 0, 0, -1.57, 0]
  velocities: [0, 0, 0, 0, 0, 0, 0]
  accelerations: [0, 0, 0, 0, 0, 0, 0]
  effort: [0, 0, 0, 0, 0, 0, 0]
  time_from_start: {secs: 1, nsecs: 0}"
  
4. Copy and Paste your error message on terminal

SUMMARY
========

PARAMETERS
 * /dynamixel_info: /home/ros/catkin_...
 * /dynamixel_workbench/dxl_read_period: 0.01
 * /dynamixel_workbench/dxl_write_period: 0.01
 * /dynamixel_workbench/publish_period: 0.01
 * /dynamixel_workbench/use_cmd_vel_topic: False
 * /dynamixel_workbench/use_joint_states_topic: True
 * /dynamixel_workbench/use_moveit: True
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    dynamixel_workbench (dynamixel_workbench_controllers/dynamixel_workbench_controllers)

auto-starting new master
process[master]: started with pid [7730]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to bdbce9c4-3abb-11e9-b0c2-309c23d0fc30
process[rosout-1]: started with pid [7743]
started core service [/rosout]
process[dynamixel_workbench-2]: started with pid [7754]
[ INFO] [1551291363.551966619]: Name : id_1, ID : 1, Model Number : 1020
[ INFO] [1551291363.586926056]: Name : id_2, ID : 2, Model Number : 1120
[ INFO] [1551291363.621909133]: Name : id_3, ID : 3, Model Number : 1020
[ INFO] [1551291363.656890261]: Name : id_4, ID : 4, Model Number : 1020
[ INFO] [1551291363.691885838]: Name : id_5, ID : 5, Model Number : 1020
[ INFO] [1551291363.726912205]: Name : id_6, ID : 6, Model Number : 1020
[ INFO] [1551291363.761892022]: Name : id_7, ID : 7, Model Number : 1020
[ INFO] [1551291363.992902435]: [DynamixelDriver] Succeeded to add sync write handler
[ INFO] [1551291363.992920825]: [DynamixelDriver] Succeeded to add sync write handler
[ERROR] [1551291365.960031441]: [TxRxResult] There is no status packet!
[ERROR] [1551291365.960055771]: groupSyncRead getdata failed
[ERROR] [1551291365.960064731]: Failed to get Present Position
[ INFO] [1551291365.960073591]: 'id_1, id_2, id_3, id_4, id_5, id_6, id_7' is ready to move
[ INFO] [1551291365.960087621]: Succeeded to get joint trajectory!
[ INFO] [1551291365.960123531]: Complete Execution
================================================================================REQUIRED process [dynamixel_workbench-2] has died!
process has died [pid 7754, exit code -11, cmd /home/ros/catkin_ws/devel/lib/dynamixel_workbench_controllers/dynamixel_workbench_controllers /dev/ttyUSB0 3000000 __name:=dynamixel_workbench __log:=/home/ros/.ros/log/bdbce9c4-3abb-11e9-b0c2-309c23d0fc30/dynamixel_workbench-2.log].
log file: /home/ros/.ros/log/bdbce9c4-3abb-11e9-b0c2-309c23d0fc30/dynamixel_workbench-2*.log
Initiating shutdown!
================================================================================
[dynamixel_workbench-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
  
5. Please, describe detailedly what difficulty you are in 

We are trying to manually publish to dynamixel_workbench/joint_trajectory with the above format and inputs. As a result we get the above errors.

routiful, 2019-04-09:
@haltermc 
Sorry for late reply.
If you didn't have moveit config, you should set `use_moveit` param to false

RyanJaehyunShim, 2019-05-27:
@haltermc 

As there has been no update since the last question, I will close this issue. 

Regards,
Ryan

closed 2019-05-27 by RyanJaehyunShim