ROBOTIS-GIT/dynamixel-workbench #233: Can't find dynamixel_workbench_moveit_bridge

KMiyawaki, 2019-03-15:
1. How to setup? (ex, U2D2, OpenCR,...)

  - Connect XM540-W270-R x 6 and Dynamixel XM430-W210-R x 2 to JetsonTX2 via [DXHUB](http://www.besttechnology.co.jp/modules/onlineshop/index.php?fct=photo&p=151) and USB.
  - Follow [e-manual](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/#tutorials) 3.1.2 Downloads ROS Packages, and compile dynamixel-workbench, dynamixel-workbench-msgs, DynamixelSDK and open_manipulator_msgs.
  - Install dependencies.

2. Which Dynamixel have you used? and how many? **(Please describe below format to all connected Dynamixels)**

   - Model Name:XM540-W270-R
   - ID:30, (1, 11), (2, 12), 3
     - 1 and 11 are master-slave.
     - 2 and 12 are master-slave.
   - Model Name:XM430-W210-R
   - ID: 4, 5
   - Baud Rate of Dynamixels:1000000
   - Protocol Version:2.0

3. Write down the commands you used in order

```shell
git clone https://github.com/ROBOTIS-GIT/dynamixel-workbench.git
git clone https://github.com/ROBOTIS-GIT/dynamixel-workbench-msgs.git
git clone https://github.com/ROBOTIS-GIT/DynamixelSDK.git
git clone https://github.com/ROBOTIS-GIT/open_manipulator_msgs.git
sudo apt-get install ros-kinetic-moveit-core ros-kinetic-moveit-ros-planning ros-kinetic-moveit-ros-planning-interface
```

- After that, I wrote `moveit_bridge.launch` like this,

```xml
<launch>
  <arg name="robot_name"       default="oit_arm_201903"/>
  <arg name="planning_group"   default="arm"/>

  <node name="$(arg planning_group)" pkg="dynamixel_workbench_moveit_bridge" type="moveit_bridge" output="screen">
    <param name="planning_group"      value="$(arg planning_group)"/>
    <remap from="$(arg planning_group)/joint_trajectory" to="$(arg robot_name)/joint_trajectory"/>
  </node>
</launch>
```

1. Copy and Paste your error message on terminal

```shell
started roslaunch server http://10.42.0.1:39266/

SUMMARY
========

PARAMETERS
 * /arm/planning_group: arm
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    arm (dynamixel_workbench_movewit_bridge/moveit_bridge)

auto-starting new master
process[master]: started with pid [31446]
ROS_MASTER_URI=http://10.42.0.1:11311

setting /run_id to bcf48970-4704-11e9-8785-00044b8ccfdf
process[rosout-1]: started with pid [31469]
started core service [/rosout]
ERROR: cannot launch node of type [dynamixel_workbench_moveit_bridge/moveit_bridge]: dynamixel_workbench_moveit_bridge
```
  
5. Please, describe detailedly what difficulty you are in

    - I built original arm using some dynamisel servos, and want to control it by MoveIT.
    - However, `dynamixel_workbench_moveit_bridge` has gone. It seems to be deleted about 2 months ago.
    - How can I interface dynamixel servos to MoveIT ?

routiful, 2019-04-09:
@KMiyawaki 
Sorry for late reply.
`moveit_bridge` package are moved to another branch (feature-moveit-bridge)
https://github.com/ROBOTIS-GIT/dynamixel-workbench/tree/feature-moveit-bridge

KMiyawaki, 2019-04-10:
Thank you for your response.
I have found the feature-moveit-bridge branch, and used it.

In my understandings, MoveIt! talks with the robot via FollowJointTrajectoryAction interface.
https://moveit.ros.org/documentation/concepts/
However, moveit_bridge do not prepare the action server, and uses trajectory_msgs/JointTrajectory topic.

Are you planning to implement FollowJointTrajectoryAction server ?

If so, I can easily apply my another robot arm control program.
The program communicate with MoveIt! by using action client of FollowJointTrajectoryAction.

routiful, 2019-04-10:
Unfortunately, as team policy, we don't have any maintain or upgrade moveit_bridge package.

KMiyawaki, 2019-04-10:
OK. I understand. Thank you.

closed 2019-04-10 by KMiyawaki

5730289021-NN, 2021-08-03:
>Are you planning to implement FollowJointTrajectoryAction server ?

>If so, I can easily apply my another robot arm control program.
>The program communicate with MoveIt! by using action client of FollowJointTrajectoryAction.

As far as I know, using [dynamixel_control_hw](https://github.com/resibots/dynamixel_control_hw) to provide hardware interface then apply [JointTrajectoryController](http://wiki.ros.org/joint_trajectory_controller)[dynamixel_control] is the simplest way to work on for ROS1 

Alternatively on ROS2, [dynamixel_control](https://github.com/youtalk/dynamixel_control) is also found to be a very great "one-shot" solution.

I'm really interested if here, as official dynamixel provider, have an official solution on this. 
This feature could be useful in various use case.

ROBOTIS-Will, 2021-08-06:
Hi @5730289021-NN 
We've recently contacted the dynamixel_control package maintainer, and will contribute to update / upgrade this package to support ROS2.
The dynamixel_control is dependent to dynamixel workbench, but planning to replace with dynamixel_sdk.
Thank you for your input!