ROBOTIS-GIT/dynamixel-workbench #235: Unable to Read Motor Shutdown

adithyamurali, 2019-04-10:
Hi everyone, I am using the dynamixel_workbench_controller and SDK to connect to a manipulator arm from Trossen Robotics. The arm is essentially a chain of XM430 and XL430s.

Sometimes the dynamixel motors shuts down (RED LED flickering) so I wanted to understand why they were failing. I tried to read from the Shutdown (63) register with both itemRead and getSyncReadData but I always get the initial value ‘52’ for all the motors (even after the motor shuts down). For syncRead, the control_items address is 63 and data_length is 1 (so I’m assuming I’m reading at the right spot).

I’m not even sure if I’m interpreting the value correctly. Even when the motor is functioning, I’m reading a value of 52 (binary 00110100) means there is a Overheating, Electrical shock and Overload errors, according to the documentation screenshot below?

[image]

Documentation link: http://support.robotis.com/en/product/actuator/dynamixel_x/xm_series/xm430-w350.htm#bookmark16

Any insights and suggestions would be much appreciated, thanks!

yhna, 2019-04-11:
Hello, @adithyamurali 
The Shutdown(63) register in the control table of dynamixel is an option to shut down dynamixel when an error condition occurs.
If dynamixel has been shut down,  the **Hardware Error Status** (70) register indicates the shutdown reason.

adithyamurali, 2019-04-12:
Hi @yhna , thanks for the reply! I can now read the Hardware Error Status (70). It seems that when the motors are shutting down I'm seeing a Overload error (error 32, 0x20). I notice that this happens mostly for some motors on the arm (carrying more payload I believe). The motors work fine for several iterations of commanding the same joint positions but shutdown a while later. When I power cycle the motors, they start working again but for a while. Do you have any thoughts on reducing the frequency of these shutdowns (e.g. reducing profile velocity)?

yhna, 2019-04-15:
@adithyamurali 
Since the persistent load exceeds the maximum output of the Dynamixel, it can not be resolved by changing the parameters of Dynamixel. I recommend using a higher specification Dynamixel or reducing the weight of the robot.

adithyamurali, 2019-04-15:
Great, thanks for the clarifications!

closed 2019-04-15 by adithyamurali