ROBOTIS-GIT/dynamixel-workbench #236: OpenCm firmware - ROS ?

GMahmoud, 2019-04-18:
Hey Guys,

I am trying to control dynamixel servos with an openCM9.04 board. I managed to flash it and run some arduino scripts on it, for example `find_dynamixel`  . However,  when I want to bind it with ROS ie `rosrun dynamixel_workbench_controllers find_dynamixel /dev/ttyACM0` no motor is detected. 
So my question is : Should I install some firmware to make it run with ros packages or Am I missing something ? If yes where could I find this firmware please ?

Sorry if it seems so obvious but I do not have a big experience with dyna servos.

Cheers :)

yhna, 2019-04-22:
Hello, @GMahmoud 
If you want to control dynamixel in ros environment, you need a communication interface part like [U2D2.](http://emanual.robotis.com/docs/en/parts/interface/u2d2/) 
If you want to control dynamixel in OpenCM, you can control it by writing OpenCM [dynamixel workbench example](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/#opencr-and-opencm-tutorials) in OpenCM. OpenCM and ROS are separate systems.

GMahmoud, 2019-04-24:
Hello @yhna , 
Thanks for your answer. I have already figured it out. I used the OpenCM as USBtoDynamixel and then I could use dynamixel workbench ros package :)

closed 2019-04-24 by GMahmoud

Roger-Lo2000, 2022-03-11:
Hello @GMahmoud ,
I am facing the same issue of ROS find dynamixel recently,
Could you please tell me how do you use openCM as USB2Dynamixel?
Do I have to upload firmware to openCM or what ?
If so , Could you please tell what to do?
I am beginner of this.

Thank you.