ROBOTIS-GIT/dynamixel-workbench #239: PWM mode not possible with XL430-W250-T

ruffson, 2019-05-11:
1. How to setup? (ex, U2D2, OpenCR,...)

- `OpenCM 9.04` + `OpenCM 485 Expansion Board`

2. Which Dynamixel have you used? and how many? **(Please describe below format to all connected Dynamixels)**

   - 4x `XL4300-W250-T` daisy-chained

   - 1-4

   - 57600

   - 2.0
 
3. Write down the commands you used in order

  ```
#include <DynamixelWorkbench.h>

#if defined(__OPENCM904__)
#define DEVICE_NAME "3" //Dynamixel on Serial3(USART3)  <-OpenCM 485EXP
#else
#define DEVICE_NAME ""
#endif

#define BAUDRATE  57600
#define DXL_ID    1

DynamixelWorkbench dxl_wb;
const uint8_t handler_index = 0;

void setup()
{
  Serial.begin(57600);
  while (!Serial);

  const char *log;
  bool result = false;

  uint8_t dxl_id = DXL_ID;
  uint16_t model_number = 0;

  result = dxl_wb.init(DEVICE_NAME, BAUDRATE, &log);
  if (result == false)
  {
    Serial.println(log);
    Serial.println("Failed to init");
  }
  else
  {
    Serial.print("Succeeded to init : ");
    Serial.println(BAUDRATE);
  }

   result = dxl_wb.ping(dxl_id, &model_number, &log);
   if (result == false)
   {
     Serial.println(log);
     Serial.println("Failed to ping");
   }
   else
   {
     Serial.println("Succeeded to ping");
     Serial.print("id : ");
     Serial.print(dxl_id);
     Serial.print(" model_number : ");
     Serial.println(model_number);
   }

  result = dxl_wb.setPWMControlMode(dxl_id, &log);

  if (result == false)
  {
    Serial.println(log);
    Serial.println("Failed to set mode");
  }
  else
  {
    Serial.println("Succeeded to set mode");
  }

}

void loop()
{}
  ```
 
4. Copy and Paste your error message on terminal

  ```
 -> Succeeded to init : 57600
 -> Succeeded to ping
 -> id : 1 model_number : 1060
 -> [DynamixelWorkbench] Failed to set Operating Mode!
 -> Failed to set mode
  ```
  
5. Please, describe detailedly what difficulty you are in 

I want to control my motors in PWM mode. And in the snippet above I just try to enable PWM mode for one motor. However I cannot change the operation mode. I pretty much just modifed one of the example codes and modified it slightly. I can do position control just fine. I also tried  `result = dxl_wb.setOperatingMode(dxl_id, 16, &log);` which did not work either. According to the docs PWM should be supported for that servo model.
Any help is very much appreciated!

ruffson, 2019-05-17:
Is this the right repository to raise this issue or can someone recommend another channel that is more fitting?

yhna, 2019-05-21:
@ruffson 
Sorry for the delay.
Currently the Dynamixel Workbench can not set the PWM Control Mode of xl430 Dynamixel.
Do not set it with Dynamixel Workbench, but set xl430 to PWM Control Mode in advance using [Dynamixel Wizard 2.0](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/).

This bug is fixed and will be updated.

Thanks.

ruffson, 2019-05-21:
> @ruffson
> Sorry for the delay.
> Currently the Dynamixel Workbench can not set the PWM Control Mode of xl430 Dynamixel.
> Do not set it with Dynamixel Workbench, but set xl430 to PWM Control Mode in [...]

Thank you!
Is there any alternative for macOS? The Wizard is only available for Win and Linux.

yhna, 2019-05-21:
Unfortunately, macOS requires the use of the Dynamixel SDK.

Thanks.

RyanJaehyunShim, 2019-05-27:
@ruffson 

Wizard2.0 is available for macOS as of last Friday. Please refer to the following link
(http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/)

Regards,
Ryan

closed 2019-05-27 by RyanJaehyunShim

ruffson, 2019-05-27:
> @ruffson
> 
> Wizard2.0 is available for macOS as of last Friday. Please refer to the following link
> (http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/)
> 
> Regards,
> Ryan

Thank you Ryan.
I don't know if you tested this before but this does not work: 

The macOS file is a `DynamixelWizard2Setup_x64.app.zip.mdlp`. If I remove the `mdlp` file ending and unzip the file I get a `Contents` folder (which is usually inside an app bundle). Okay fair enough I know how to navigate to the binary inside a macOS app.
If I run the binary `Contents/MacOS/DynamixelWizard2Setup` the Setup Wizard to install the program is opened. However, after the "Ready to Install" during installation, I get the error "Maintenance tool is not a bundle" and the last screen's title is: "The DYNAMIXEL Wizard 2.0 Wizard failed". I think you might have made a mistake bundling that app or I messed up here.

RyanJaehyunShim, 2019-05-28:
@ruffson 

If you have any questions or found any bugs about Wizard2.0, please ask them at the link below.
(http://en.robotis.com/service/forum.php)

Regards,
Ryan

ruffson, 2019-05-28:
> @ruffson
> 
> If you have any questions or found any bugs about Wizard2.0, please ask them at the link below.
> (http://en.robotis.com/service/forum.php)
> 
> Regards,
> Ryan

Okay will do. Thank you!

swiz23, 2019-07-23:
Has the ros-kinetic-dynamixel-workbench-toolbox from apt-get been updated with this fix? I'm trying to use the setPWMControlMode function in my own code, and I am still getting an error message for the XL430s.

rjshim, 2019-07-24:
@swiz23 

Please do set-up things using DXL Wizard. If you have any questions regarding Wizard, please ask it at the [link](http://en.robotis.com/service/forum.php)

swiz23, 2019-07-24:
Right, but for my specific application, I need the ability to switch between different operating modes including PWM while my ROS node is running. I would like to abstract away setting up the Dynamixel configs so that a potential user of my robot does not have to install the Dynamixel wizard to begin with.

swiz23, 2019-08-08:
Just as an update, I was able to work around this issue in code by just using the itemWrite function directly to set the PWM operating mode.

ruffson, 2019-08-08:
> Just as an update, I was able to work around this issue in code by just using the itemWrite function directly to set the PWM operating mode.

Oh, you set pwm operating mode without a U2D2?
Can you show a minimal example?

swiz23, 2019-08-08:
In your example code above, try substituting...

result = dxl_wb.setPWMControlMode(dxl_id, &log);

with...

result = dxl_wb.itemWrite(dxl_id, "Operating_Mode", 16, &log);

For my application, I'm compiling all my code on my Linux machine and just using U2D2 as a pass-through to the motor firmware. But I believe this should still work for your application. If you look at the source code [here](https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/master/dynamixel_workbench_toolbox/src/dynamixel_workbench_toolbox/dynamixel_workbench.cpp), setPWMControlMode calls setOperatingMode which ends up calling the writeRegister function (which is essentially the same thing as the itemWrite function). If you take a look at the setOperatingMode function, there's a bunch of checks to make sure the motor you want to write PWM mode to actually supports it. I'm not 100% sure why it's failing, but if you just use the itemWrite function directly, it seems to work.

ruffson, 2019-08-09:
> In your example code above, try substituting...
>
> result = dxl_wb.setPWMControlMode(dxl_id, &log);
>
> with...
>
> result = dxl_wb.itemWrite(dxl_id, "Operating_Mode", 16, &log);
>
> For my application, I'm [...]

Thanks so much, really appreciate it!
Will give it a try.

ruffson, 2019-08-14:
> In your example code above, try substituting...
>
> result = dxl_wb.setPWMControlMode(dxl_id, &log);
>
> with...
>
> result = dxl_wb.itemWrite(dxl_id, "Operating_Mode", 16, &log);
>
> For my application, I'm [...]

As a little update: Directly trying to cann `itemWrite` or even `writeRegister` did not work for me. However, using yet another Robotis Dynamixel library worked perfectly. It's called `Dynamixel2Arduino`: http://emanual.robotis.com/docs/en/parts/controller/opencm904/#library-api

Xiaodao-chen, 2023-11-21:
> @ruffson Sorry for the delay. Currently, the Dynamixel Workbench can not set the PWM Control Mode of xl430 Dynamixel. Do not set it with Dynamixel Workbench, but set xl430 to PWM Control Mode in [...]

now does it support?

ruffson, 2023-11-23:
> > Just as an update, I was able to work around this issue in code by just using the itemWrite function directly to set the PWM operating mode.
> 
> could you tell me how did you set PWM mode, I just want to advertise PWM to make the servo move at my desired velocity

I'm sorry, I don't use Dynamixels anymore.