ROBOTIS-GIT/dynamixel-workbench #241: Why don't you describe about binary install in the emanual?

yoshimalucky, 2019-05-21:
I'm wondering why you don't describe about its binary install in the e-manual.

Unless they would like to modify or develop the packages, the binary install seems to be enough.

yhna, 2019-05-21:
@yoshimalucky 
If the bug has been fixed in this source code or there is a new update, the user must obtain the latest version of the source code.
For this reason, we recommend downloading the source code through github.

yoshimalucky, 2019-05-21:
In some cases, the binary package works better.

This is an example.
https://github.com/ROBOTIS-GIT/dynamixel-workbench/issues/240

This happens only if we use the source codes.

yoshimalucky, 2019-05-24:
Never mind. This is the problem only for the sample codes. Not for ROS package.

closed 2019-05-24 by yoshimalucky