ROBOTIS-GIT/dynamixel-workbench #244: dynamixel-workbench does not work..

lolongin, 2019-05-24:
ISSUE TEMPLATE ver. 1.0.0

Before you open issue, please refer to [ROBOTIS e-Manual](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/)

1. How to setup? (ex, U2D2, OpenCR,...)

2. Which Dynamixel have you used? and how many? **(Please describe below format to all connected Dynamixels)**

   - Model Name
xl320

   - ID
1
   - Baud Rate of Dynamixels
1000000

   - Protocol Version
2.0
 
3. Write down the commands you used in order

#include <DynamixelWorkbench.h>

#if defined(__OPENCM904__)
  #define DEVICE_NAME "1" //Dynamixel on Serial3(USART3)  <-OpenCM 485EXP
#elif defined(__OPENCR__)
  #define DEVICE_NAME ""
#endif          

#define BAUDRATE  1000000

DynamixelWorkbench dxl_wb;

void setup() 
{
  Serial.begin(57600);
  while(!Serial); // Wait for Opening Serial Monitor

  const char *log = NULL;
  bool result = false;

  uint8_t scanned_id[16];
  uint8_t dxl_cnt = 0;
  uint8_t range = 100;

  result = dxl_wb.init(DEVICE_NAME, BAUDRATE, &log);
  if (result == false)
  {
    Serial.println(log);
    Serial.println("Failed to init");
  }
  else
  {
    Serial.print("Succeeded to init : ");
    Serial.println(BAUDRATE);  
  }

  Serial.println("Wait for scan...");
  result = dxl_wb.scan(scanned_id, &dxl_cnt, range, &log);
  if (result == false)
  {
    Serial.println(log);
    Serial.println("Failed to scan");
  }
  else
  {
    Serial.print("Find ");
    Serial.print(dxl_cnt);
    Serial.println(" Dynamixels");

    for (int cnt = 0; cnt < dxl_cnt; cnt++)
    {
      Serial.print("id : ");
      Serial.print(scanned_id[cnt]);
      Serial.print(" model name : ");
      Serial.println(dxl_wb.getModelName(scanned_id[cnt]));
    }
  }  
}

void loop() 
{

}
 
4. Copy and Paste your error message on terminal

Succeeded to init : 1000000
Wait for scan...
Find 0 Dynamixels

5. Please, describe detailedly what difficulty you are in 

   all the dynamixel-workbench examples do not work on opencm 9.04 with XL320

lolongin, 2019-05-24:
ID_change example,
Succeeded to init : 1000000
[RxPacketError] Hardware error occurred. Check the error at Control Table (Hardware Error Status)!
Failed to ping
[DynamixelWorkbench] Failed to change torque status!

RyanJaehyunShim, 2019-05-27:
@lolongin 

We provide [U2D2](http://emanual.robotis.com/docs/en/parts/interface/u2d2/) and [Wizards2.0](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/) for Dynamixel setup. 

If you have any question, please ask at the link below.
(http://en.robotis.com/service/forum.php)

Thank you,
Ryan

closed 2019-05-27 by RyanJaehyunShim

lolongin, 2019-05-27:
got it

 
 
 (http://en.robotis.com/service/forum.php)