ROBOTIS-GIT/dynamixel-workbench #248: Torque_Enable does not work from yaml file

yoshimalucky, 2019-05-30:
In this yaml file:
https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/master/dynamixel_workbench_controllers/config/basic.yaml#L6

When I write such as:
Torque_Enable : 0

Torque can not be disabled. Therefore, I have to call a service dynamixel_command at initialization in the case of willing to disable motor torque.

LED : 0 works, so motor LED can be turned off.

Why not Torque_Enable? It seems to be a bag.

yoshimalucky, 2019-06-26:
Fixed this issue with this PR #249

RyanJaehyunShim, 2019-07-08:
@yoshimalucky 

As torque is set to be ON when initialising, you cannot set it the yaml file. We noted it on our [eManual](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/#controllers)

Thank you for your contribution and I will close your PR as well. 
Ryan

closed 2019-07-08 by RyanJaehyunShim