ROBOTIS-GIT/dynamixel-workbench #252: can't connect to dynamixel pro plus servo

tomsepe, 2019-06-16:
1. How to setup? (ex, U2D2, OpenCR,...)
U2D2

2. Which Dynamixel have you used? and how many? **(Please describe below format to all connected Dynamixels)**

   - Model Name
 two H42P-020-S300-R and one M42P-010-S260-R  
   - ID 1,3,5

   - Baud Rate of Dynamixels
57600

   - Protocol Version
 2.0

3. Write down the commands you used in order

  ```
  $ roscore    
  $ roslaunch dynamixel_workbench_controllers dynamixel_controllers.launch
  ```
 
4. Copy and Paste your error message on terminal

  ```
thomas@localhost:~/mw_robot/catkin_ws/src$ roslaunch dynamixel_workbench_controllers dynamixel_controllers.launch
... logging to /home/thomas/.ros/log/33f16c22-8fea-11e9-bc7d-6036dd6e4083/roslaunch-localhost-1438.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
started roslaunch server http://localhost:38059/
SUMMARY
========
PARAMETERS
 * /dynamixel_info: /home/thomas/mw_r...
 * /dynamixel_workbench/dxl_read_period: 0.01
 * /dynamixel_workbench/dxl_write_period: 0.01
 * /dynamixel_workbench/mobile_robot_config/radius_of_wheel: 0.033
 * /dynamixel_workbench/mobile_robot_config/seperation_between_wheels: 0.16
 * /dynamixel_workbench/publish_period: 0.01
 * /dynamixel_workbench/use_cmd_vel_topic: False
 * /dynamixel_workbench/use_joint_states_topic: True
 * /dynamixel_workbench/use_moveit: False
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
NODES
  /
    dynamixel_workbench (dynamixel_workbench_controllers/dynamixel_workbench_controllers)
ROS_MASTER_URI=http://localhost:11311
process[dynamixel_workbench-1]: started with pid [1455]
[ INFO] [1560662384.451304869]: Name : joint1, ID : 1, Model Number : 2000
[ INFO] [1560662384.491237307]: Name : joint2, ID : 3, Model Number : 2000
[ INFO] [1560662384.531255364]: Name : joint3, ID : 5, Model Number : 2100
[ERROR] [1560662384.621227034]: [DynamixelDriver] Failed to get the Tool
[ERROR] [1560662384.621296825]: Failed to write value[0] on items[Return_Delay_Time] to Dynamixel[Name : joint3, ID : 5]
[ERROR] [1560662384.621334566]: Please check control table (http://emanual.robotis.com/#control-table)
================================================================================REQUIRED process [dynamixel_workbench-1] has died!
process has finished cleanly
log file: /home/thomas/.ros/log/33f16c22-8fea-11e9-bc7d-6036dd6e4083/dynamixel_workbench-1*.log
Initiating shutdown!
================================================================================
  
5. Please, describe detailedly what difficulty you are in 

  Can't start the controller. seem like it doesn't recognize that model of the  M42P-010-S260-R servo

also when I run find_dynamixel I get:

[ INFO] [1560662622.127288022]: id : 1, model name : PRO-PLUS-H42P-020-S300-R
[ INFO] [1560662622.127328705]: id : 3, model name : PRO-PLUS-H42P-020-S300-R
[ INFO] [1560662622.127356779]: id : 5, model name : (null)

rittersb, 2019-06-17:
@tomsepe I am experiencing a similar issue but with the M54P-040-S250-R. I did a little digging and it doesn't seem that this family is defined in [dynamixel_item.h](https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/master/dynamixel_workbench_toolbox/include/dynamixel_workbench_toolbox/dynamixel_item.h) and [dynamixel_item.cpp](https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/master/dynamixel_workbench_toolbox/src/dynamixel_workbench_toolbox/dynamixel_item.cpp)

tomsepe, 2019-06-17:
@rittersb I was able to add my own definitions to the file and I am able to use my servo now. I can send you the modified files if you like and also I'll make a pull request

rittersb, 2019-06-17:
@tomsepe If you don't mind that would be great. Thanks!

tomsepe, 2019-06-17:
here is a link to the modified dynamixel_workbench_toolbox with added description for the M42P-010-S260. https://www.dropbox.com/sh/hv56be11q3ye92u/AABF121hQgPf3kse8VPE6L-wa?dl=0

you'll still need to add descriptions for your specific servo. if you just do a search for "M42P_010_S260_R" you'll find where I modified.

The modified files are:
dynamixel_item.cpp
dynamixel_tool.cpp

and 

dynamixel_item.h

tomsepe, 2019-06-17:
Specific changes I made:

File: dynamixel_item.cpp

>  static const ModelInfo info_PRO_PLUS[] = {
>       {0.01, -303454, 0, 303454, -3.14159265, 3.14159265},  // PRO_H42P_020_S300_R
>       {0.01, -501433, 0, 501433, -3.14159265, 3.14159265},  // PRO_H54P_100_S500_R
>       {0.01, -501433, 0, 501433, -3.14159265, 3.14159265},  // PRO_H54P_200_S500_R
>       {0.01, -262931, 0, 262931, -3.14159265, 3.14159265}}; // PRO_M42P_010_S260_R

>    else if (num == PRO_PLUS_H42P_020_S300_R || num == PRO_PLUS_H54P_100_S500_R || num == PRO_PLUS_H54P_200_S500_R || num == PRO_PLUS_M42P_010_S260_R)

>    else if (num == PRO_PLUS_M42P_010_S260_R)
>   {
>     info = &info_PRO_PLUS[3];
>   }
> 

File: dynamixel_tool.cpp:
add description to dynamixe_model_table[]:

>     {PRO_PLUS_M42P_010_S260_R, "PRO-PLUS-M42P-010-S260-R"},

File: dynamixel_item.h:
add definition

> #define PRO_PLUS_M42P_010_S260_R  2100

ROBOTIS-Will, 2019-06-18:
@tomsepe , @rittersb Thank you for reporting the issue.
Any missing DYNAMIXEL in DYNAMIXEL Workbench will be updated by the end of this week.

RyanJaehyunShim, 2019-06-18:
@tomsepe @tomsepe 

Pro plus M models have been added to the library. Let us know if you run into any other problems.

I will close this issue,
Ryan

closed 2019-06-18 by RyanJaehyunShim