ROBOTIS-GIT/dynamixel-workbench #264: wheel_operator is not working

yama-kan, 2019-07-18:
Hi,  I tried "Wheel Operators" tutorial at "dynamixel_workbench e-manual" with XM430-W210.

I set the use_cmd_vel parameter of the dynamixel_controllers.lanch file to true but I can't control motor using wasdx key.

1. How to setup? (ex, U2D2, OpenCR,...)

    U2D2

2. Which Dynamixel have you used? and how many? **(Please describe below format to all connected Dynamixels)**

   - Model Name
XM430-W210
   - ID
1,2
   - Baud Rate of Dynamixels
3000000
   - Protocol Version
2.0

3. Write down the commands you used in order

  ```
  $ roscore    
  $ roslaunch dynamixel_workbench_controllers dynamixel_controllers.launch 
  $ roslaunch dynamixel_workbench_operators wheel_operator.launch 
  ```
 
4. Copy and Paste your error message on terminal

  ```
$ roslaunch dynamixel_workbench_operators wheel_operator.launch 
... logging to /home/yamaguchi/.ros/log/0732b16e-a926-11e9-afe3-0c9d9265015b/roslaunch-jroboPC14-22688.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:36235/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /dynamixel_workbench/
    wheel_operator (dynamixel_workbench_operators/wheel_operator)

ROS_MASTER_URI=http://localhost:11311

process[dynamixel_workbench/wheel_operator-1]: started with pid [22705]
[ INFO] [1563431652.467609198]: You can set '-lin_vel_step' and  '-ang_vel_step' arguments (default is 0.01 and 0.1)
[ INFO] [1563431652.468159249]: 
  Control Your Mobile Robot! 
  --------------------------- 
  Moving around:
          w
     a    s    d
          x
  
  w/x : increase/decrease linear velocity
  a/d : increase/decrease angular velocity
  
  s : force stop
  
  CTRL-C to quit
  

  ```
  
5. Please, describe detailedly what difficulty you are in 

    - Wheel Operators is not working.

yoshimalucky, 2019-09-11:
@yama-kan Your servo motors have non-default value of 57600 for baud rate.

try:
roslaunch dynamixel_workbench_controllers dynamixel_controllers.launch dxl_baud_rate:=3000000

Or, modify here:
https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/master/dynamixel_workbench_controllers/launch/dynamixel_controllers.launch#L3

rjshim, 2020-01-08:
Since there have been no updates, I will close this issue.

closed 2020-01-08 by rjshim