ROBOTIS-GIT/dynamixel-workbench #265: Unable to address an XM430-W210 motor with ID 32

bholmes451, 2019-07-31:
ISSUE TEMPLATE ver. 1.0.0

Before you open issue, please refer to [ROBOTIS e-Manual](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/)

1. How to setup? (ex, U2D2, OpenCR,...)
Using an OpenCM board and the dynamixel workbench. We set the ID using the latest Dynamixel Wizard 2.0 and can successfully control and communicate with it with the Wizard software.

2. Which Dynamixel have you used? and how many? **(Please describe below format to all connected Dynamixels)**

   - Model Name
        XM430-W210
   - ID
       32
   - Baud Rate of Dynamixels
       1000000
   - Protocol Version
       2.0

3. Write down the commands you used in order

 
4. Copy and Paste your error message on terminal
We are using Arduino, not ROS.
  
5. Please, describe detailedly what difficulty you are in 

   We can successfully use IDs 1-31 with no issues but as soon as we attempt to communicate with ID 32 it returns an "Unable to find tool" error. We are able to see it as part of the scan.

rjshim, 2020-01-09:
@bholmes451 

Would you provide us with more information (e.g., terminal command input or error code)? Also, as you changed the baudrate of your motor, test it again with adding dxl_baud_rate:=1000000 at the end and see if it works. 

`$ roslaunch dynamixel_workbench_controllers dynamixel_controllers.launch dxl_baud_rate:=1000000`

Ryan

JaehyunShim, 2020-10-16:
I will close this thread as it has been no recent updates.

closed 2020-10-16 by JaehyunShim