ROBOTIS-GIT/dynamixel-workbench #272: <DynamixelWorkbench.h> with <ros.h> serial port issues

domiinio, 2019-09-22:
Hello everyone,
I'm trying to controll a Dynamixel ax-12a servo using the OpenCM9.04 board.
For programming I'm using Arduino IDE.

I was trying out the Dynamixel Workbench library for the servo control, and it was working fine, but since I'll be needing ROS in my project later on, I tried to install the rosserial library and when I include the library into a single example h_Position, 
...
#include <DynamixelWorkbench.h>
#include <ros.h>
.....
I get the following error and I don't know how to get rid of it:

 _In file included from /home/dominik/Arduino/libraries/Rosserial_Arduino_Library/src/ros.h:39:0,
                 from /tmp/arduino_modified_sketch_218835/h_Position.ino:20:
/home/dominik/Arduino/libraries/Rosserial_Arduino_Library/src/ArduinoHardware.h: In constructor 'ArduinoHardware::ArduinoHardware()':
/home/dominik/Arduino/libraries/Rosserial_Arduino_Library/src/ArduinoHardware.h:55:16: error: cannot convert 'USBSerial*' to 'HardwareSerial*' in assignment
       iostream = &Serial;_

If anyone has an Idea, what could be the cause of this error, I'd be more than happy to hear it.
Thanks in advance

rjshim, 2020-01-08:
@domiinio 

Can you just download this [ROS package](https://github.com/ROBOTIS-GIT/dynamixel-workbench) and test it instead of adding the ros.h header to your source code. 

Ryan

JaehyunShim, 2020-10-16:
I will close this thread as it has been no recent updates.

closed 2020-10-16 by JaehyunShim