ROBOTIS-GIT/dynamixel-workbench #273: ROS tutorial Find Dynamixel (Question)

juanjqo, 2019-10-08:
ISSUE TEMPLATE ver. 1.0.0

Before you open issue, please refer to [ROBOTIS e-Manual](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/)

1. How to setup? (ex, U2D2, OpenCR,...)

2. Which Dynamixel have you used? and how many? **(Please describe below format to all connected Dynamixels)**
I am using one MX-64 dynamixel motor.

   - Model Name
     MX-64
   - ID 
   24
   - Baud Rate of Dynamixels
   1000000
   - Protocol Version
   1.0
3. Write down the commands you used in order

  ```
  $ roscore    
  $ rosrun dynamixel_workbench_controllers find_dynamixel /dev/ttyACM0
  ```
 
4. Copy and Paste your error message on terminal

  ```
 rosrun dynamixel_workbench_controllers find_dynamixel /dev/ttyACM0
[ INFO] [1570558208.359211508]: Succeed to init(9600)
[ INFO] [1570558208.359240606]: Wait for scanning...
[ INFO] [1570558218.544527907]: Find 1 Dynamixels
[ INFO] [1570558218.544544379]: id : 253, model name : (null)
[ INFO] [1570558218.545097372]: Succeed to init(57600)
[ INFO] [1570558218.545107961]: Wait for scanning...
[ INFO] [1570558227.412740896]: Find 1 Dynamixels
[ INFO] [1570558227.412756887]: id : 253, model name : (null)
[ INFO] [1570558227.413323749]: Succeed to init(115200)
[ INFO] [1570558227.413334547]: Wait for scanning...
[ INFO] [1570558236.149227454]: Find 1 Dynamixels
[ INFO] [1570558236.149243353]: id : 253, model name : (null)
[ INFO] [1570558236.149829648]: Succeed to init(1000000)
[ INFO] [1570558236.149836589]: Wait for scanning...
[ INFO] [1570558244.735730700]: Find 2 Dynamixels
[ INFO] [1570558244.735746436]: id : 23, model name : MX-64
[ INFO] [1570558244.735751673]: id : 253, model name : (null)
[ INFO] [1570558244.736335895]: Succeed to init(2000000)
[ INFO] [1570558244.736341872]: Wait for scanning...
[ INFO] [1570558253.347920265]: Find 1 Dynamixels
[ INFO] [1570558253.347933172]: id : 253, model name : (null)
[ INFO] [1570558253.348494113]: Succeed to init(3000000)
[ INFO] [1570558253.348504647]: Wait for scanning...
[ INFO] [1570558261.957616396]: Find 1 Dynamixels
[ INFO] [1570558261.957631975]: id : 253, model name : (null)
[ INFO] [1570558261.958306302]: Succeed to init(4000000)
[ INFO] [1570558261.958317212]: Wait for scanning...
[ INFO] [1570558270.565982163]: Find 1 Dynamixels
[ INFO] [1570558270.565997972]: id : 253, model name : (null)

  
  ```
  
5. Please, describe detailedly what difficulty you are in 

    - In the [tutorial](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/) the output was:

```
[ INFO] [1544589715.841211668]: Succeed to init(9600)
[ INFO] [1544589715.841236741]: Wait for scanning...
[ INFO] [1544589737.539083688]: Find 0 Dynamixels
[ INFO] [1544589737.539526809]: Succeed to init(57600)
[ INFO] [1544589737.539570059]: Wait for scanning...
[ INFO] [1544589755.441019922]: Find 2 Dynamixels
[ INFO] [1544589755.441086482]: id : 1, model name : XM430-W350
[ INFO] [1544589755.441109032]: id : 2, model name : XM430-W350
[ INFO] [1544589755.441504892]: Succeed to init(115200)
[ INFO] [1544589755.441548969]: Wait for scanning...
[ INFO] [1544589773.031677244]: Find 0 Dynamixels
[ INFO] [1544589773.032153380]: Succeed to init(1000000)
[ INFO] [1544589773.032178580]: Wait for scanning...
[ INFO] [1544589790.291943770]: Find 0 Dynamixels
[ INFO] [1544589790.292404604]: Succeed to init(2000000)
[ INFO] [1544589790.292418207]: Wait for scanning...
[ INFO] [1544589807.530702991]: Find 0 Dynamixels
[ INFO] [1544589807.531286252]: Succeed to init(3000000)
[ INFO] [1544589807.531331656]: Wait for scanning...
[ INFO] [1544589824.762803705]: Find 0 Dynamixels
[ INFO] [1544589824.763461821]: Succeed to init(4000000)
[ INFO] [1544589824.763506935]: Wait for scanning...
[ INFO] [1544589841.990120553]: Find 0 Dynamixels
```
This is, Find 0 motors when the baudrate is different to the dynamixel baudrate.  However, in my case, always find one motor in those cases. This is expected?

In other words, what it means the following? (Case when the baudrate is different to the dynamixel baurate)
```
[ INFO] [1570558270.565982163]: Find 1 Dynamixels
[ INFO] [1570558270.565997972]: id : 253, model name : (null)
```

why detect two motors when the baud is the same to the dynamixel baudrate ( There is only one dynamixel motor connected)
```
[ INFO] [1570558244.735730700]: Find 2 Dynamixels
[ INFO] [1570558244.735746436]: id : 23, model name : MX-64
[ INFO] [1570558244.735751673]: id : 253, model name : (null)
```

Cheers, 

Juan

rjshim, 2020-01-09:
@juanjqo 

[image]

It worked without any issues for me. Try to change the motor id or baudrate with [Dyanmixel Wizard](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/) and test it again. If it still does not work, the problem might be on your motor. 

Ryan

nathangeorge1, 2020-02-06:
I'm having the same problem, the model name is "(null)" which is preventing node "dynamixel_workbench_controllers" from working  

```
selfsub@selfsub-1:~/catkin_ws/src/selfsub$ rosrun dynamixel_workbench_controllers find_dynamixel /dev/ttyUSB0
[ INFO] [1581024627.363922489]: Succeed to init(9600)
[ INFO] [1581024627.364008089]: Wait for scanning...
[ INFO] [1581024649.145637970]: Find 0 Dynamixels
[ INFO] [1581024649.155095478]: Succeed to init(57600)
[ INFO] [1581024649.155132437]: Wait for scanning...
[ INFO] [1581024667.186529892]: Find 0 Dynamixels
[ INFO] [1581024667.196587945]: Succeed to init(115200)
[ INFO] [1581024667.196626237]: Wait for scanning...
[ INFO] [1581024684.855150641]: Find 0 Dynamixels
[ INFO] [1581024684.866384056]: Succeed to init(1000000)
[ INFO] [1581024684.866421115]: Wait for scanning...
[ INFO] [1581024693.513519518]: Find 1 Dynamixels
[ INFO] [1581024693.513544444]: id : 1, model name : (null)
[ INFO] [1581024693.525412708]: Succeed to init(2000000)
[ INFO] [1581024693.525436993]: Wait for scanning...
```

It's outputting this instead:

```
process[dynamixel_workbench-1]: started with pid [14591]
[ INFO] [1581024702.304939238]: Name : pan, ID : 1, Model Number : 360
[ERROR] [1581024702.306275812]: [DynamixelDriver] Failed to get the Tool
[ERROR] [1581024702.306298575]: Failed to set Dynamixel SDK Handler
================================================================================REQUIRED process [dynamixel_workbench-1] has died!
process has finished cleanly
log file: /home/selfsub/.ros/log/ec55591c-4922-11ea-a93d-0cdd24c1d0ae/dynamixel_workbench-1*.log
Initiating shutdown!
================================================================================

```

I tried setting the ID to 5 but it didn't seem to have any effect. I was able to get the motor moving in Dynamixel Wizard 2.

[image]

@rjshim How do I get my MX-12W to be recognized by the node "find_dynamixel"?

ROBOTIS-Will, 2020-03-18:
@nathangeorge1 
Could you check which DYNAMIXEL Workbench branch you are on?
If you've been using the `kinetic-devel` branch, please try with the `mater` branch and let us know.

nathangeorge1, 2020-03-18:
Hi @ROBOTIS-Will ,

I used apt to install on Kinetic:

`sudo apt-get install ros-kinetic-dynamixel-sdk ros-kinetic-dynamixel-workbench-controllers`

I didn't build from source. Should I try building from source or upgrading to Melodic?

nathangeorge1, 2020-03-19:
@ROBOTIS-Will an update,

I actually tested this motor through the U2D2 on both primary boot Ubuntu 18.04 Melodic and VirtualBox Ubuntu 16.04.2 Kinetic. Both give the same null error.

rjshim, 2020-03-19:
@nathangeorge1 

Click this [link](https://github.com/ROBOTIS-GIT/dynamixel-workbench) then you will see the green click and download button in the top right corner. Click it and download the source package and catkin_make it with your own packages together. 

Test it and let us know if you still see the same error code.

Ryan

nathangeorge1, 2020-03-21:
It worked! I'm able to find the name.

[image]

I'm on a VirtualBox Melodic now. Is there a way to get it working using the apt installer?

I'm also having another problem with the tutorial, but I'll open up another issue for that.

ROBOTIS-Will, 2020-03-23:
@nathangeorge1 
Thank you for the feedback. The binary installer has not been updated for a while so it is recommended to use the latest source directly from GitHub using below command until we update the binary installation files.
```
$ git clone https://github.com/ROBOTIS-GIT/dynamixel-workbench.git
```

nathangeorge1, 2020-03-23:
Sounds good; will do that in the mean time

ROBOTIS-Will, 2020-05-08:
This issue will be closed as there wasn't any recent activity. Please feel free to reopen when needed.

closed 2020-05-08 by ROBOTIS-Will