ROBOTIS-GIT/dynamixel-workbench #274: Read and Write "Homing_Offset" with OpenCM9.04c and XM430-W350 T

tj60647, 2019-11-01:
ISSUE TEMPLATE ver. 1.0.0

Before you open issue, please refer to [ROBOTIS e-Manual](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/)

1. How to setup? (ex, U2D2, OpenCR,...)

I am using OpenCM9.04c + Arduino IDE

2. Which Dynamixel have you used? and how many? **(Please describe below format to all connected Dynamixels)**

In this setup there is a single: 
   - Model Name:  XM430-W350 T
   - ID:   1
   - Baud Rate of Dynamixels:    57600
   - Protocol Version:  2.0
 
3. Write down the commands you used in order:  The following is the code being tested for reading and writing of "Homing_Offset"

#include <DynamixelWorkbench.h>

#define DEVICE_NAME "1"
#define BAUDRATE  57600
#define DXL_ID    1

DynamixelWorkbench dxl_wb;

int32_t homing_offset = 1000;//test value
int32_t operating_mode = 0;
int32_t present_position = 0;

void setup()
{
  Serial.begin(57600);

  dxl_wb.begin(DEVICE_NAME, BAUDRATE);
  dxl_wb.ping(DXL_ID);
  dxl_wb.itemWrite(DXL_ID, "Torque_Enable", 0);
  dxl_wb.itemWrite(DXL_ID, "Operating_Mode", 4);//extended position
  dxl_wb.itemWrite(DXL_ID, "Homing_Offset", homing_offset);
  dxl_wb.itemWrite(DXL_ID, "Torque_Enable", 1);
  dxl_wb.ping(DXL_ID);

  delay(1000);
  dxl_wb.itemWrite(DXL_ID, "Torque_Enable", 0);//set to move freely
}

int timer = 0;
void loop()
{
  operating_mode = dxl_wb.itemRead(DXL_ID, "Operating_Mode");
  present_position = dxl_wb.itemRead(DXL_ID, "Present_Position");
  homing_offset = dxl_wb.itemRead(DXL_ID, "Homing_Offset");

  Serial.println("[ ID :"     + String(DXL_ID)                +
                 " OpMode :" + String(operating_mode)         +
                 " PresPos :" + String(present_position)      +
                 " HoOff :" + String(homing_offset)           +
                 " ]");
  delay(500);

  timer++;

  //every 10 seconds change Homing_Offset to new value
  if(timer > 10)
  {
    dxl_wb.itemWrite(DXL_ID, "Homing_Offset", 0);
    present_position = dxl_wb.itemRead(DXL_ID, "Present_Position");
    Serial.println("update Homing_Offset: " + String(present_position));
    dxl_wb.itemWrite(DXL_ID, "Homing_Offset", present_position);
    timer = 0;
  }
}
 
4. Copy and Paste your error message on terminal
  The error is that the Homing_Offset does not change.

5. Please, describe in detail what difficulty you are in 

How do I Read and Write to "Homing_Offset" with OpenCM9.04c + Arduino IDE?

Neither:
    -  dxl_wb.itemWrite(DXL_ID, "Homing_Offset", homing_offset);
nor
    -  homing_offset = dxl_wb.itemRead(DXL_ID, "Homing_Offset");
appear to work.

I can set the homing offset using Dynamixel Wizard 2.0.
I can read the homing offset using Dynamixel Wizard 2.0.
Reading and writing other parameters seems to work fine:  Torque_Enable, Operating_Mode, Present_Position, Goal_Position.

rjshim, 2020-01-09:
@tj60647 

Any parameters on the control table of EEPROM area (including homing offset) cannot be changed while torque is on. If you have more questions about Dynamixel, leave them at the [link](http://en.robotis.com/model/board.php?bo_table=robotis_faq_en).

tj60647, 2020-02-19:
Correct.

Doesn't...

dxl_wb.itemWrite(DXL_ID, "Torque_Enable", 0);

...disable torque?  

Also,  in the code above writing to "Operating_Mode" works.

Thanks for your help.

JaehyunShim, 2020-10-16:
Please use Dynamixel-SDK instead of this package. We do not support this repository anymore. 

Link: https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/overview/

closed 2020-10-16 by JaehyunShim