ROBOTIS-GIT/dynamixel-workbench #275: Could not file the dynamixel moter using the package?

ipa-mah, 2019-11-04:
ISSUE TEMPLATE ver. 1.0.0

Before you open issue, please refer to [ROBOTIS e-Manual](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/)

1. How to setup? (ex, U2D2, OpenCR,...)

2. Which Dynamixel have you used? and how many? **(Please describe below format to all connected Dynamixels)**

   - Model Name: MX-12W, AX-12W, CM-530 controller board

   - ID

   - Baud Rate of Dynamixels

   - Protocol Version
 
3. Write down the commands you used in order

  ```  
  $ rosrun dynamixel_workbench_controllers find_dynamixel /dev/ttyUSB0
  ```
 
4. Copy and Paste your error message on terminal

  ```
  SUMMARY
  ========

  PARAMETERS
   * /baud_rate: 57600
   * /device_name: /dev/ttyUSB0
   * /ping: False
   * /ping_id: 1
   * /rosdistro: melodic
   * /rosversion: <unversioned>

   * /scan_range: 10

  NODES
    /
      single_dynamixel_controller (dynamixel_workbench_single_manager/single_dynamixel_controller)
      single_dynamixel_monitor (dynamixel_workbench_single_manager/single_dynamixel_monitor)

  auto-starting new master
  process[master]: started with pid [8594]
  ROS_MASTER_URI=http://192.168.1.3:11311

  setting /run_id to de3bd71e-728e-11e8-9f66-d8cb8af3d300
  process[rosout-1]: started with pid [8607]
  started core service [/rosout]
  process[single_dynamixel_monitor-2]: started with pid [8618]
  process[single_dynamixel_controller-3]: started with pid [8626]
  [PortHandlerLinux::SetupPort] Error opening serial port!
  ================================================================================REQUIRED process [single_dynamixel_monitor-2] has died!
  process has died [pid 8618, exit code -11, cmd /home/darby/catkin_ws/devel/lib/dynamixel_workbench_single_manager/single_dynamixel_monitor __name:=single_dynamixel_monitor __log:=/home/darby/.ros/log/de3bd71e-728e-11e8-9f66-d8cb8af3d300/single_dynamixel_monitor-2.log].
  log file: /home/darby/.ros/log/de3bd71e-728e-11e8-9f66-d8cb8af3d300/single_dynamixel_monitor-2*.log
  Initiating shutdown!
  ================================================================================
  [single_dynamixel_controller-3] killing on exit
  [single_dynamixel_monitor-2] killing on exit
  [rosout-1] killing on exit
  [master] killing on exit
  shutting down processing monitor...
  ... shutting down processing monitor complete
  ```
  
5. Please, describe detailedly what difficulty you are in 

    - I connect 2 dynamixel motors to the controller CM-530 and use the package to find the dynamixel motors. The attached file shows no dynamixel connected?
Do you have any idea to figure out the problem?
Thank you
 
[image]

rjshim, 2020-01-09:
@ipa-mah 

You have to set up motor id numbers first with [Dynamixel Wizard](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/) (e.g., 1 for the first motor and 2 for the second motor). Motors won't be detected if they have the same id number. 

Ryan

JaehyunShim, 2020-10-16:
I will close this thread as there have been no updates.

closed 2020-10-16 by JaehyunShim