ROBOTIS-GIT/dynamixel-workbench #278: 2XL430-W250-T - Able to ping but not to write...

swinterbotix, 2019-12-11:
1. Setup was done using a U2D2

2. Which Dynamixel have you used? and how many? **(Please describe below format to all connected Dynamixels)**

   - Model Name: 2XL430-W250-T (a.k.a 1090)

   - ID: 8 and 9

   - Baud Rate of Dynamixels: 1000000

   - Protocol Version: 2.0
 
3. Write down the commands you used in order

Using the 'itemWrite' function that is part of the dynamixel_toolbox.
Specifically... 

const char* log;
itemWrite(8, "Drive_Mode", 0, &log);
 
4. Copy and Paste your error message on terminal

[DynamixelDriver] Failed to get the Tool
Failed to write value[0] on items[Drive_Mode] to [ID : 8]

5. Please, describe detailedly what difficulty you are in 

    - If I try using the itemWrite function to write to the servo (model shown above), I get the above error when there should be no error. Note that I am able to ping the motor successfully - but seem to have trouble writing to the registers with the itemWrite function. Also note that when using two separate XL430 servos, the same code works just fine.

ROBOTIS-Will, 2020-01-03:
@swinterbotix 
Thank you for reporting the issue. I'll test and look into the code and get back to you.

ROBOTIS-Will, 2020-01-03:
@swinterbotix 
I've just tested with 2XL430-W250-T(ID 1 & 2, 1Mbps) + U2D2(/dev/ttyUSB0) on Linux.
I modified the example dynamixel_workbench_toolbox/examples/src/k_Read_Write.cpp to change the "Drive_Mode" instead of turning on the LED.
After running the code for ID 1 and 2, I checked the modified value from DYNAMIXEL Wizard 2.0 without any problem.
In case of 2XL430-W250-T, two actuators are connected internally, therefore, if one of them is in Torque On state, the other will also be in the same status that will not accept any change in EEPROM.
Please try changing the Drive_Mode under Torque Off state.
Thank you.

swinterbotix, 2020-01-03:
@ROBOTIS-Will 
I made sure to torque off both actuators before writing to the EEPROM, but I'm still getting the error message. I'll try running the example cpp file you mentioned though and see if I have any luck with that.

ROBOTIS-Will, 2020-01-03:
@swinterbotix 
Below is the source code that I modified and used to test the DYNAMIXEL.
[k_Read_Write.cpp.zip](https://github.com/ROBOTIS-GIT/dynamixel-workbench/files/4018092/k_Read_Write.cpp.zip)

swinterbotix, 2020-01-05:
Hi @ROBOTIS-Will,

I downloaded your zip file and tested it out. As you mentioned, the code was able to run successfully. However, that was only when running the *master* branch of the dynamixel_workbench repository. Afterwards, I switched to the *kinetic-devel* branch (which I assume contains the current version of the code available via apt-get) and recompiled the code in the *dynamixel_workbench*. When running the same cpp file, I got an error message ONLY when working with the 2XL430-W250-T servos. The other servos had no issue. Also, with the *kinetic-devel* branch code, the pinging was successful for the 2XL430-W250_T servos - but read/writing resulted in the error message 'Failed to get the Tool'. So it looks like you might have to reupdate the 'dynamixel_workbench_toolbox' code available on apt-get to your most recent version.

rjshim, 2020-01-08:
@swinterbotix 

Dynamixel workbench does not support the 2xl series. Please use [DynamixelSDK](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/overview/)

Regards,
Ryan

swinterbotix, 2020-01-08:
@rjshim,

I'm a bit confused. The Dynamixel workbench seems to work with the 2xl series assuming ROS is not used. Are you saying that the dynamixel_workbench ROS metapackage does not support the 2xl series?

rjshim, 2020-01-09:
@swinterbotix 

The 2XL and 2XC models are not compatible with the current version of DXLWB. It might be updated in the future but we do not have any concrete plans about it. 

Ryan

ROBOTIS-Will, 2020-01-09:
@swinterbotix 
We apologize that some of the recent DYNAMIXEL had not been updated to the Kinetic branch of DYNAMIXEL Workbench.
As master branch already supports 2XL and recent products, it will soon be merged into Kinetic branch as well.
Once it is completed, we'll give you a notification.
Thank you very much for your patience.

swinterbotix, 2020-01-09:
Thank you! I really appreciate this! Do you think it would be possible to also merge the master branch into the Melodic one as well? My company plans to use the 2XL servos in a new line of robot platforms and the sooner we can get working ROS wrappers for the melodic version of the dynamixel workbench, the sooner we can complete development.

Thanks,
Solomon

ROBOTIS-Will, 2020-01-10:
@swinterbotix 
I believe merging the source code into the Melodic wouldn't be a much trouble once we complete the Kinetic version. However, due to the characteristics of 2XL hardware configuration, some features may not work as other single axle DYNAMIXEL products (such as modifying EEPROM of ID A while ID B is in Torque on status).
Thank you.

swinterbotix, 2020-01-10:
That's fine. I can always check for that in my code

closed 2025-07-08 by yun-goon