ROBOTIS-GIT/dynamixel-workbench #280: [workbench_toolbox] DynamixelWorkbench::getPresentVelocityData() reads wrong item

yoshito-okada, 2019-12-24:
I believe the line below should read the "Present_Velocity" item, instead of "Goal_Velocity".
https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/646b229955e413f9e805b94859219cfd72b543b0/dynamixel_workbench_toolbox/src/dynamixel_workbench_toolbox/dynamixel_workbench.cpp#L1132

rjshim, 2020-01-09:
@yoshito-okada 

Thank you for the report. I just fixed it and it will be updated soon. 

Thank you, 
Ryan

closed 2020-01-09 by rjshim

yoshito-okada, 2020-11-16:
@JaehyunShim 
This issue fixed wrongly.  What to read is "Present_Velocity", not "Present_Speed" !
https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/455fb5a02f03a06cca6e3cd6b0f3196a8072030e/dynamixel_workbench_toolbox/src/dynamixel_workbench_toolbox/dynamixel_workbench.cpp#L1151

yoshito-okada, 2020-12-16:
@ROBOTIS-Will 
Can I ask to re-open this issue because not yet fixed?