ROBOTIS-GIT/dynamixel-workbench #290: Possible bug when connecting AX with MX

dhiegomaga, 2020-02-12:
_1. How to setup? (ex, U2D2, OpenCR,...)_

I'm using the standard usb interface and connecting an MX-64 to a AX-12, both powered to 12V, and controlled via ROS on Ubuntu 18.04, using the package **dynamixel_workbench_controllers**. 

Following is the YAML

```
ax12:
    ID: 4
    Return_Delay_Time: 250

mx64:
    ID: 2
    Return_Delay_Time: 250
    P_gain: 32
    I_gain: 0
    D_gain: 0
```

_2. Which Dynamixel have you used? and how many?_

   - Model Name: MX-64 and AX-12

   - ID: ?

   - Baud Rate of Dynamixels: 1M

   - Protocol Version: 1.0
 
_My issue_

I want to use both AX-12 and MX-64 on the same setup, sharing the same cascading network architecture. I had to **downgrade** the MX to protocol version 1.0, since I wasn't able to use both on the same connection with different protocols (got some errors), AND at the time of this writing, AX-12 doesn't support protocol 1.0.

When I use them separately it works fine, but when they are both connected and I send some angle near 3 or -3 rads, they behave weird: 

[image]

Notice how I sent the command to go from -3 to 3, but the MX keeps "flickering" back and fourth (for ever), without ever converging and stopping. To be clear, this does not happen when the AX-12 is not connected to the MX-64.

ROBOTIS-Will, 2020-04-08:
@dhiegomaga 
I'm sorry for the delayed response.
While we are looking into this issue, please try with Protocol 2.0 of MX-64AT.
You can actually set MX-64T with Protocol 2.0 to use Protocol 1.0.
Please see below control table where allows you to select a specific Protocol version to communicate.
[http://emanual.robotis.com/docs/en/dxl/mx/mx-64-2/#protocol-type13](http://emanual.robotis.com/docs/en/dxl/mx/mx-64-2/#protocol-type13)
Thank you.

yun-goon, 2025-01-13:
This issue appears to be resolved, so it will be closed.

closed 2025-01-13 by yun-goon