ROBOTIS-GIT/dynamixel-workbench #294: Building dynamixel_workbench_toolbox without ROS on a Raspberry Pi

swinterbotix, 2020-05-05:
While following the instructions to build the dynamixel_workbench_toolbox package without ROS on a Raspberry Pi, I ran into the following error (while running the 'make' command).

[image]

After some debugging, I found that if I changed [line 20](https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/master/dynamixel_workbench_toolbox/examples/CMakeLists.txt#L20) in the CMakeLists.txt file to read...

`target_link_libraries(dynamixel_workbench LINK_PUBLIC "/usr/local/lib/libdxl_sbc_cpp.so")`

... then the 'make' command worked. 

I'm not exactly sure how it should be done, but I believe that the 'if condition' block should check to see not just if the code is being built on a MAC or x64 Linux system, but also a SBC system (like the raspberry Pi).

eisulimma, 2024-06-04:
Hey I had the same issue but your solution didn't work for me. I get the same error but with a different name:
make[2]: *** No rule to make target '/usr/local/lib/libdxl_sbc_cpp.so', needed by 'find_dynamixel'. Stop.
make[1]: *** [CMakeFiles/Makefile2:108: CMakeFiles/find_dynamixel.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

Please help!
Thank you !!

yun-goon, 2025-01-21:
Hi @swinterbotix 
Sorry for the delayed response.
The issue seems to be caused by a whitespace on line 56 of the c++/build/linux_sbc/Makefile, which resulted in the make command failing. This has been fixed in [pull request #634](https://github.com/ROBOTIS-GIT/DynamixelSDK/pull/634).

Additionally, you can modify the CMakeLists.txt file to ensure it builds without issues on SBCs (like the Raspberry Pi). Add the following line at the specified location:
https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/31f1c97ef612f7daa810a8ed67debf6677a0f281/dynamixel_workbench_toolbox/examples/CMakeLists.txt#L19

Thank you for raising the issue!

yun-goon, 2025-03-14:
This feature has been updated in the Humble branch!

Please try it out, and if you encounter any issues, feel free to reopen the issue.

Thank you!

closed 2025-03-14 by yun-goon