ROBOTIS-GIT/dynamixel-workbench #298: Linux64 build: error: atoi was not find in this scope

cyphunk, 2020-06-18:
- I'm using `DynamixelSDK git, ver. 3.7.21 or greater`
- I'm using `C++` Language
- I'm using `USB2Dynamixel2` serial converter
- I'm using `XM430-W250-T`
- and I'm having an `issue` like what

Error on compile:

```
[ 16%] Building CXX object CMakeFiles/find_dynamixel.dir/src/o_Find_Dynamixel.cpp.o
[ 18%] Linking CXX executable find_dynamixel
[ 18%] Built target find_dynamixel
Scanning dependencies of target sync_read_write
[ 21%] Building CXX object CMakeFiles/sync_read_write.dir/src/m_Sync_Read_Write.cpp.o
/motors/dynamixel/U2D2_SMPS_XL430/dynamixel-workbench/dynamixel_workbench_toolbox/examples/src/m_Sync_Read_Write.cpp: In function ‘int main(int, char**)’:
/motors/dynamixel/U2D2_SMPS_XL430/dynamixel-workbench/dynamixel_workbench_toolbox/examples/src/m_Sync_Read_Write.cpp:39:17: error: ‘atoi’ was not declared in this scope
39 |     baud_rate = atoi(argv[2]);
|                 ^~~~
/motors/dynamixel/U2D2_SMPS_XL430/dynamixel-workbench/dynamixel_workbench_toolbox/examples/src/m_Sync_Read_Write.cpp:134:12: error: ‘abs’ was not declared in this scope
134 |     }while(abs(goal_position[0] - present_position[0]) > 15 &&
|            ^~~
make[2]: *** [CMakeFiles/sync_read_write.dir/build.make:83: CMakeFiles/sync_read_write.dir/src/m_Sync_Read_Write.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:155: CMakeFiles/sync_read_write.dir/all] Error 2
make: *** [Makefile:104: all] Error 2
```

This is on archlinux, cmake version 3.17.3, gcc (GCC) 10.1.0

cyphunk, 2020-06-18:
I was able to resolve this by adding the following to all source files:
```
#include <iostream>
#include <cstring>
#include <cstdlib>
using namespace std;
```
I doubt this is the smartest solution but in-case someone runs into the same issue, this should work

yun-goon, 2025-02-03:
Apologies for the delayed response. We kindly ask that you try again with the latest version. If the same issue arises, please feel free to resubmit it, and we will address it promptly.

closed 2025-02-03 by yun-goon

AizarHafizhS, 2025-02-12:
@yun-goon I still get same issue today. Fixed it using @cyphunk  solution stated above