ROBOTIS-GIT/dynamixel-workbench #301: No rule to make target x64

01binary, 2020-06-30:
Hello,

I am following this tutorial to build dynamixel workbench on Raspberry Pi 4 (Raspbian) and ROS Melodic:
https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/#linux-and-macos

Getting this error after running "make":

```
No rule to make target '/usr/local/lib/libdxl_x64_cpp.iso'
```

I went into `/lib` and all the output files I have are for the `sbc` platform because I am on Pi. Why is it trying to make it for x64 if I am on SBC?

```
libdxl_sbc_cpp.so
libdxl_sbc_cpp.so.2
libdxl_sbc_cpp.so.2.0
libdxl_sbc_cpp.so.2.0.0
```

Thank you!

01binary, 2020-06-30:
Oh, got it. `CMakeLists.txt` in `/dynamixel_workbench/dynamixel_workbench_toolbox/examples` has this statement:

```
if (APPLE)
 target_link_libraries(....libdxl_mac_cpp.dylib")
else()
 target_link_libraries(...libdxl_linux64_cpp.so")
endif()
```

I changed the 64-bit instruction to `sbc` and now it builds on raspberri pi. That must be why it built on mac and linux VM with no issue.

01binary, 2020-07-01:
Looks like this is the same as https://github.com/ROBOTIS-GIT/dynamixel-workbench/issues/294

eisulimma, 2024-06-04:
Hey I had the same issue but your solution didn't work for me. I get the same error but with a different name: 
make[2]: *** No rule to make target '/usr/local/lib/libdxl_sbc_cpp.so', needed by 'find_dynamixel'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:108: CMakeFiles/find_dynamixel.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

Please help! 
Thank you !!

01binary, 2025-01-21:
@eisulimma can you verify the fix worked for you by chance so this GitHub issue can be closed cleanly?

I moved on to using NUCs which run ordinary Ubuntu desktop, and no longer using Raspberry Pi for robotics.

yun-goon, 2025-03-14:
This feature has been updated in the Humble branch!

Please try it out, and if you encounter any issues, feel free to reopen the issue.

Thank you!

closed 2025-03-14 by yun-goon