ROBOTIS-GIT/dynamixel-workbench #305: Increase Number of Dynamixels

swinterbotix, 2020-08-12:
Hi,

Recently, I've been during some development on a robot project that consists of 36 XM540-W270 servos. Unfortunately, I kept running into an error at startup that the dynamixel_workbench_ toolbox 'failed to get the tool'. It took a few hours, but I located the source of the issue being with a hard-coded limit to the number of servos that the dynamixel_workbench_toolbox can handle. As specified at https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/master/dynamixel_workbench_toolbox/include/dynamixel_workbench_toolbox/dynamixel_tool.h#L30, the max limit is set to '30'. Do you think you could bump up this number to 36? After manually changing this number, I had no issues running my code.

Thanks

swinterbotix, 2020-10-21:
Any update on this?

yun-goon, 2025-01-22:
Hi @swinterbotix 
Sorry for the delayed response.

Although there are no previous records addressing this specific matter, based on the information that OpenCR’s maximum power output should not exceed approximately 55W (as noted in [OpenCR Manual](https://emanual.robotis.com/docs/en/parts/controller/opencr10/#specifications)), it seems that the limit of 30 devices was set to ensure system stability.

I have also tested with 35 devices connected, and I found no issues as long as the power output limits are maintained.

Thank you.

closed 2025-02-03 by yun-goon