ROBOTIS-GIT/dynamixel-workbench #307: itemRead function does not work after Motor Error...

swinterbotix, 2020-08-27:
1. How to setup? U2D2

2. Which Dynamixel have you used? and how many? **(Please describe below format to all connected Dynamixels)**

   - Model Name: Combination of XL430 and XM430 servos (7 in total)

   - ID: 1 - 7

   - Baud Rate of Dynamixels: 1 Mbps

   - Protocol Version: 2.0
 
3. Write down the commands you used in order: 

```
const char *log;
int32_t value = 0;
bool success = dxl_wb.itemRead(7, 'Hardware_Error_Status', &value, &log);
if (!success)
    ROS_ERROR("%s", log);
```

4. Copy and Paste your error message on terminal

`[RxPacketError] Hardware error occurred. Check the error at Control Table (Hardware Error Status)!`
  
5. Please, describe detailedly what difficulty you are in 

I'm purposely putting the motor with ID 7 into an error state (by overloading it), and then trying to read the Hardware_Error_Status register using the 'itemRead' function in the dynamixel_workbench_toolbox. Unfortunately, I always recieve the above error message - no matter what register I try to read from. Note that before putting the motor in an 'overloaded' state, the itemRead function works just fine. Is there a reason this function breaks after a motor error occurs? I'm not trying to enable the torque or even write to a register - just trying to read the register so I can know in software if the motor is in an error state or not.

Thanks

swiz23, 2021-07-04:
Any update on this?

ROBOTIS-Will, 2021-07-05:
Hi @swiz23 
Sorry about the delayed response on this.
I've created a new branch #335 for this issue which dxl_error won't block reading from DYNAMIXEL while blocking writing to DYNAMIXEL for safety purpose.

DYNAMIXEL Workbench has been updated with a limitation such as adding a new DYNAMIXEL, but without fixing bugs or adding new features.
We're planning to upgrade the DYNAMIXEL SDK so that it can be a substitute for the Workbench though there could be some significant changes.

Thank you.

swiz23, 2021-07-05:
Got it. Thanks! Will the changes be merged into master eventually?

ROBOTIS-Will, 2021-07-06:
I think these changes are pretty safe to be merged as it only allows to "read" from DYNAMIXEL.
I'll merge this change into master and release after receiving a confirmation from you application.
Thanks!

ROBOTIS-Will, 2021-07-30:
Hi @swiz23
Have you had chance to try the code #335?
Unless there's an issue with this change, I'll merge it to develop and prepare for the next release.
Thanks!

swiz23, 2021-07-30:
Hi @ROBOTIS-Will, 
I have not tested this since I don't have any Dynamixels on me. But I'll ask a friend of mine who does to test this and get back to you.

ROBOTIS-Will, 2021-07-30:
@swiz23 
Not a problem. I'll merge this into the develop branch for now :)

LSinterbotix, 2022-01-27:
@ROBOTIS-Will We were able to verify that the patch does work. You can feel free to close this. Thanks!

ROBOTIS-Will, 2022-01-27:
@LSinterbotix 
Thanks for the confirmation! 
I'll include this in the next release.

yun-goon, 2025-01-13:
This issue appears to be resolved, so it will be closed.

PR #335 (2021-07-30): Hotfix dxl error
This may resolve #307 
While there's any hardware error flag set in Hardware Error Status register, the Workbench had been blocking the use of reading from / writing to DYNAMIXEL register.
This hotfix will allow reading from DYNAMIXEL even if the Hardware Error Status is set.
Writing to DYNAMIXEL will still remain the same for the safety purpose as it is hard to identify the use case.
closed 2025-01-13 by yun-goon