ROBOTIS-GIT/dynamixel-workbench #316: [ERROR] [1604201796.570353381]: [TxRxResult] There is no status packet!

julianher, 2020-11-01:
ISSUE TEMPLATE ver. 1.0.0

Before you open issue, please refer to [ROBOTIS e-Manual](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/)

1. How to setup? (ex, U2D2, OpenCR,...)
    USB2Dynamixel
2. Which Dynamixel have you used? and how many? **(Please describe below format to all connected Dynamixels)**

   - Model Name AX-12A

   - ID 1

   - Baud Rate of Dynamixels 1000000

   - Protocol Version i guess 1
 
3. Write down the commands you used in order

  ```
 $ rosrun dynamixel_workbench_controllers find_dynamixel /dev/ttyUSB0

it found my servo: 

[ INFO] [1604200550.830647675]: Succeed to init(1000000)
[ INFO] [1604200550.830667731]: Wait for scanning...
[ INFO] [1604200559.467147520]: Find 1 Dynamixels
[ INFO] [1604200559.467172052]: id : 1, model name : AX-12A

but then when I try:

  ```
 
4. Copy and Paste your error message on terminal
[ERROR] [1604201166.874318684]: [TxRxResult] There is no status packet!
[ERROR] [1604201166.875144574]: Can't find Dynamixel ID '1'
[ERROR] [1604201166.875169860]: Please check Dynamixel ID or BaudRate
================================================================================REQUIRED process SUMMARY
========

PARAMETERS
 * /dynamixel_info: /opt/ros/melodic/...
 * /dynamixel_workbench/dxl_read_period: 0.01
 * /dynamixel_workbench/dxl_write_period: 0.01
 * /dynamixel_workbench/mobile_robot_config/radius_of_wheel: 0.033
 * /dynamixel_workbench/mobile_robot_config/seperation_between_wheels: 0.16
 * /dynamixel_workbench/publish_period: 0.01
 * /dynamixel_workbench/use_cmd_vel_topic: False
 * /dynamixel_workbench/use_joint_states_topic: True
 * /dynamixel_workbench/use_moveit: False
 * /rosdistro: melodic
 * /rosversion: 1.14.9

NODES
  /
    dynamixel_workbench (dynamixel_workbench_controllers/dynamixel_workbench_controllers)

auto-starting new master
process[master]: started with pid [10712]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 71685dae-1bf3-11eb-be4e-3417ebbecf4e
process[rosout-1]: started with pid [10723]
started core service [/rosout]
process[dynamixel_workbench-2]: started with pid [10726]
[ERROR] [1604201796.570353381]: [TxRxResult] There is no status packet!
[ERROR] [1604201796.571159650]: Can't find Dynamixel ID '1'
[ERROR] [1604201796.571189273]: Please check Dynamixel ID or BaudRate
================================================================================REQUIRED process [dynamixel_workbench-2] has died!
process has finished cleanly
log file: /home/julian/.ros/log/71685dae-1bf3-11eb-be4e-3417ebbecf4e/dynamixel_workbench-2*.log
Initiating shutdown!
================================================================================
[dynamixel_workbench-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
  
5. Please, describe detailedly what difficulty you are in 

    - Hey maybe this is a nooby question cause I am new using ROS and Dynamixel I'm connecting my Servo by a USB2Dynamixel and I am getting that error. As you can see the servo works with a baud rate of 1000000 I have already set-up that in the dynamixel_controllers.launch but I keep getting the same error.

ROBOTIS-Will, 2020-11-04:
Hi @julianher 
I'll review this issue and get back to you.
In the meantime, could you list up the commands you ran?
Thanks.

julianher, 2020-11-04:
> Hi @julianher
> I'll review this issue and get back to you.
> In the meantime, could you list up the commands you ran?
> Thanks.

Hi thank you for helping me with this.

ok so first I am using ROS melodic, AX-12A, USB2Dynamixel and the corresponding power supply. 

I'm following this tutorial: https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/

the commands I ran are:
1. In order to give permissions on the USB  
$ sudo chmod 777 /dev/ttyUSB0    

2. To find  my Servo, ID and baut rate:
$rosrun dynamixel_workbench_controllers find_dynamixel /dev/ttyUSB0
the result is this:

[image]

As you can see the servo has an ID of 1 and a baut rate of 1000000

3. to start the dynamixel_controllers
$roslaunch dynamixel_workbench_controllers dynamixel_controllers.launch

and the result of this is:
[image]

The launch file is this:

[image]

where you can see that I have set the baut rate to 1000000

and the yaml file is this:

[image]

julianher, 2020-11-05:
> Hi @julianher
> I'll review this issue and get back to you.
> In the meantime, could you list up the commands you ran?
> Thanks.

I think making any change in the .launch file is having no effect. I mean, I am supposed to set the baud rate in this file. 

To check if any change in this file has effect I deleted all the information in this file and the result is the same.   I have even deleted the file and when I call
roslaunch dynamixel_workbench_controllers dynamixel_controllers.launch  it still does the same. shouldn't I get a no launch found error?   I have also done catkin build in my ws.  Thus I think when I am setting baud rate to 1000000 this changes are having no effect.

ROBOTIS-Will, 2020-11-05:
Hi @julianher 
I have tested with my AX-18A with U2D2. When you are modifying the yaml or launch file, you don't have to rebuild the source.

- file structure and build
  - /home/willson/catkin_ws/src/dynamixel-workbench
  - /home/willson/catkin_ws/src/dynamixel-workbench-msgs
  - ran `catkin_make` on `/home/willson/catkin_ws`

- basic.yaml file
```
pan:
  ID: 1
  Return_Delay_Time: 0
# tilt:
#   ID: 2
#   Return_Delay_Time: 0
```

- dynamixel_controllers.launch file
```
<launch>
  <arg name="usb_port"                default="/dev/ttyUSB0"/>
  <arg name="dxl_baud_rate"           default="1000000"/>
  <arg name="namespace"               default="dynamixel_workbench"/>
...
```

- Result 
  - `roscore` on the top terminal
  - `rosrun dynamixel_workbench_controllers find_dynamixel /dev/ttyUSB0` and `roslaunch dynamixel_workbench_controllers dynamixel_controllers.launch` on the bottom terminal
[image]

julianher, 2020-11-05:
@ROBOTIS-Will yes it worked. My actual dynamixel_workbench is installed as an apt so the .launch file I am supposed to modify is located in the /opt ... I modified the .launch and the .yaml using sudo gedit.

this happened because I used sudo apt-get install instead of  git clone

yuya-0411, 2021-05-14:
@ROBOTIS-Will 
I also need your help.

I have the similar issue with @julianher, so I tired your solution. But it didn't work and I still the problem. 

I use XL-320 and USB2Dynamixel. 
ID: 3
Baud Rate of Dynamixels 1000000
Protocol Version : 2.0

I'll show the situation below.
[image]

As you can see, ID is found but items is not. I doublechecked items on https://emanual.robotis.com/docs/en/dxl/x/xl320/ . I can't understand why the program failed to write Return_Delay_Time to Dynamixel.

I tried to fix it by myself over and over but I couldn't.
I hope your help. thank you

ROBOTIS-Will, 2021-05-17:
@yuya-0411 The default baudrate of XL-320 is 1000000bps. Have you checked the dxl_baud_rate argument in the launch file?
https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/master/dynamixel_workbench_controllers/launch/dynamixel_controllers.launch#L3

yuya-0411, 2021-05-17:
@ROBOTIS-Will 
Thank you for your reply.

I checked it over and over. And I doublechecked the baud rate of dynamixels I use.
However, the same error appears.

I think baud rate and connection of dynamixels are well because all dynamixels can be found in 1000000 bps when I also run find_dynamixel program. 

I'm not sure though, is VMware related to the error?
I have ROS on Ubuntu18.04 on VMware.

ROBOTIS-Will, 2021-05-17:
@yuya-0411 
Hmmm... I haven't run on a virtual machine, but I don't think it isn't much different.

Make sure that the physical USB port is connected to the VMware and check if your VMware uses the same port(`/dev/ttyUSB0`) addressed in the launch file.

USB2Dynamixel has a TTL / RS-485 conversion switch and check if this switch is properly set to `TTL` side.

Have you also configured the basic.yaml file to match your DYNAMIXEL?
https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/master/dynamixel_workbench_controllers/config/basic.yaml

yuya-0411, 2021-05-18:
@ROBOTIS-Will 

I checked USB port. It is /dev/ttyUSB0. Actually, I use its port when I run find_dynamixel program.
[image]

I always make sure that a switch on USB2Dynamixel is set to TTL side.

I've configured the basic.yaml file along with XL-320 specification.
https://emanual.robotis.com/docs/en/dxl/x/xl320/
I will show my basic.yaml and error the below.
[image]
[image]
I don't know why the error is different with what I showed before a little bit. But there is still 'no status packet' message. These errors last until C-ctrl is pressed.

Are there anything wrong in my basic.yaml?

ROBOTIS-Will, 2021-07-30:
@yuya-0411 
I'm sorry for taking long to get back on this.
Has this issue been resolved?

yuya-0411, 2021-07-30:
@ROBOTIS-Will 

I couldn't solve the above problem. But I leaned to use dynamixels on ROS by shifting from dynamixel_workbench to DynamixelSDK. DynamixelSDK allow me to use dynamixels by just doing the below.

```
~/catkin/src 
git clone https://github.com/ROBOTIS-GIT/DynamixelSDK.git 
cd .. 
catkin_make 
```

Of course, I need to check USB PORT.
I actually control dynamixels (not only XL-320 but also XM430-W350-R).

However other problem happened. somehow, some of 5 dynamixels can't constantly receive command. That's why, the robot behaves a little bit weirdly. But this problem is not related to here problem.

thank you

ROBOTIS-Will, 2021-07-30:
@yuya-0411 
Thank you for the updates.
Please feel free to create a new issue thread regarding the problem you are experiencing so we can freely discuss more about it as well as creating some useful examples to resolve your problem.
Thank you.

sunzhon, 2022-02-17:
I also encounter this issue when I used a USB-RS485 transfer instead of D2D to connect XM430-W350 motors. So I am curious whether the workbence can also be used to communicate with dyanmixel motors via a common USB-RS485 transfer, such as FTDI FT2232H.

ROBOTIS-Will, 2022-02-18:
Hi @sunzhon 

In case of customized communication interface, I cannot guarantee whether will it work or not.
The Workbench should be working fine if the interface device works correctly.
Thank you.

yun-goon, 2025-01-13:
This issue will be closed due to prolonged inactivity. If the issue persists, feel free to reopen it.

closed 2025-01-13 by yun-goon