ROBOTIS-GIT/dynamixel-workbench #319: XL320 control_table Present_Position-element only gives value between 0-255

Benoit-LdL, 2020-12-16:
1. How to setup? (ex, U2D2, OpenCR,...)
OpenCM9.04 and using the dynamixel-workbench library in arduino

2. Which Dynamixel have you used? and how many? **(Please describe below format to all connected Dynamixels)**

   - Model Name
XL320
   - ID
1-16
   - Baud Rate of Dynamixels
1000000
   - Protocol Version
 v2
3. Code used
! this is only the first part of the code where data from the XL320 servos get read:
'
if (cmd[0] == "control_table")
      {
        uint8_t id = cmd[1].toInt();

        const ControlItem *control_item =  dxl_wb.getControlTable(id);
        uint8_t the_number_of_control_item = dxl_wb.getTheNumberOfControlItem(id);

        uint16_t last_register_addr = control_item[the_number_of_control_item-1].address;
        uint16_t last_register_addr_length = control_item[the_number_of_control_item-1].data_length;

        uint32_t getAllRegisteredData[last_register_addr+last_register_addr_length];
...
'

4. Please, describe what difficulty you are in 

I came across this issue on both the OpenCM9.04 arduino example "p_monitor" AND my custom code where I use the same code as the control_table-function in the "p_monitor" arduino example. 

The "Present_Position" element only gives a value back between 0 and 255 while the "Goal_Position" element accepts a value between -1024 and 1024 (if i'm not mistaking) OR a radian value which can also be negative. This is very useful because you can send an angle, converted to radian, and the servo wil move to that position (both positive angle as negative angle).

THE PROBLEM
I'm using the "control_table" elements to get the state of each servo and publish all this data to ROS topics.

As the incoming "Present_Position" can only be between 0-255 it can never show the exact position of the servo...

If this is not a bug, how can I get a correct reading of the present servo postition?
If this explanation is not clear I will add images/videos to this issue.

I hope someone can help me with this!

ROBOTIS-David, 2020-12-22:
using the "[getPresentPositionData](https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/#bool-getpresentpositiondatauint8_t-id-int32_t-data-const-char-log--null)" or "[itemRead](https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/#bool-itemreaduint8_t-id-const-char-item_name-int32_t-data-const-char-log--null)" method would be a better option if the user wants to read the current position valute. 

I simply tested the methods mentioned above, and it works as below. 

[image]

If you want to read multiple data on Control Table, consider using the [Sync Read](https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/#bool-addsyncreadhandleruint16_t-address-uint16_t-length-const-char-log--null) method.

ROBOTIS-Will, 2020-12-22:
Hi @Benoit-LdL 
I'm somewhat unclear about your issue and how to reproduce it, but below is the modified `h_Position` example.
I was able to get 0 ~1023 present position using the "Present_Position" control table item name.
Please let me know if I'm misunderstanding what you are trying to explain.

```cpp
#include <DynamixelWorkbench.h>

#if defined(__OPENCM904__)
  #define DEVICE_NAME "1" //Serial1 : OpenCM9.04 DXL port. I directly connected XL-320 to OpenCM9.04 without OpenCM485EXP board.
#elif defined(__OPENCR__)
  #define DEVICE_NAME ""
#endif   

#define BAUDRATE  1000000  //default baudrate of XL-320
#define DXL_ID    1

DynamixelWorkbench dxl_wb;

void setup() 
{
  Serial.begin(57600);
  while(!Serial); // Wait for Opening Serial Monitor

  const char *log;
  bool result = false;

  uint8_t dxl_id = DXL_ID;
  uint16_t model_number = 0;
  int32_t present_pos = 0;

  result = dxl_wb.init(DEVICE_NAME, BAUDRATE, &log);
  if (result == false)
  {
    Serial.println(log);
    Serial.println("Failed to init");
  }
  else
  {
    Serial.print("Succeeded to init : ");
    Serial.println(BAUDRATE);  
  }

  result = dxl_wb.ping(dxl_id, &model_number, &log);
  if (result == false)
  {
    Serial.println(log);
    Serial.println("Failed to ping");
  }
  else
  {
    Serial.println("Succeeded to ping");
    Serial.print("id : ");
    Serial.print(dxl_id);
    Serial.print(" model_number : ");
    Serial.println(model_number);
  }

  result = dxl_wb.jointMode(dxl_id, 0, 0, &log);
  if (result == false)
  {
    Serial.println(log);
    Serial.println("Failed to change joint mode");
  }
  else
  {
    Serial.println("Succeed to change joint mode");
    Serial.println("Dynamixel is moving...");

    for (int count = 0; count < 3; count++)
    {
      dxl_wb.goalPosition(dxl_id, (int32_t)0);
      delay(3000);
      if(dxl_wb.itemRead(dxl_id, "Present_Position", &present_pos)) {
        Serial.println(present_pos);
      }

      dxl_wb.goalPosition(dxl_id, (int32_t)1023);
      delay(3000);
      if(dxl_wb.itemRead(dxl_id, "Present_Position", &present_pos)) {
        Serial.println(present_pos);
      }
    }
  }
}

void loop() 
{

}
```

Benoit-LdL, 2020-12-22:
> using the "[getPresentPositionData](https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/#bool-getpresentpositiondatauint8_t-id-int32_t-data-const-char-log--null)" or "[itemRead](https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/#bool-itemreaduint8_t-id-const-char-item_name-int32_t-data-const-char-log--null)" method would be a better option if you would like to read the current position valute.
> 
> I simply tested the methods mentioned above, and it works as below.
> 
> [image]
> 
> If you want to read multiple data on Control Table, consider using the [Sync Read](https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/#bool-addsyncreadhandleruint16_t-address-uint16_t-length-const-char-log--null) method.

Hi @ROBOTIS-David , this does indeed work and I used the  "getPresentPositionData" function for my project.
Hi @ROBOTIS-Will, in the "h_position" arduino example it does indeed read 2 bytes to get a value between 0-1023.
However, the problem lays in the "p_monitor" arduino example where the "control_table" function only reads 1 byte for the "present_position" and thus only shows a value between 0-255

video link: https://photos.app.goo.gl/A8jKuxeK98SvcSTeA 

In the video I use an original, un-edited version (except device_name =1) of the "p_monitor" arduino sketch and you can see that I use the "control_table" function for the servo with ID 3. When looking at the "present_position" element you can see that when I turn the servo with ID 3 the value stays between 0-255. (sorry for the small text size on the serial monitor, but I wanted to show the servo too ;) ). 

If my explanation is still not clear, please tell me and I'll try to explain it better haha ;)

Kind regards and merry Christmas,

Benoît Lagasse

ROBOTIS-Will, 2020-12-23:
@Benoit-LdL 
Oh, thank you very much for the video attachment!
There was a missing code in the source code that processes data length.
I'll fix the issue today and you should be able to update the OpenCM9.04 Arduino board manager by tomorrow at the latest.
Thank you.

ROBOTIS-Will, 2020-12-23:
@Benoit-LdL 
OpenCM9.04 board manager [has been updated](https://github.com/ROBOTIS-GIT/OpenCM9.04/releases).
Please update your manager from Arduino IDE.
Thank you for your patience.

Benoit-LdL, 2020-12-23:
@ROBOTIS-Will thanks for the quick response!

Kind regards,
Benoît Lagasse

closed 2021-07-30 by ROBOTIS-Will