ROBOTIS-GIT/dynamixel-workbench #320: Bulk read/write Dynamixel XM-430s

stonkens, 2020-12-24:
ISSUE TEMPLATE ver. 1.0.0

Before you open issue, please refer to [ROBOTIS e-Manual](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/)

1. _How to setup? (ex, U2D2, OpenCR,...)_

2. _Which Dynamixel have you used? and how many? **(Please describe below format to all connected Dynamixels)**_

   - Model Name: XM-430

   - ID: 3

   - Baud Rate of Dynamixels: 57600

   - Protocol Version: 2.0
 
3. _Write down the commands you used in order_
Not relevant
4. _Copy and Paste your error message on terminal_
Not relevant
  
5. Please, describe detailedly what difficulty you are in 
I have a couple of questions re the documentation and `dynamixelSDK` vs `dynamixel_workbench`. My setup is as follows, 4 XM-430-W350-T, for which I want to be able to do position, velocity and torque control - not at the same time obviously. They are daisy-chained with a U2D2 
- The tutorials on [Workbench tutorials](http://wiki.ros.org/dynamixel_workbench_controllers/Tutorials) seem obsolete (clicking on the link guides to E-manual but they are still available, e.g. [Torque Control](http://wiki.ros.org/dynamixel_workbench_controllers/Tutorials/TorqueControl) ), is this correct? I followed the tutorials in the e-manual but these are a bit limited in terms of functionality.
- I want to be able to set different Goal (current, position or velocity) for each dynamixel. If I understand correctly, I need to use some form of `bulkWrite` to achieve this. What is the best way to achieve this in interface with ROS. I want to send a new command at minimally 20 Hz, ideally higher, with a different Goal target. Should I use DynamixelSDK or Dynamixel_workbench? For both, could you guide me to an example implementation

Thank you for getting back to me! Happy Holidays!

ROBOTIS-Will, 2020-12-24:
Hi @stonkens 
DYNAMIXEL Workbench was originally designed to support DYNAMIXEL configuration on Linux, but as we have introduced DYNAMIXEL Wizard 2.0 for all three major operating systems, the Workbench will be deprecated soon.
For this reason, I recommend to use DYNAMIXEL SDK.
Also, if you wish to accomplish higher control frequency, it is critical to minimize the communication packet size and delays.

For your case, [Sync Read](https://emanual.robotis.com/docs/en/dxl/protocol2/#sync-read) and [Sync Write](https://emanual.robotis.com/docs/en/dxl/protocol2/#sync-write) will be the feasible solution. However, in order to use them, you have to meet certain conditions; data has to be sequentially located in the memory address.

Your case, Goal Current(102, 2byte) and Goal Velocity(104, 4byte) are sequentially saved in the control table while the Goal Position(116, 4byte) is separated. Adopting Indirect Address and Indirect Data can resolve this problem as shown below diagram.
Once you assign the memory addresses of desired items to the value of each indirect address, you can access those data by accessing indirect data addresses.
Therefore, you can read or write addresses from 224 to 233 to access Goal Current, Goal Velocity, and Goal Position with sync read/write command.
[image]

You can refer to DYNAMIXEL SDK example for Sync feature.
https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/master/c%2B%2B/example/protocol2.0/sync_read_write/sync_read_write.cpp

Below video clip will be helpful to install DYNAMIXEL SDK for ROS.
https://youtu.be/SpdxjsCO9sE

Happy Holidays and a happy new year!

stonkens, 2020-12-30:
Perfect, thanks a lot. That indeed enables much higher control frequencies!

closed 2020-12-30 by stonkens