ROBOTIS-GIT/dynamixel-workbench #321: position, effort, vel feedback are no any change

codyjing, 2021-01-04:
1. setup with U2D2

2. used 3 xm430-w350-r

   - joint0,joint1,joint3

   - 1,2,3

   - 115200

   - Protocol 2.0
 
3. Write down the commands you used in order

  $ roscore    
  $ roslaunch dynamixel_workbench_controllers dynamixel_controllers.launch
  $ roslaunch dynamixel_workbench_operators joint_operator.launch
  $ rosservice call /dynamixel_workbench/execution 

 
 
4. Copy and Paste your error message on the terminal 
 when the joints are moving, the topic joint_states publishes at 100Hz but the position, effort, vel are no any change. it is completely wrong.
[image]

closed 2021-01-05 by codyjing

Helloworld-Le, 2022-07-03:
Hi I am in a very similar situation:

1. setup with U2D2 (PC to U2D2 to one XL430W250, the other port of  XL430W250 is connected to 12 DC output from OpenCR board)
2.  code used:
 $ roscore
$ roslaunch dynamixel_workbench_controllers dynamixel_controllers.launch
$ roslaunch dynamixel_workbench_operators joint_operator.launch
$ rosservice call /dynamixel_workbench/execution
3. both SBC and remote PC are running ubuntu20.04
4. `PARAMETERS
 * /dynamixel_info: /home/le/Install/...
 * /dynamixel_workbench/dxl_read_period: 0.01
 * /dynamixel_workbench/dxl_write_period: 0.01
 * /dynamixel_workbench/mobile_robot_config/radius_of_wheel: 0.033
 * /dynamixel_workbench/mobile_robot_config/seperation_between_wheels: 0.16
 * /dynamixel_workbench/publish_period: 0.01
 * /dynamixel_workbench/use_cmd_vel_topic: False
 * /dynamixel_workbench/use_joint_states_topic: True
 * /dynamixel_workbench/use_moveit: False
 * /rosdistro: noetic
 * /rosversion: 1.15.14

NODES
  /
    dynamixel_workbench (dynamixel_workbench_controllers/dynamixel_workbench_controllers)

ROS_MASTER_URI=http://129.215.90.113:11311

process[dynamixel_workbench-1]: started with pid [15423]
[ INFO] [1656881305.060908573]: Name : camera_motor, ID : 10, Model Number : 1060
[ INFO] [1656881305.142854829]: [DynamixelDriver] Succeeded to add sync write handler
[ INFO] [1656881305.143087024]: [DynamixelDriver] Succeeded to add sync write handler
[ INFO] [1656881349.741417869]: 'camera_motor' is ready to move
[ INFO] [1656881349.745809422]: Succeeded to get joint trajectory!
[ INFO] [1656881349.746493176]: Dynamixel is moving
[ INFO] [1656881350.746536474]: Dynamixel is moving
`
5. problem: 
 my motor is moving as expected, however the /dynamixel_workbench/joint_states is not updating the velocity or position. I have noticed that when i only run  `roslaunch dynamixel_workbench_controllers dynamixel_controllers.launch` and before run   `roslaunch dynamixel_workbench_operators joint_operator.launch` , I can see the /dynamixel_workbench/joint_states is updating correctly when I try to manually twist the motor a tiny little bit. But after run `roslaunch dynamixel_workbench_operators joint_operator.launch` ,  the position and velocity in  /dynamixel_workbench/joint_states will just remain unchanged no matter how the motor is moving.

jan-krueger, 2022-11-30:
Hello,

I would like to add that this issue seems to persist. I have some data that looks like a step-lock function when in reality the motors make small, smooth, continues movements. The data looks like this. It almost appears as if the controller is not reading back the data frequently enough from the motors. 

[image]